open62541/open62541 #3596: Crash due to use of NULL session object in server

juangburgos, 2020-05-12:
## Crash Description

In [`processMSGDecoded`](https://github.com/open62541/open62541/blob/9ed0e090ccff4392dc8a2aa9994ba559b9a9c78f/src/server/ua_server_binary.c#L546) should there be a test condition on `session` when calling `((UA_SessionService)(uintptr_t)service)(server, channel, session ...` ?

I see there is a comment saying `/* Session lifecycle service. The session pointer can still be NULL. */`, but further in the stack there is an attempt to close on that `NULL` session with `Service_CloseSession` which causes a crash.

[image]

[image]

[image]

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [x] Critical issue

jpfr, 2020-05-13:
Fixed via #3597.

juangburgos, 2020-05-13:
Beautiful!

closed 2020-05-13 by juangburgos