open62541/open62541 #3604: Can't connect to 62541server(1.1-rc1) from 62541client(1.1-rc1)

yoosYyd, 2020-05-15:
But UaExpert connect to that UAserver endpoint  without any problems.
code of client below

```
#pragma comment(lib, "Ws2_32.lib")
#include <iostream>
#include <conio.h>
#include <signal.h>
#include "open62541/open62541.h"

UA_Client* client = NULL;
UA_StatusCode res;
void WriteTag4B(const char *tagName,int type,unsigned char *data)
{
	UA_WriteRequest wReq;
	UA_WriteRequest_init(&wReq);
	wReq.nodesToWrite = UA_WriteValue_new();
	wReq.nodesToWriteSize = 1;
	wReq.nodesToWrite[0].nodeId = UA_NODEID_STRING(2, (char*)tagName);
	wReq.nodesToWrite[0].attributeId = UA_ATTRIBUTEID_VALUE;
	wReq.nodesToWrite[0].value.hasValue = true;
	wReq.nodesToWrite[0].value.value.type = &UA_TYPES[type];
	wReq.nodesToWrite[0].value.value.storageType = UA_VARIANT_DATA;
	wReq.nodesToWrite[0].value.value.data = data;
	UA_WriteResponse wResp = UA_Client_Service_write(client, wReq);
	if (wResp.responseHeader.serviceResult != UA_STATUSCODE_GOOD)
	{
		printf("tag write err:0x%X\n", wResp.responseHeader.serviceResult);
	}
	UA_WriteRequest_deleteMembers(&wReq);
	UA_WriteResponse_deleteMembers(&wResp);
}
int main()
{
	/*WSADATA wsd = { 0 };
	WSAStartup(MAKEWORD(2, 2), &wsd);*/
	setlocale(LC_ALL, "Russian");
	///////////////////////////////////////////////////////////////////////////////////////
	client = UA_Client_new();
	UA_ClientConfig_setDefault(UA_Client_getConfig(client));
	UA_ClientConfig* config = UA_Client_getConfig(client);
	res = UA_Client_connectUsername(client, "opc.tcp://Computer:4888/", "User", "111");
	if (res != UA_STATUSCODE_GOOD)
	{
		printf("UA_Client_connectUsername: err: %X\n", res);
	}//80AE0000
	_getch();
	float fVal = 33;
	WriteTag4B("Z1.Density", UA_TYPES_FLOAT,(unsigned char*)&fVal);
	UA_Client_disconnect(client);
	UA_Client_delete(client);
}
```

And log

> [2020-05-15 13:51:05.086 (UTC+0300)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
> [2020-05-15 13:51:05.092 (UTC+0300)] warn/network       DNS lookup of Computer failed with error Или приложение не вызвало WSAStartup, или произошла ошибка в WSAStartup.
> 
> [2020-05-15 13:51:05.098 (UTC+0300)] warn/client        Could not open a TCP connection to opc.tcp://Computer:4888/
> UA_Client_connectUsername: err: 80AE0000

Looks like some problems with sockets initialization. Can any help me ?

jpfr, 2020-05-15:
Dns lookup failed.
Try the IP address instead of the host name.

yoosYyd, 2020-05-15:
> Dns lookup failed.
> Try the IP address instead of the host name.

How to do it?

> [2020-05-15 14:57:42.554 (UTC+0300)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
> [2020-05-15 14:57:42.559 (UTC+0300)] warn/network       DNS lookup of 127.0.0.1 failed with error Или приложение не вызвало WSAStartup, или произошла ошибка в WSAStartup.
> 
> [2020-05-15 14:57:42.564 (UTC+0300)] warn/client        Could not open a TCP connection to opc.tcp://127.0.0.1:4888/
> UA_Client_connectUsername: err: 80AE0000

jpfr, 2020-05-15:
Maybe the port is not correct.

yoosYyd, 2020-05-15:
> Maybe the port is not correct.

Here's log from UaExpert

> 15:06:41.393 | AddressSpaceModel  | open62541-based OPC UA Appl... | Browse succeeded.
> 15:06:41.387 | AddressSpaceModel  | open62541-based OPC UA Appl... | Browse succeeded.
> 15:06:41.386 | Server Node        | open62541-based OPC UA Appl... | Successfully connected UA server.
> 15:06:41.371 | Server Node        | open62541-based OPC UA Appl... | Connection status of server 'open62541-based OPC UA Application - None - None' changed to 'Connected'.
> 15:06:41.370 | AddressSpaceModel  | open62541-based OPC UA Appl... | Register for ModelChangeEvents failed: createMonitoredItems returned Good (0x00000000) for item 0
> 15:06:41.363 | Server Node        | open62541-based OPC UA Appl... | ApplicationUri: 'urn:open62541.server.application'
> 15:06:41.362 | Server Node        | open62541-based OPC UA Appl... | Found security policy 'http://opcfoundation.org/UA/SecurityPolicy#None'
> 15:06:41.361 | Server Node        | open62541-based OPC UA Appl... | Found endpoint 'opc.tcp://Computer:4888/'

port the same

jpfr, 2020-05-18:
Should be resolved via #3609.
Can you check?

yoosYyd, 2020-05-20:
> Should be resolved via #3609.
> Can you check?

Yes.It solution works.

closed 2020-05-21 by jpfr