open62541/open62541 #3605: Outdated top-level documents in master branch

ichrispa, 2020-05-15:
On behalf of AEbner as per Core Dev conference:

There appear to exist several outdated top-level document in master that require attention prior ot the 1.1 release (AUTHORS and CHANGELOG) where mentioned.

@andreasebner : Could you please specify which documents caught your attention?

andreasebner, 2020-06-03:
I started with the files in the root folder. Maybe it would make sense to create similar tables for folders like arch, deps and tools. In addition, there are also some outdated parts in the documentation. For example, the UA_BUILD_SELFSIGNED_CERTIFICATE is central mentioned in 'Building open6254', but within the code I can’t find the option anymore.

File | Status| Comment
-- | -- | --
  .clang-format | seems Up-to-Date |  |
  .clang-tidy | seems Up-to-Date | |
  .cquery | unknown | |
  .dockerignore | seems Up-to-Date | |
  .editorconfig | seems Up-to-Date | |
  .gitignore | Up-to-Date | |
  .gitmodules | Up-to-Date | Maybe we should use Tag Errata-1.04.6 for the UA Nodeset|
  .travis-apt-pin.preferences | unknown | 3 years unchanged|
  .travis.yml | Up-to-Date | |
  AUTHORS | seems outdated | no changes in 2 years? |
  CHANGELOG | outdated | last entry from 2018|
  CMakeLists.txt | Up-to-Date | |
  CODE_OF_CONDUCT.md | seems Up-to-Date | |
  CONTRIBUTING.md | Up-to-Date | |
  CPPLINT.cfg | seems Up-to-Date | |
  FEATURES.md | outdated | e.g. 'WIP for Release 0.4' |
  ISSUE_TEMPLATE.md | Up-to-Date | |
  LICENSE | Up-to-Date | |
  LICENSE-CC0 | Up-to-Date | |
  README.md | seems Up-to-Date | |
  TinyDockerfile | unknown | 3 years unchanged|
  appveyor.yml | Up-to-Date | |
  azure-pipelines.yml | Up-to-Date | |
  cppcheck-suppressions.txt | seems Up-to-Date | |
  open62541.pc.in | seems Up-to-Date | |
  open62541.spec | seems outdated | e.g. Version:  0.3 + changelog from 2017 |
  sonar-project.properties | seems Up-to-Date|

NoelGraf, 2021-05-28:
File | Status | Comment
-- | -- | --
.clang-format | seems Up-to-Date | &nbsp;
.clang-tidy | seems Up-to-Date | &nbsp;
.cquery | unknown | &nbsp;
.dockerignore | seems Up-to-Date | &nbsp;
.editorconfig | seems Up-to-Date | &nbsp;
.gitignore | Up-to-Date | &nbsp;
.gitmodules | Up-to-Date | Maybe we should use Tag Errata-1.04.6 for the UA Nodeset
.travis-apt-pin.preferences | unknown | 4 years unchanged
.travis.yml | Up-to-Date | Use of Travis discontinued?
AUTHORS | outdated | no changes in 3 years
CHANGELOG | outdated | last entry from 2018
CMakeLists.txt | Up-to-Date | &nbsp;
CODE_OF_CONDUCT.md | seems Up-to-Date | &nbsp;
CONTRIBUTING.md | Up-to-Date | &nbsp;
CPPLINT.cfg | seems Up-to-Date | &nbsp;
FEATURES.md | seems Up-to-Date | &nbsp;
ISSUE_TEMPLATE.md | Up-to-Date | &nbsp;
LICENSE | Up-to-Date | &nbsp;
LICENSE-CC0 | Up-to-Date | &nbsp;
README.md | seems Up-to-Date | &nbsp;
TinyDockerfile | unknown | 4 years unchanged
appveyor.yml | Up-to-Date | &nbsp;
azure-pipelines.yml | Up-to-Date | &nbsp;
cppcheck-suppressions.txt | seems Up-to-Date | &nbsp;
open62541.pc.in | seems Up-to-Date | &nbsp;
open62541.spec | seems outdated | e.g. Version: 1.0.1
sonar-project.properties | seems Up-to-Date | &nbsp;