open62541/open62541 #3609: Server-client connection cannot be established on Windows 10 localhost

saben23, 2020-05-18:
## Description
I am trying to establish a connection on **localhost** between the [Server](https://open62541.org/doc/1.0/tutorial_server_firststeps.html) and [Client](https://open62541.org/doc/1.0/tutorial_client_firststeps.html) with the example code that has been given in the Tutorial section of the documentation. For an extra I add a custom variable to the server which I want to read on the Client side. I use C++ and CMake in my sample project.

**My problem** is that after starting the Server when I start my Client a **connection cannot be established**. The logs represents that the Server is up and running but the Client could not connect to it because the DNS lookup failed for the host.

### Server log:
```bash
[2020-05-18 10:58:21.991 (UTC+0200)] debug/session      Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | Read the attribute 2
[2020-05-18 10:58:21.991 (UTC+0200)] warn/server        Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2020-05-18 10:58:21.992 (UTC+0200)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-05-18 10:58:21.996 (UTC+0200)] info/network       TCP network layer listening on opc.tcp://localhost:4840/
```
### Client log:
```bash
[2020-05-18 10:58:56.325 (UTC+0200)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-05-18 10:58:56.326 (UTC+0200)] warn/network       DNS lookup of localhost failed with error Either the application has not called WSAStartup, or WSAStartup failed.
[2020-05-18 10:58:56.327 (UTC+0200)] warn/client        Could not open a TCP connection to opc.tcp://localhost:4840/
[2020-05-18 10:58:56.327 (UTC+0200)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadConnectionClosed`
```
### Finding:
After some searching in the Issues I found a solution in this [issuse](https://github.com/open62541/open62541/issues/3481) that solved my problem. 
The solution was to add the **UA_initialize_architecture_network()** function call to **UA_ClientConnectionTCP_init** function in the **network_tcp.c** file. 

### Question:
My question is that what is the reason that I cannot establish a connection without modifying the Library code? Am I missing something somwhere? 

## Background Information / Reproduction Steps
* **CMake version:** 3.17.0
* **Visual Studio**: 2019 Enterprise Version 16.4.6

Used CMake options:

```bash
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_LOGLEVEL=200 ..
```
* After the above command I built in Visual Studio for RelWithDebInfo target. 
* After the build I installed the library with the following command
```bash
cmake --build . --target install 
```
I attached the project code with the example server and client code for easier reproduction. I usually built it for x64 targets. 
[test_opc.zip](https://github.com/open62541/open62541/files/4643341/test_opc.zip)

Thank you for your help in advance.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.1.0-rc1-13-gcdcb97de
         - Downloaded the .zip from the **pack/master** branch
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10 Enterprise (64-bit)
         - Version: 1909 
         - OS build: 18363.778 
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue

jpfr, 2020-05-18:
As you have a local fix (and a Github account), could you post a pull-request?

If not, we can do it if you give short heads-up.

saben23, 2020-05-18:
Thanks for the quick response. I posted a pull request with the change. 
Link:
https://github.com/open62541/open62541/pull/3610

saben23, 2020-05-18:
The pull-request has been accepted and rebased on the [1.1 branch](https://github.com/open62541/open62541/tree/1.1) therefore I close this issue.

closed 2020-05-18 by saben23