open62541/open62541 #3612: User/Role based authorization of a node in open62541

OPCUAUser, 2020-05-19:
Hi Community,

Is there a support for role based authorization in open62541. If its not supported, what is the roadmap for the same? I need to implement a user based authorization for the nodes.

For example, I have User/Roles A, B and C whereas the nodes are 1001, 1002.
User/Role A: can read/write the node 1001 & no permission for 1002
User/Role B: can read/write the node 1002 & no permission for 1001
User/Role A: can read/write the node 1001 & 1002 (all permissions)

Your support would be really helpful for me.

Thanks!

jpfr, 2020-05-19:
We don't have the "Roles" which were added in OPC UA v1.04.
But there is fine-grained access control.

You can fork the "AccessControl" plugin:
https://github.com/open62541/open62541/blob/master/plugins/ua_accesscontrol_default.c
The code is licensed as CC0. So you can combine it with commercial code.

Here, you get the credentials of a user during login.
To verify the credentials and bind custom data to the session.

The session context is forwarded back into the AccessControl plugin to verify every action on the information model.

OPCUAUser, 2020-05-21:
Hello @jpfr:

Thanks for the information, it would be helpful for me. Also, Is there a roadmap for the addition of Roles in the stack?

Best regards!

viktor-morin, 2020-10-01:
@OPCUAUser 
Did you implement this?
Any example you can share?
Thanks!

NoelGraf, 2021-04-16:
No, there is currently no roadmap to add this feature.
I would then close the issue. If there should be something else, the issue can be opened again.

closed 2021-04-16 by NoelGraf