open62541/open62541 #3614: Function UA_ServerConfig_setDefaultWithSecurityPolicies for encryption doesn't work

luigicapogrosso, 2020-05-19:
## Description
I'm trying to implement Encryption for my "open62541" Server which i connect with the UaExpert-Client. I tried to modify the server config, but i have the following linking error:
`undefined reference to UA_ServerConfig_setDefaultWithSecurityPolicies`

What could be the problem? Has the function become deprecated after a few updates?

## Background Information / Reproduction Steps
I'm using _exactly_ this example  on how to use encryption: 
https://github.com/open62541/open62541/blob/master/examples/encryption/server_encryption.c

An excerpt of the code is here:
```c
    /* Load certificate and private key */
    UA_ByteString certificate = loadFile(argv[1]);
    UA_ByteString privateKey = loadFile(argv[2]);

    /* Load the trustlist */
    size_t trustListSize = 0;
    if(argc > 3)
        trustListSize = (size_t)argc-3;
    UA_STACKARRAY(UA_ByteString, trustList, trustListSize);
    for(size_t i = 0; i < trustListSize; i++)
        trustList[i] = loadFile(argv[i+3]);

    /* Loading of a issuer list, not used in this application */
    size_t issuerListSize = 0;
    UA_ByteString *issuerList = NULL;

    /* Loading of a revocation list currently unsupported */
    UA_ByteString *revocationList = NULL;
    size_t revocationListSize = 0;

    UA_Server *server = UA_Server_new();
    UA_ServerConfig *config = UA_Server_getConfig(server);

    UA_StatusCode retval =
        UA_ServerConfig_setDefaultWithSecurityPolicies(config, 4840,
                                                       &certificate, &privateKey,
                                                       trustList, trustListSize,
                                                       issuerList, issuerListSize,
                                                       revocationList, revocationListSize);
```

## Checklist
 - [X] open62541 Version (release number or git tag): v1.1-rc1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Ubuntu 18.04 LTS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [X] Critical issue

jpfr, 2020-05-19:
Did you enable encryption support in CMake?

luigicapogrosso, 2020-05-19:
Yes, on my `CMakeLists.txt` I added:

```
option(UA_ENABLE_ENCRYPTION "Enable encryption support (uses mbedTLS by default)" OFF)
option(UA_ENABLE_ENCRYPTION_OPENSSL "Enable encryption support (uses openssl)" OFF)
option(UA_ENABLE_ENCRYPTION_MBEDTLS "Enable encryption support (uses mbedTLS)" OFF)

# security provider 
if(UA_ENABLE_ENCRYPTION)
    if(NOT UA_ENABLE_ENCRYPTION_OPENSSL)
        set(UA_ENABLE_ENCRYPTION_MBEDTLS ON)
    endif()
endif()
if(UA_ENABLE_ENCRYPTION_MBEDTLS)
    set(UA_ENABLE_ENCRYPTION ON)
    set(UA_ENABLE_ENCRYPTION_OPENSSL OFF)
endif()
if(UA_ENABLE_ENCRYPTION_OPENSSL)
    set(UA_ENABLE_ENCRYPTION ON)
    set(UA_ENABLE_ENCRYPTION_MBEDTLS OFF)
endif()  
```

jpfr, 2020-05-19:
You don't have to edit the CMakeLists.txt.
You can use cmake to set the config variables. Or graphical tools like `ccmake` or `cmake-gui` to edit the config.

luigicapogrosso, 2020-05-19:
Yes, I did it.
For encryption support  I also installed `libmbedtls-dev`.

luigicapogrosso, 2020-05-25:
Okai, I see where the problem might be. 
I have installed the library with the stable package in Ubuntu, using these lines:
```
sudo add-apt-repository ppa:open62541-team/ppa
sudo apt-get update
sudo apt-get install libopen62541-1-dev
```
With this installation, however, I understand that the part relating to the activation of the encryption is not compiled.

So, I cloned the library here from GitHub's repo and I use the graphical tools `ccmake` to edit the config variables, putting:
1. UA_ENABLE_ENCRYPTION -> ON
2. UA_ENABLE_ENCRYPTION_OPENSSL -> ON
3. UA_ENABLE_ENCRYPTION_MBEDTLS -> ON
but, it doesn't work anyway.

jpfr, 2020-05-27:
You only need either of
UA_ENABLE_ENCRYPTION_OPENSSL
UA_ENABLE_ENCRYPTION_MBEDTLS

Do you get any log output from the SecurityPolicy?
Which endpoints does the UAExpert show?

NoelGraf, 2021-05-06:
@luigicapogrosso Did it work? And can the issue be closed?

luigicapogrosso, 2021-05-10:
Yes, it works, therefore, this issue can be closed.

closed 2021-05-10 by luigicapogrosso