open62541/open62541 #3619: "Aes128-Sha256-RsaOaep" security policy

johndzilva, 2020-05-21:
Hello community,

As you all know that the "Basic128Rsa15" and "Basic256" security policies are no longer considered secure any more and it is deprecated in v1.04 spec.

I have implemented the "Aes128-Sha256-RsaOaep" security policy for my purpose. If you think that it is good feature for you then I can raise a pull request. (tested against uaexpert, OPCF reference client, UA browser)

If not, users can refer to the codes below and use it in their project. 
mbedtls->
https://github.com/johndzilva/open62541/commit/e9c5d0d1ddf0afc6218b4e5cc3a3b1b8c53ee048
openssl->
https://github.com/johndzilva/open62541/commit/ce928432a5b0952490efd39c6d19b231bef44d6f

jpfr, 2020-05-21:
Sounds great.
We can review/merge a PR from you on the topic.

johndzilva, 2020-05-27:
PR - https://github.com/open62541/open62541/pull/3632

Thanks @jpfr !

Before I close this issue, would you be interested in "Aes256-Sha256-RsaPss" security policy in open62541. If yes, then I can support and raise a PR for that too.
Currently "Aes256-Sha256-RsaPss" and "Basic256Rsa256" are the only security policy for configurations with **high security** needs.

jpfr, 2020-05-27:
Hehe, sure!
Do you the possibility to use more common functionality between the very similar SecurityPolicies?

johndzilva, 2020-05-27:
Do you mean to reduce the code size?

closed 2020-05-27 by johndzilva