open62541/open62541 #3620: rpc_for_client?

dotChris90, 2020-05-21:
hello all,

first I want to say how amazing this project is. never though that a so highly industry project could be open source and looks so brilliant.

I just was wondering if or if not it would be possible to extend server methods by a client. I know that the server can create method nodes with callbacks just like we all knows it from OOP.

now I thought if a client with administrator permissions could connect to the server and extending server model by add methods nodes with callbacks from client (like subscribing and waiting for other client to call, server would forward this rpc to client and also forward return type from client to client ).

my current understanding is that this would not be possible currently since the rpc (method callbacks) should be part of the server process and cannot be part of a client process. am I right? 

dw the issue here is just a question. no proposal. thanks for your time and might this project become more popular day by day 😉

ccvca, 2020-05-22:
As far as I know, this is not easily possible in OPC UA (https://reference.opcfoundation.org/v104/Core/docs/Part4/)
To achieve something like this, your "client" needs to spawn an OPC UA Server and your "Server" hat to connect to this (as an OPC UA Client). Then it could pass calls of a method to this Client.

jpfr, 2020-05-22:
Hello @dotChris90 ,

do you have a specific use case for that?
Note that object instantiation (from object types) reuses the existing method node from the type. That already covers many use cases.

We can create method nodes with the `AddNodes` service.
But we cannot currently automatically transfer binary code as the method callback.
The protocol does not support it. And there are obvious security reasons.

What you could do at the moment is use the constructor-callback for new nodes with a special convention, e.g. take the `description` attribute, to bind callbacks to new method nodes.
You could also look up symbols from a DLL/.so at runtime.

Or you add a special method to the information model that takes as argument both the attributes and the callback for the new method node.

There are bindings for TCL and Perl available (see the wiki). With those, it should be possible to attach executable source code as well.

dotChris90, 2020-05-22:
hello guys,

thanks for sharing your ideas.

@jpfr there is no direct use case. I just was investigating about the possibilities of dynamic extending the server methods on runtime.

something like different teams building services / methods for a server how to add them dynamicly. so mostly just theoretical thinking or brain storming about what is possible with opc ua.

we already were thinking about c++ based plug-in system with open dynamic libs on runtime and defining interfaces which always must exist in lib.

as alternative we thought about a forwarding system over the subscriber system. if a request on server happens a listener variable is set and client is informed. server forwarding is done on specific topic in specific message format.

also what was mentioned here with multiple server approache sound good. 

the version of plug in system and the multiple server approache sounds best so far. just not sure about the performance influence if multiple servers. but as my understanding the opc ua is optimized for embedded devices even running on microcontroller without any operating system. so I think the multiple server approach would be suites.

ccvca, 2020-05-22:
When you want to provide the address space of several OPC UA-Servers I do see two more options:

- Aggregating, as it is discussed here: https://github.com/open62541/open62541/issues/3560
- Local/Global Discovery Servers to find the Server by specific names

PS: If there are no more question this issue can be closed.

dotChris90, 2020-05-24:
@ccvca  - i close. and thanks for your fast reply ;)

closed 2020-05-24 by dotChris90