open62541/open62541 #3624: open62541.dll is not inside /bin or /lib directories after install

saukijan, 2020-05-22:
# Description

This issue is more of a note and acknowledgment of it's existence. I'll investigate it further when I have a bit of time.

While running a CI setup for [conan-central-index](https://github.com/conan-io/conan-center-index/pull/1687#issuecomment-632786531) I found that `open62541.dll` is built within the `/build` directory and not moved to `/bin` or `/lib` directories. This is fixed within the conan package, so it won't be a problem for users, just some unexpected cmake behavior.

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS=ON  -DUA_NAMESPACE_ZERO="FULL"  -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_ENABLE_METHODCALLS=ON -DUA_ENABLE_NODEMANAGEMENT=ON -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_SEMAPHORE=ON ..
```

## Solution

The following command should solve the issue, thought I am not certain why it comes up [install](https://github.com/open62541/open62541/blob/master/CMakeLists.txt#L1363) command should take care of it itself.

```cmake
if(WIN32)
   add_custom_command(TARGET open62541 POST_BUILD
            COMMAND ${CMAKE_COMMAND} -E copy_directory
            "${CMAKE_BINARY_DIR}/open62541.dll"
             ${CMAKE_INSTALL_LIBDIR}
    )
endif()
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): v1.0
 - [ ] Other OPC UA SDKs used (client or server):  server
 - [ ] Operating system: Windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached [CMake.log](https://c3i.jfrog.io/c3i/misc/logs/pr/1687/9/open62541/1.0.0/6cf96db5765ec2c04c7435986e096ab00f2a89d3/create_stdout.txt)
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

saukijan, 2020-05-22:
Just noticed that it was fixed in the latest maseter (`${CMAKE_INSTALL_PREFIX}` changed to `${CMAKE_INSTALL_BINDIR}`) I patched it for v1.0 and v1.0.1 conan packages

closed 2020-05-22 by saukijan

traversaro, 2020-05-22:
For reference, the PR that fixed it was https://github.com/open62541/open62541/pull/3347 .