open62541/open62541 #3626: About certificate signing encryption mode for OPCUA Client

wuhao1121, 2020-05-24:
hello all,
       When I used the opcua client to connect to the server in the way of signature encryption, I found that the opcua secure channel was always None. I don't know why and how to modify it. Do you have any information? Thank you very much.
Here is my code，
        UA_ByteString certificate = loadFile("");
	UA_ByteString privateKey = loadFile("");
	const char* username = "Administrator";
	const char* password = "123456";
	UA_Client *client = UA_Client_new();
	UA_ClientConfig *cc = UA_Client_getConfig(client);

	UA_ClientConfig_setDefaultEncryption(cc, certificate, privateKey,NULL, 0,NULL, 0);
	UA_ByteString_clear(&certificate);
	UA_ByteString_clear(&privateKey);
	cc->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
	cc->securityPolicyUri = 
        UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15");
	
	cc->clientDescription.applicationUri = UA_STRING_ALLOC("urn:feng- 
         PC");

	UA_StatusCode retval = UA_Client_connect_username(client, endpointUrl, username, password);

Here is my question，
       When you run the program，will log opened SecureChannel with SecurityPolicy 
       http://opcfoundation.org/UA/SecurityPolicy#None.
       I realized it wasn't an encrypted SecureChannel.

jpfr, 2020-05-24:
Which version do you use?
The first channel will be unencrypted to retrieve the endpoint configuration.

Then a new channel is opened that used encryption.

wuhao1121, 2020-05-24:
> Which version do you use?
> The first channel will be unencrypted to retrieve the endpoint configuration.
> 
> Then a new channel is opened that used encryption.
Hello, I am using pack\master, I wonder if this version is feasible? 
However, I found that if I shut down the server http://opcfoundation.org/UA/SecurityPolicy#None, the client would not be able to connect to the server and And certificate validation doesn't work。
Do you have any Suggestions，Please?

ZekeLandon, 2020-05-30:
Hi @wuhao1121, as @jpfr explained before, the very first connection would be using security policy None to discover the available endpoints in the network. Once it is identified, you shall any security policy there after to create secure channel.
 
Example here: https://github.com/open62541/open62541/blob/master/examples/encryption/client_encryption.c would be a good starting point.

Regards,
Zeke

NoelGraf, 2021-04-30:
@wuhao1121 can the issue be closed?

closed 2021-05-06 by NoelGraf