open62541/open62541 #3627: A crash occurs when the UA Server pointer is released using the UA Server delete method.

linghuqingyun, 2020-05-25:
Hello everyone:
I implemented a uaserver with open62541, but it crashed when I called the UA Server delete method to release the UA Server pointer.How do I release resources when I stop ua services?
This can happen with any version of an operating system.I'm using open62541-v1.0.

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issu

jpfr, 2020-05-25:
That’s not good.
Can you run the server with valgrind or drmemory to get more visibility in the crash?

We need logs to help you.

linghuqingyun, 2020-05-26:
Here are some debug logs.I used drmemory to get information.
[results202005251828.txt](https://github.com/open62541/open62541/files/4679405/results202005251828.txt)

jpfr, 2020-05-26:
The first problem is this one:
```
Error #5: INVALID HEAP ARGUMENT to free 0x00112804
# 0 replace_free                                              [d:\drmemory_package\common\alloc_replace.c:2706]
# 1 UA_Array_delete
# 2 ATL::CComCriticalSection::Init
# 3 ATL::CComCriticalSection::Init
# 4 UA_clear       
# 5 UA_ServerConfig_clean
# 6 UA_Server_delete
# 7 COPCUAManager::DeleteUaServer                             [f:\svn\chengs\chengs\io驱动\opc\uaserver\psopcuaserver\opcuamanager.cpp:312]
# 8 COPCUAManager::~COPCUAManager                             [f:\svn\chengs\chengs\io驱动\opc\uaserver\psopcuaserver\opcuamanager.cpp:63]
# 9 COPCUAManager::`scalar deleting destructor'
#10 COPCUAServerDlg::OnOpcuaserverExit                        [f:\svn\chengs\chengs\io驱动\opc\uaserver\psopcuaserver\opcuaserverdlg.cpp:1714]
```

Is it possible that you set a stack-allocated string in the config?

jpfr, 2020-06-11:
Seems to be resolved.
Reopen if required.

closed 2020-06-11 by jpfr