open62541/open62541 #3631: Problem with connection between UAExpert and open62541 server

nokem61, 2020-05-26:
## Description

UAExpert can't open Secure Channel under security policy
OPCF Sample Client work without any problem

## Background Information / Reproduction Steps

UA_STATUSCODE_BADINTERNALERROR is returned by mbedtls_decrypt_rsaOAep from code:
 if(data->length % rsaContext->len != 0)
        return UA_STATUSCODE_BADINTERNALERROR;
 
because data->length = 510

In decryptAndVerifyChunk chunk->length = 1728, offset = 1218

Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS:BOOL="1" -DUA_ENABLE_HISTORIZING:BOOL="1" -DUA_LOGLEVEL:STRING="100" -DUA_BUILD_EXAMPLES:BOOL="1" -DUA_ENABLE_ENCRYPTION:BOOL="1" -DUA_BUILD_TOOLS:BOOL="1" -DUA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL="1" -DUA_NAMESPACE_ZERO:STRING="FULL"  ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.1rc1
 - [ ] Other OPC UA SDKs used (client or server):  UAExpert
 - [ ] Operating system: Windows10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

nokem61, 2020-05-26:
[conversation.zip](https://github.com/open62541/open62541/files/4681258/conversation.zip)

jpfr, 2020-05-27:
How did you generate the certificate?
If not via our generator-script in /tools/certs, please show what information you can share.

See here for dumping the certificate details https://support.qacafe.com/knowledge-base/how-do-i-display-the-contents-of-a-ssl-certificate/

nokem61, 2020-05-27:
```
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number: 1 (0x1)
        Signature Algorithm: sha256WithRSAEncryption
        Issuer: DC = localhost, O = xx, L = xx, C = PL, CN = OPCUASrv
        Validity
            Not Before: May 27 09:40:58 2020 GMT
            Not After : May 27 09:40:58 2021 GMT
        Subject: DC = localhost, O = xx, L = xx, C = PL, CN = OPCUASrv
        Subject Public Key Info:
            Public Key Algorithm: rsaEncryption
                RSA Public-Key: (2046 bit)
                Modulus:
                    2e:50:33:27:4b:bb:98:24:8b:d9:36:10:08:40:7e:
                    a3:9a:fe:e5:36:96:a4:53:83:a0:86:17:1e:44:c4:
                    af:48:8d:a2:17:0e:d2:01:b1:68:ca:e8:30:2a:54:
                    09:cf:b8:f5:c8:71:46:0b:07:11:4e:f1:1c:e5:97:
                    25:0f:3a:c0:66:e1:4d:be:73:42:90:5d:a1:6f:ac:
                    af:38:95:06:8a:97:6c:b8:ad:bc:97:74:e3:b1:4a:
                    71:4c:ad:a9:5d:23:c1:0a:44:ca:0f:73:43:67:c6:
                    30:0d:cb:f7:e4:72:08:7c:63:86:95:d2:38:7e:80:
                    a8:94:e1:d9:99:bc:e8:65:41:fd:14:08:7f:ad:11:
                    b3:20:35:39:65:64:98:20:37:5f:6f:bf:87:1e:04:
                    48:6e:23:c9:39:de:24:f3:c0:23:9c:21:98:9e:73:
                    5c:56:04:52:0f:27:43:21:64:0e:86:09:b1:1f:ed:
                    9e:2c:8e:98:be:e1:2f:00:4a:bc:bc:d0:c9:ab:31:
                    ab:3b:aa:5c:87:5f:d8:f3:2f:a2:03:c1:1d:b0:b9:
                    89:5b:03:0c:ee:67:bb:8d:12:aa:e6:98:78:fa:72:
                    17:a4:1e:00:c8:34:41:49:23:60:90:51:f9:f5:6a:
                    0e:24:19:86:1b:08:ff:0d:66:22:84:c3:54:06:1c:
                    29
                Exponent: 65537 (0x10001)
        X509v3 extensions:
            X509v3 Subject Key Identifier:
                7D:F5:8C:DD:C1:33:E1:98:AC:6A:DC:B7:DD:54:82:C6:F2:CC:BD:68
            X509v3 Authority Key Identifier:
                keyid:7D:F5:8C:DD:C1:33:E1:98:AC:6A:DC:B7:DD:54:82:C6:F2:CC:BD:68
                DirName:/DC=localhost/O=xx/L=xx/C=PL/CN=OPCUASrv
                serial:01

            X509v3 Subject Alternative Name:
                URI:urn:localhost:OPCUASrv, DNS:localhost
            X509v3 Basic Constraints: critical
                CA:FALSE
            X509v3 Key Usage: critical
                Digital Signature, Non Repudiation, Key Encipherment, Data Encipherment, Certificate Sign
            X509v3 Extended Key Usage: critical
                TLS Web Server Authentication, TLS Web Client Authentication
    Signature Algorithm: sha256WithRSAEncryption
         1b:79:e1:80:94:42:8e:9c:c2:e4:80:a1:8f:d4:74:07:41:da:
         89:f5:de:f1:51:7c:7d:37:28:42:be:5d:4a:8b:2a:7d:2d:77:
         28:a8:50:18:27:ea:8d:6b:36:f4:38:44:ab:0e:90:3d:df:2e:
         e4:11:48:fe:94:27:dd:46:b5:aa:b0:a0:9c:a8:f0:00:ee:a4:
         f3:93:1e:2d:9e:6e:2c:81:6d:86:1a:53:20:f6:8c:4c:1e:56:
         30:57:bf:d6:5f:4c:30:24:29:f9:15:49:b9:0f:5d:bc:52:18:
         d9:f8:d9:b6:bc:b2:de:f9:41:bd:80:20:13:70:03:cf:8e:fb:
         d7:e0:dd:af:01:0e:77:86:53:d1:7f:0f:71:b9:b9:e1:35:01:
         ea:95:2b:b2:ae:ca:89:4c:d4:62:81:b8:bc:4d:07:e2:2b:c3:
         4b:ee:a1:d2:92:ab:7e:d6:da:6f:7b:74:9f:92:1b:c0:53:47:
         90:ce:ce:1b:a9:1d:02:2d:a7:43:dc:f1:8b:82:02:62:b7:25:
         24:26:76:98:4a:90:60:32:9b:29:f8:7b:0a:0a:41:c5:7e:68:
         fc:1b:7e:62:05:76:e9:5a:12:b1:ae:63:c5:61:b0:7a:cb:b1:
         49:57:c1:0a:cc:4d:88:2b:64:a5:0d:75:6d:db:da:2f:2d:72:
         5a:70:38:f7
-----BEGIN CERTIFICATE-----
MIIENjCCAx6gAwIBAgIBATANBgkqhkiG9w0BAQsFADBVMRkwFwYKCZImiZPyLGQB
GRYJbG9jYWxob3N0MQswCQYDVQQKDAJ4eDELMAkGA1UEBwwCeHgxCzAJBgNVBAYT
AlBMMREwDwYDVQQDDAhPUENVQVNydjAeFw0yMDA1MjcwOTQwNThaFw0yMTA1Mjcw
OTQwNThaMFUxGTAXBgoJkiaJk/IsZAEZFglsb2NhbGhvc3QxCzAJBgNVBAoMAnh4
MQswCQYDVQQHDAJ4eDELMAkGA1UEBhMCUEwxETAPBgNVBAMMCE9QQ1VBU3J2MIIB
ITANBgkqhkiG9w0BAQEFAAOCAQ4AMIIBCQKCAQAuUDMnS7uYJIvZNhAIQH6jmv7l
NpakU4OghhceRMSvSI2iFw7SAbFoyugwKlQJz7j1yHFGCwcRTvEc5ZclDzrAZuFN
vnNCkF2hb6yvOJUGipdsuK28l3TjsUpxTK2pXSPBCkTKD3NDZ8YwDcv35HIIfGOG
ldI4foColOHZmbzoZUH9FAh/rRGzIDU5ZWSYIDdfb7+HHgRIbiPJOd4k88AjnCGY
nnNcVgRSDydDIWQOhgmxH+2eLI6YvuEvAEq8vNDJqzGrO6pch1/Y8y+iA8EdsLmJ
WwMM7me7jRKq5ph4+nIXpB4AyDRBSSNgkFH59WoOJBmGGwj/DWYihMNUBhwpAgMB
AAGjggEQMIIBDDAdBgNVHQ4EFgQUffWM3cEz4Zisaty33VSCxvLMvWgwfQYDVR0j
BHYwdIAUffWM3cEz4Zisaty33VSCxvLMvWihWaRXMFUxGTAXBgoJkiaJk/IsZAEZ
Fglsb2NhbGhvc3QxCzAJBgNVBAoMAnh4MQswCQYDVQQHDAJ4eDELMAkGA1UEBhMC
UEwxETAPBgNVBAMMCE9QQ1VBU3J2ggEBMCwGA1UdEQQlMCOGFnVybjpsb2NhbGhv
c3Q6T1BDVUFTcnaCCWxvY2FsaG9zdDAMBgNVHRMBAf8EAjAAMA4GA1UdDwEB/wQE
AwIC9DAgBgNVHSUBAf8EFjAUBggrBgEFBQcDAQYIKwYBBQUHAwIwDQYJKoZIhvcN
AQELBQADggEBABt54YCUQo6cwuSAoY/UdAdB2on13vFRfH03KEK+XUqLKn0tdyio
UBgn6o1rNvQ4RKsOkD3fLuQRSP6UJ91GtaqwoJyo8ADupPOTHi2ebiyBbYYaUyD2
jEweVjBXv9ZfTDAkKfkVSbkPXbxSGNn42ba8st75Qb2AIBNwA8+O+9fg3a8BDneG
U9F/D3G5ueE1AeqVK7KuyolM1GKBuLxNB+Irw0vuodKSq37W2m97dJ+SG8BTR5DO
zhupHQItp0Pc8YuCAmK3JSQmdphKkGAymyn4ewoKQcV+aPwbfmIFdulaErGuY8Vh
sHrLsUlXwQrMTYgrZKUNdW3b2i8tclpwOPc=
-----END CERTIFICATE-----
```

nokem61, 2020-05-27:
Certificate is generate from code:
```
// OpenSSL_tests.cpp : This file contains the 'main' function. Program execution begins and ends there.
//

#include <iostream>
#include "openssl/x509v3.h"
#include "openssl/pem.h"
#include "openssl/rsa.h"
#include "openssl/rsaerr.h"
#include "openssl/applink.c"
#include <Windows.h>

#define new new(__FILE__, __LINE__)

constexpr auto CERTIFICATE_DIR = "certs\\";
constexpr auto CERTIFICATE_FILE = "OPCUASrv.der";
constexpr auto KEY_FILE = "OPCUASrv.key";

bool generateX509(const std::string& certFileName, const std::string& keyFileName, long daysValid)
{
  bool result = false;

  std::unique_ptr<BIO, void(*)(BIO *)> certFile{BIO_new_file(certFileName.data(), "wb"), BIO_free_all};
  std::unique_ptr<BIO, void(*)(BIO *)> keyFile{BIO_new_file(keyFileName.data(), "wb"), BIO_free_all};

  if (certFile && keyFile)
  {
    std::unique_ptr<RSA, void(*)(RSA *)> rsa{RSA_new(), RSA_free};
    std::unique_ptr<BIGNUM, void(*)(BIGNUM *)> bn{BN_new(), BN_free};

    BN_set_word(bn.get(), RSA_F4);
    int rsa_ok = RSA_generate_key_ex(rsa.get(), 2046, bn.get(), nullptr);

    if (rsa_ok == 1)
    {
      // --- cert generation ---
      std::unique_ptr<X509, void(*)(X509 *)> cert{X509_new(), X509_free};
      std::unique_ptr<EVP_PKEY, void(*)(EVP_PKEY *)> pkey{EVP_PKEY_new(), EVP_PKEY_free};

      X509_set_version(cert.get(), 2);

      // The RSA structure will be automatically freed when the EVP_PKEY structure is freed.
      EVP_PKEY_assign(pkey.get(), EVP_PKEY_RSA, reinterpret_cast<char*>(rsa.release()));
      ASN1_INTEGER_set(X509_get_serialNumber(cert.get()), 1); // serial number

      X509_gmtime_adj(X509_get_notBefore(cert.get()), 0); // now
      X509_gmtime_adj(X509_get_notAfter(cert.get()), daysValid * 24 * 3600); // accepts secs

      X509_set_pubkey(cert.get(), pkey.get());

      // 1 -- X509_NAME may disambig with wincrypt.h
      // 2 -- DO NO FREE the name internal pointer
      X509_name_st* name = X509_get_subject_name(cert.get());

      const unsigned char country[] = "PL";
      const unsigned char company[] = "xx";
      const unsigned char common_name[] = "OPCUASrv";
      const unsigned char locality[] = "xx";

      unsigned long size = MAX_PATH;
      char buffer[MAX_PATH];
      GetComputerNameExA(ComputerNameDnsFullyQualified, buffer, &size);
     //std::string hostname(buffer, size);
      std::string hostname("localhost");

      auto x =X509_NAME_add_entry_by_txt(name, "DC", MBSTRING_ASC, (const unsigned char*)hostname.c_str(), hostname.size(), -1, 0);
      x = X509_NAME_add_entry_by_txt(name, "O", MBSTRING_ASC, company, -1, -1, 0);
      X509_NAME_add_entry_by_txt(name, "L", MBSTRING_ASC, locality, -1, -1, 0);
      X509_NAME_add_entry_by_txt(name, "C", MBSTRING_ASC, country, -1, -1, 0);
      X509_NAME_add_entry_by_txt(name, "CN", MBSTRING_ASC, common_name, -1, -1, 0);
      
      X509_set_issuer_name(cert.get(), name);
      
      X509_EXTENSION *ex;
      X509V3_CTX ctx;
      X509V3_set_ctx_nodb(&ctx);
      X509V3_set_ctx(&ctx, cert.get(), cert.get(), NULL, NULL, 0);

      
      ex = X509V3_EXT_conf_nid(NULL, &ctx, NID_subject_key_identifier, "hash");
      X509_add_ext(cert.get(), ex, -1);

      ex = X509V3_EXT_conf_nid(NULL, &ctx, NID_authority_key_identifier, "keyid:always,issuer:always");
      X509_add_ext(cert.get(), ex, -1);

      std::string subject_alt_name;
      subject_alt_name.append("URI:urn:");
      subject_alt_name.append(hostname);
      subject_alt_name.append(":OPCUASrv");
      subject_alt_name.append(",DNS:");
      subject_alt_name.append(hostname);

      ex = X509V3_EXT_conf_nid(NULL, &ctx, NID_subject_alt_name, subject_alt_name.c_str());
      X509_add_ext(cert.get(), ex, -1);

      ex = X509V3_EXT_conf_nid(NULL, &ctx, NID_basic_constraints, "critical, CA:FALSE");
      X509_add_ext(cert.get(), ex, -1);

      ex = X509V3_EXT_conf_nid(NULL, &ctx, NID_key_usage, "critical, nonRepudiation, digitalSignature, keyEncipherment, dataEncipherment, keyCertSign");
      X509_add_ext(cert.get(), ex, -1);

      ex = X509V3_EXT_conf_nid(NULL, &ctx, NID_ext_key_usage, "critical, TLS Web Server Authentication, TLS Web Client Authentication");
      X509_add_ext(cert.get(), ex, -1);

      X509_EXTENSION_free(ex);
      X509_sign(cert.get(), pkey.get(), EVP_sha256());

      int ret = i2d_PrivateKey_bio(keyFile.get(), pkey.get());
      int ret2 = i2d_X509_bio(certFile.get(), cert.get());

      result = (ret >0) && (ret2 >0); // OpenSSL return codes    
    }
  }

  return result;
}

int main()
{
    generateX509(CERTIFICATE_FILE, KEY_FILE, 365);
}
```

nokem61, 2020-05-27:
[OPCUASrvCerts.zip](https://github.com/open62541/open62541/files/4687927/OPCUASrvCerts.zip)

nokem61, 2020-05-27:
Prosys OPC UA Browser also has a problem:
[image]

Server log
```
[2020-05-27 11:54:46.247 (UTC+0200)] trace/network      Connection 468 | New TCP connection on server socket 268
[2020-05-27 11:54:46.248 (UTC+0200)] info/network       Connection 468 | New connection over TCP from 169.254.198.216
[2020-05-27 11:54:46.269 (UTC+0200)] trace/network      Connection 468 | Activity on the socket
[2020-05-27 11:54:46.269 (UTC+0200)] trace/network      Connection 468 | Received a packet.
[2020-05-27 11:54:46.269 (UTC+0200)] info/channel       Creating a new SecureChannel
[2020-05-27 11:54:46.270 (UTC+0200)] trace/channel      Connection 468 | SecureChannel 0 | Process a HEL message
[2020-05-27 11:54:46.280 (UTC+0200)] trace/network      Connection 468 | Activity on the socket
[2020-05-27 11:54:46.281 (UTC+0200)] trace/network      Connection 468 | Received a packet.
[2020-05-27 11:54:46.281 (UTC+0200)] trace/channel      Connection 468 | SecureChannel 0 | Process an OPN message
[2020-05-27 11:54:46.281 (UTC+0200)] debug/channel      Connection 468 | SecureChannel 0 | Decrypt an OPN message
[2020-05-27 11:54:46.282 (UTC+0200)] trace/channel      Connection 468 | SecureChannel 0 | Verifying chunk signature
[2020-05-27 11:54:46.282 (UTC+0200)] debug/channel      Connection 468 | SecureChannel 1 | Generating new local nonce
[2020-05-27 11:54:46.282 (UTC+0200)] debug/channel      Connection 468 | SecureChannel 1 | Generating SecureChannel keys[2020-05-27 11:54:46.282 (UTC+0200)] trace/channel      Connection 468 | SecureChannel 1 | Generating new local keys
[2020-05-27 11:54:46.283 (UTC+0200)] trace/channel      Connection 468 | SecureChannel 1 | Generating new remote keys
[2020-05-27 11:54:46.283 (UTC+0200)] info/channel       Connection 468 | SecureChannel 1 | Opened SecureChannel
[2020-05-27 11:54:46.296 (UTC+0200)] trace/network      Connection 468 | Activity on the socket
[2020-05-27 11:54:46.296 (UTC+0200)] trace/network      Connection 468 | Received a packet.
[2020-05-27 11:54:46.296 (UTC+0200)] trace/channel      Connection 468 | SecureChannel 1 | Process a MSG
[2020-05-27 11:54:46.296 (UTC+0200)] debug/session      Connection 468 | SecureChannel 1 | Session g=00000000-0000-0000-0000-000000000000 | Processing GetEndpointsRequest with endpointUrl "opc.tcp://NWOLOWIEC.TEL-STER.local:6666"
[2020-05-27 11:54:46.297 (UTC+0200)] debug/session      Connection 468 | SecureChannel 1 | Session g=00000000-0000-0000-0000-000000000000 | Sending response for RequestId 2 of type GetEndpointsResponse
[2020-05-27 11:54:46.300 (UTC+0200)] trace/network      Connection 468 | Activity on the socket
[2020-05-27 11:54:46.301 (UTC+0200)] trace/network      Connection 468 | Received a packet.
[2020-05-27 11:54:46.301 (UTC+0200)] trace/channel      Connection 468 | SecureChannel 1 | Process a CLO
[2020-05-27 11:54:46.301 (UTC+0200)] info/channel       Connection 468 | SecureChannel 1 | CloseSecureChannel
[2020-05-27 11:54:46.305 (UTC+0200)] trace/network      Connection 468 | Activity on the socket
[2020-05-27 11:54:46.306 (UTC+0200)] info/network       Connection 468 | Closed
[2020-05-27 11:54:48.566 (UTC+0200)] trace/network      Connection 468 | New TCP connection on server socket 268
[2020-05-27 11:54:48.567 (UTC+0200)] info/network       Connection 468 | New connection over TCP from 169.254.198.216
[2020-05-27 11:54:48.568 (UTC+0200)] trace/network      Connection 468 | Activity on the socket
[2020-05-27 11:54:48.568 (UTC+0200)] trace/network      Connection 468 | Received a packet.
[2020-05-27 11:54:48.568 (UTC+0200)] info/channel       Creating a new SecureChannel
[2020-05-27 11:54:48.569 (UTC+0200)] trace/channel      Connection 468 | SecureChannel 0 | Process a HEL message
[2020-05-27 11:54:48.572 (UTC+0200)] trace/network      Connection 468 | Activity on the socket
[2020-05-27 11:54:48.572 (UTC+0200)] trace/network      Connection 468 | Received a packet.
[2020-05-27 11:54:48.573 (UTC+0200)] trace/channel      Connection 468 | SecureChannel 0 | Process an OPN message
[2020-05-27 11:54:48.573 (UTC+0200)] debug/channel      Connection 468 | SecureChannel 0 | Decrypt an OPN message
[2020-05-27 11:54:48.574 (UTC+0200)] trace/channel      Connection 468 | SecureChannel 0 | Verifying chunk signature
[2020-05-27 11:54:48.574 (UTC+0200)] debug/channel      Connection 468 | SecureChannel 2 | Generating new local nonce
[2020-05-27 11:54:48.575 (UTC+0200)] debug/channel      Connection 468 | SecureChannel 2 | Generating SecureChannel keys[2020-05-27 11:54:48.575 (UTC+0200)] trace/channel      Connection 468 | SecureChannel 2 | Generating new local keys
[2020-05-27 11:54:48.576 (UTC+0200)] trace/channel      Connection 468 | SecureChannel 2 | Generating new remote keys
[2020-05-27 11:54:48.576 (UTC+0200)] info/channel       Connection 468 | SecureChannel 2 | Opened SecureChannel
[2020-05-27 11:54:48.578 (UTC+0200)] trace/network      Connection 468 | Activity on the socket
[2020-05-27 11:54:48.578 (UTC+0200)] trace/network      Connection 468 | Received a packet.
[2020-05-27 11:54:48.579 (UTC+0200)] trace/channel      Connection 468 | SecureChannel 2 | Process a MSG
[2020-05-27 11:54:48.580 (UTC+0200)] debug/session      Connection 468 | SecureChannel 2 | Session g=00000000-0000-0000-0000-000000000000 | Processing GetEndpointsRequest with endpointUrl "opc.tcp://NWOLOWIEC.TEL-STER.local:6666"
[... cut ...]
[2020-05-27 11:54:50.858 (UTC+0200)] trace/network      Connection 464 | Activity on the socket
[2020-05-27 11:54:50.858 (UTC+0200)] trace/network      Connection 464 | Received a packet.
[2020-05-27 11:54:50.859 (UTC+0200)] info/channel       Creating a new SecureChannel
[2020-05-27 11:54:50.859 (UTC+0200)] trace/channel      Connection 464 | SecureChannel 0 | Process a HEL message
[2020-05-27 11:54:55.036 (UTC+0200)] trace/network      Connection 464 | Activity on the socket
[2020-05-27 11:54:55.036 (UTC+0200)] info/network       Connection 464 | Closed
```

nokem61, 2020-05-27:
Prosys Logs:

```
05/27/2020 11:54:28.123 INFO  [main] com.prosysopc.ua.app.uabrowser.UaBrowserLauncher  [] - Starting UaBrowser version 4.0.2-231..
05/27/2020 11:54:28.127 INFO  [main] com.prosysopc.ua.app.uabrowser.UaBrowserLauncher  [] - Conf location: C:\Users\nwolowiec.TEL-STER\.prosysopc\prosys-opc-ua-browser\conf.xml
05/27/2020 11:54:31.165 INFO  [main] com.prosysopc.fx.library.utility.managedservices.ManagedServiceManager  [] - Services for start order S2_CONF_FILE_LOAD : [h [NEW]]
05/27/2020 11:54:31.196 INFO  [main] com.prosysopc.fx.library.utility.managedservices.ManagedServiceManager  [] - Services for start order S3_AFTER_CONF_FILE_LOAD : [b [NEW], a [NEW]]
05/27/2020 11:54:31.197 INFO  [main] com.prosysopc.fx.library.utility.managedservices.ManagedServiceManager  [] - Services for start order S5_UI_START : [h [NEW]]
05/27/2020 11:54:31.199 INFO  [main] com.prosysopc.fx.library.utility.managedservices.ManagedServiceManager  [] - Starting: {NEW=[h [NEW]]}
05/27/2020 11:54:31.238 INFO  [h STARTING]                  com.prosysopc.ua.UaApplication  [] - Prosys OPC UA SDK for Java v4.2.0-955
05/27/2020 11:54:31.240 INFO  [h STARTING]                  com.prosysopc.ua.UaApplication  [] - (c) Prosys OPC Ltd. <http://www.prosysopc.com>
05/27/2020 11:54:31.246 INFO  [main] com.prosysopc.fx.library.utility.managedservices.ManagedServiceManager  [] - Starting: {NEW=[b [NEW], a [NEW]]}
05/27/2020 11:54:31.257 INFO  [pool-3-thread-1]           com.prosysopc.ua.app.uabrowser.core.a  [] - Loading ApplicationInstanceCertificates
05/27/2020 11:54:31.348 INFO  [main] com.prosysopc.fx.library.utility.managedservices.ManagedServiceManager  [] - Starting: {NEW=[h [NEW]]}
05/27/2020 11:54:31.485 INFO  [pool-3-thread-1]                              com.prosysopc.ua.f  [] - Reading application certificate from C:\Users\nwolowiec.TEL-STER\.prosysopc\prosys-opc-ua-browser\PKI\CA\private\UaBrowser@NWOLOWIEC_2048.der
05/27/2020 11:54:31.492 INFO  [pool-3-thread-1]                              com.prosysopc.ua.f  [] - Reading private key from keystore C:\Users\nwolowiec.TEL-STER\.prosysopc\prosys-opc-ua-browser\PKI\CA\private\UaBrowser@NWOLOWIEC_2048.pem
05/27/2020 11:54:31.913 INFO  [JavaFX Application Thread]                com.prosysopc.ua.app.uabrowser.h  [] - Loaded font:Font[name=Roboto, family=Roboto, style=Regular, size=12.0]
05/27/2020 11:54:31.930 INFO  [JavaFX Application Thread]                com.prosysopc.ua.app.uabrowser.h  [] - Loaded font:Font[name=Roboto Bold, family=Roboto, style=Bold, size=12.0]
05/27/2020 11:54:31.944 INFO  [JavaFX Application Thread]                com.prosysopc.ua.app.uabrowser.h  [] - Loaded font:Font[name=Roboto Italic, family=Roboto, style=Italic, size=12.0]
05/27/2020 11:54:31.959 INFO  [JavaFX Application Thread]                com.prosysopc.ua.app.uabrowser.h  [] - Loaded font:Font[name=Roboto Bold Italic, family=Roboto, style=Bold Italic, size=12.0]
05/27/2020 11:54:32.161 INFO  [pool-3-thread-1]           com.prosysopc.ua.app.uabrowser.core.a  [] - Loaded ApplicationInstanceCertificates
05/27/2020 11:54:32.421 INFO  [JavaFX Application Thread]                com.prosysopc.ua.app.uabrowser.h  [] - Showing main window
05/27/2020 11:54:46.204 INFO  [JavaFX Application Thread]   com.prosysopc.ua.app.uabrowser.singleclient.j  [] - Normal connection mode
05/27/2020 11:54:46.228 INFO  [Thread-9]       com.prosysopc.ua.stack.transport.tcp.io.d  [] - NWOLOWIEC.TEL-STER.local/169.254.198.216:6666 Connecting
05/27/2020 11:54:46.271 INFO  [Thread-9]       com.prosysopc.ua.stack.transport.tcp.io.d  [] - Connected (non-reverse), handshake completed, local=/169.254.198.216:52562, remote=NWOLOWIEC.TEL-STER.local/169.254.198.216:6666
05/27/2020 11:54:46.300 INFO  [Thread-9]       com.prosysopc.ua.stack.transport.tcp.io.b  [] - 1 Closed
05/27/2020 11:54:46.305 INFO  [TcpConnection/Read]       com.prosysopc.ua.stack.transport.tcp.io.d  [] - NWOLOWIEC.TEL-STER.local/169.254.198.216:6666 Closed (graceful)
05/27/2020 11:54:46.306 INFO  [Thread-9]       com.prosysopc.ua.stack.transport.tcp.io.d  [] - NWOLOWIEC.TEL-STER.local/169.254.198.216:6666 Closed
05/27/2020 11:54:48.522 INFO  [JavaFX Application Thread] com.prosysopc.ua.app.uabrowser.singleclient.securitysettings.c  [] - Selected SecurityMode: [http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256,SignAndEncrypt]
05/27/2020 11:54:48.565 INFO  [Thread-11]       com.prosysopc.ua.stack.transport.tcp.io.d  [] - NWOLOWIEC.TEL-STER.local/169.254.198.216:6666 Connecting
05/27/2020 11:54:48.570 INFO  [Thread-11]       com.prosysopc.ua.stack.transport.tcp.io.d  [] - Connected (non-reverse), handshake completed, local=/169.254.198.216:52563, remote=NWOLOWIEC.TEL-STER.local/169.254.198.216:6666
05/27/2020 11:54:48.586 INFO  [Thread-11]       com.prosysopc.ua.stack.transport.tcp.io.b  [] - 2 Closed
05/27/2020 11:54:48.587 INFO  [Thread-11]       com.prosysopc.ua.stack.transport.tcp.io.d  [] - NWOLOWIEC.TEL-STER.local/169.254.198.216:6666 Closed
05/27/2020 11:54:48.587 INFO  [TcpConnection/Read]       com.prosysopc.ua.stack.transport.tcp.io.d  [] - NWOLOWIEC.TEL-STER.local/169.254.198.216:6666 Closed (expected)
05/27/2020 11:54:50.854 INFO  [Thread-11]                   com.prosysopc.ua.stack.cert.e  [] - Certificate 'D057E2F46B993C006238A9EB3FC919DD61E33204' added to trusted certificates.
05/27/2020 11:54:50.856 INFO  [Thread-11]       com.prosysopc.ua.stack.transport.tcp.io.d  [] - NWOLOWIEC.TEL-STER.local/169.254.198.216:6666 Connecting
[... cut ...]
	at com.prosysopc.ua.client.x.connect(SourceFile:894) ~[application-4.0.2-231.jar:4.0.2-231]
	at com.prosysopc.ua.app.uabrowser.singleclient.j.di(SourceFile:930) ~[application-4.0.2-231.jar:4.0.2-231]
	at java.lang.Thread.run(Thread.java:834) [?:?]
05/27/2020 11:54:55.035 INFO  [TcpConnection/Read]       com.prosysopc.ua.stack.transport.tcp.io.d  [] - NWOLOWIEC.TEL-STER.local/169.254.198.216:6666 Closed (graceful)
05/27/2020 11:54:55.036 INFO  [TcpConnection/Read]       com.prosysopc.ua.stack.transport.tcp.io.d  [] - NWOLOWIEC.TEL-STER.local/169.254.198.216:6666 Closed
```

nokem61, 2020-05-27:
In both cases, something is shifted by 2 bytes. Maybe it's an accident or maybe not.

jpfr, 2020-05-28:
Strange. My impression is that UAExpert is sending a faulty message.

[image]

If you look at the Wireshark dump, the message sent by UAExpert has a TCP Segment length of 1664 byte. But the message self-identifies with a length of 1662.

And our decoding routing fails because we get a buffer that is not a multiple of the encryption block length. There are exactly two bytes missing.

The exact same server works well if I use a certificate generated with another method.
So it seems the server certificate triggers some error condition in UAExpert.

nokem61, 2020-05-29:
I find problem. 
The error is found during the certificate generation procedure. 
Key length is wrong : 
`int rsa_ok = RSA_generate_key_ex(rsa.get(), 2046, bn.get(), nullptr);`
It should be like this:
`int rsa_ok = RSA_generate_key_ex(rsa.get(), 2048, bn.get(), nullptr);`

In library is everything is fine.
Thanks.

closed 2020-05-29 by nokem61