open62541/open62541 #3635: build fail on  VS2010

LetoHH, 2020-05-27:
## 
## 
platform is ： VS2010 + WIN 10 64bit
build fail on  VS2010

the build output is below, how to fix it ?
it must be build on 2017 ?
thanks

1>------ Build started: Project: open62541-generator-types, Configuration: Debug Win32 ------
2>------ Build started: Project: open62541-generator-transport, Configuration: Debug Win32 ------
3>------ Build started: Project: open62541-generator-statuscode, Configuration: Debug Win32 ------
1>Build started 2020/5/27 11:48:26.
3>Build started 2020/5/27 11:48:26.
2>Build started 2020/5/27 11:48:26.
1>InitializeBuildStatus:
1>  Creating "Win32\Debug\open62541-generator-types\open62541-generator-types.unsuccessfulbuild" because "AlwaysCreate" was specified.
3>InitializeBuildStatus:
3>  Creating "Win32\Debug\open62541-generator-statuscode\open62541-generator-statuscode.unsuccessfulbuild" because "AlwaysCreate" was specified.
2>InitializeBuildStatus:
2>  Creating "Win32\Debug\open62541-generator-transport\open62541-generator-transport.unsuccessfulbuild" because "AlwaysCreate" was specified.
3>FinalizeBuildStatus:
3>  Deleting file "Win32\Debug\open62541-generator-statuscode\open62541-generator-statuscode.unsuccessfulbuild".
3>  Touching "Win32\Debug\open62541-generator-statuscode\open62541-generator-statuscode.lastbuildstate".
3>
3>Build succeeded.
3>
3>Time Elapsed 00:00:00.15
1>FinalizeBuildStatus:
1>  Deleting file "Win32\Debug\open62541-generator-types\open62541-generator-types.unsuccessfulbuild".
1>  Touching "Win32\Debug\open62541-generator-types\open62541-generator-types.lastbuildstate".
1>
1>Build succeeded.
1>
1>Time Elapsed 00:00:00.20
2>FinalizeBuildStatus:
2>  Deleting file "Win32\Debug\open62541-generator-transport\open62541-generator-transport.unsuccessfulbuild".
2>  Touching "Win32\Debug\open62541-generator-transport\open62541-generator-transport.lastbuildstate".
2>
2>Build succeeded.
2>
2>Time Elapsed 00:00:00.12
4>------ Build started: Project: open62541-generator-namespace, Configuration: Debug Win32 ------
4>Build started 2020/5/27 11:48:26.
4>InitializeBuildStatus:
4>  Creating "Win32\Debug\open62541-generator-namespace\open62541-generator-namespace.unsuccessfulbuild" because "AlwaysCreate" was specified.
4>FinalizeBuildStatus:
4>  Deleting file "Win32\Debug\open62541-generator-namespace\open62541-generator-namespace.unsuccessfulbuild".
4>  Touching "Win32\Debug\open62541-generator-namespace\open62541-generator-namespace.lastbuildstate".
4>
4>Build succeeded.
4>
4>Time Elapsed 00:00:00.09
5>------ Build started: Project: open62541-code-generation, Configuration: Debug Win32 ------
5>Build started 2020/5/27 11:48:27.
5>InitializeBuildStatus:
5>  Creating "Win32\Debug\open62541-code-generation\open62541-code-generation.unsuccessfulbuild" because "AlwaysCreate" was specified.
5>FinalizeBuildStatus:
5>  Deleting file "Win32\Debug\open62541-code-generation\open62541-code-generation.unsuccessfulbuild".
5>  Touching "Win32\Debug\open62541-code-generation\open62541-code-generation.lastbuildstate".
5>
5>Build succeeded.
5>
5>Time Elapsed 00:00:00.10
6>------ Build started: Project: open62541-object, Configuration: Debug Win32 ------
6>Build started 2020/5/27 11:48:27.
6>InitializeBuildStatus:
6>  Touching "open62541-object.dir\Debug\open62541-object.unsuccessfulbuild".
6>CustomBuild:
6>  All outputs are up-to-date.
[... 1461 lines cut ...]
6>..\src\server\ua_subscription_datachange.c(136): fatal error C1003: error count exceeds 100; stopping compilation
6>  Generating Code...
6>
6>Build FAILED.
6>
6>Time Elapsed 00:00:15.48
========== Build: 5 succeeded, 1 failed, 1 up-to-date, 0 skipped ==========

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

LetoHH, 2020-05-27:
OK ,I have fix this problem .
choose  project Property->c/c++->advanced->Compile as  change to 
Compile as c++ code(/TP) .

jpfr, 2020-05-27:
👍

closed 2020-05-27 by jpfr