open62541/open62541 #3641: Retrieve data from opc ua server and send them via spi

Rondine89, 2020-05-28:
Hi!
I'm updating a variable value the readCurrentValue function to set the new value. After the update I want to read the current variable value and if it's 0x001E it will be sent it via SPI, but I'm not able to manage the reading functions properly. I use the UA_Server_readValue function and then I put the result from the UA_Variant variable to a uint16 one for the SPI transmission, but it seems that copying the value from the UA_Variant to the uint16 variable doesn't work. I'm I missing something?
 

`UA_StatusCode readCurrentValue(UA_Server *server,
		const UA_NodeId *sessionId, void *sessionContext,
		const UA_NodeId *nodeId, void *nodeContext,
		UA_Boolean sourceTimeStamp, const UA_NumericRange *range,
		UA_DataValue *dataValue)  
{
	UA_UInt16 new_val = UpVal;

	if(new_val <= 100)
	{
		if(red)
		{
			new_val = UpVal + 10;
			UpVal = new_val;
			XMC_GPIO_SetOutputHigh(LedR);  /*red led off*/
			red = false;
		}
		else
		{
			XMC_GPIO_SetOutputLow(LedR);  /*red led on*/
			red = true;
		}
	}
	else
	{
		UpVal = 0;
		new_val = UpVal;
	}

	UA_Variant_setScalarCopy(&dataValue->value, &new_val,
				&UA_TYPES[UA_TYPES_UINT16]);
	dataValue->hasValue = true;

	return UA_STATUSCODE_GOOD;
}

void addVariable(UA_Server *server) // aggiunge una variabile
{
	/* Define the attribute of the myInteger variable node */
	UA_VariableAttributes attr = UA_VariableAttributes_default;
	attr.description = UA_LOCALIZEDTEXT("en-US","the answer");
	attr.displayName = UA_LOCALIZEDTEXT("en-US","the answer");
	attr.dataType = UA_TYPES[UA_TYPES_UINT16].typeId;
	attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;

	/* Add the variable node to the information model */
	UA_NodeId currentNodeId = UA_NODEID_STRING(1, "the.answer");
	UA_QualifiedName currentName = UA_QUALIFIEDNAME(1, "the answer");
	UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
	UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);

	UA_NodeId variableTypeNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE);

	UA_DataSource timeDataSource;
	timeDataSource.read = readCurrentValue;
	UA_Server_addDataSourceVariableNode(server, currentNodeId, parentNodeId,
			parentReferenceNodeId, currentName,
			variableTypeNodeId, attr,
			timeDataSource, NULL, NULL);

/*** read the variable value and send it via SPI ***/
	UA_Variant value;
	UA_Variant_init(&value);
	UA_Server_readValue(server, currentNodeId, &value);
	UA_UInt16 *intValue = (UA_UInt16 *)value.data;
	uint16_t *frame = (uint16_t *)intValue;
	UA_Variant_deleteMembers(&value);

	if(frame[0] == 0x001E) /*1E = 30*/
	{
		XMC_GPIO_SetOutputLow(LedB);  /*blue red on*/
		SendSPI(frame[0]);
	}
	else
	{
		XMC_GPIO_SetOutputHigh(LedB);  /*blue red off*/
		SendSPI(0);
	}
/************************************************/

}`

jpfr, 2020-05-28:
The `readCurrentValue` callback you are using is for the read-callback.
So it is used every time a user wants to read over the network.

Don't you want to hook into the write-callback to receive changes?

Rondine89, 2020-05-28:
Ok, so i misunderstood the function of the callback. 
What's the best way to use the write-callback to get the value of a variable node?

jpfr, 2020-05-28:
See the examples here: https://github.com/open62541/open62541/blob/master/examples/tutorial_server_datasource.c

Or just read with a fixed time interval:
https://github.com/open62541/open62541/blob/master/include/open62541/server.h#L256

Or monitor the variable to get notified when it changes:
https://github.com/open62541/open62541/blob/master/examples/tutorial_server_monitoreditems.c

Rondine89, 2020-05-28:
Ok. Thank you very much!

embbo, 2020-08-13:
@jpfr , I have question .. I same like concept but I am using CAN protocol. My question is : Is this possible i can also send user data (for example can adreess) .... e,g atrr.userdata = ("CAN Adress");

NoelGraf, 2021-04-30:
@Rondine89 is something still open? can the issue be closed?
@embbo Please open a new issue if this is required.

closed 2021-05-06 by NoelGraf