open62541/open62541 #3645: Issues building open62541 for FreeRTOS+LwIP on STM32

Rixem, 2020-05-29:
## Description
Hello everyone. I am trying to port this beautiful library onto my STM32 board. On this board, I have a working FreeRTOS and LwIP package. Works like a charm. Nom, i tried to build myself the library but get some difficulties to get it working.

## Background Information / Reproduction Steps

What I've done? I've followed precisely every steps even the #2511 thread but still have issues.
Step 1 : Fetching the repo.
Step 2 : Configure Cmake building --> 
[image]

Step 3 : Open the solution in visual studio, generate the "ALL_BUILD" package, and wait for the build. Nice ! I have the open62541.c/.h files.

Step 4 : Integrate these files in my STM32CubeIDE project.
[image]

Step 5 : configure lwipopts.h and FreeRTOSCOnfig.h. I have double check these files, and the are ok.

Step 6 : (Here is where is messed up something i think) I configured STM32CubeIDE to compile this project. 
[image]

Build command is like that : `-mcpu=cortex-m4 -std=gnu99 -g3 -DUSE_HAL_DRIVER -DDEBUG -DSTM32F429xx -c -I../Middlewares/Third_Party/LwIP/src/include/netif/ppp -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/Third_Party/LwIP/src/include/compat/stdc -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 -I../Middlewares/Third_Party/LwIP/system/arch -I../Middlewares/Third_Party/LwIP/src/include -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F -I../LWIP/App -I../Middlewares/Third_Party/LwIP/src/include/compat/posix -I../Core/Inc -I../Middlewares/Third_Party/LwIP/src/include/lwip -I../Middlewares/Third_Party/LwIP/src/include/compat/posix/arpa -I../Middlewares/Third_Party/LwIP/system -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/LwIP/src/include/lwip/priv -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/Third_Party/LwIP/src/include/lwip/prot -I../Middlewares/Third_Party/LwIP/src/include/lwip/apps -I../Middlewares/Third_Party/LwIP/src/include/compat/posix/sys -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Include -I../LWIP/Target -I../Middlewares/Third_Party/LwIP/src/include/compat/posix/net -I../Middlewares/Third_Party/LwIP/src/include/netif -O0 -ffunction-sections -fdata-sections -Wall -DUA_ARCHITECTURE_FREERTOSLWIP -fstack-usage --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb`

When i try to compile this project, unfortunately i get multiple errors.
[image]
[image]
[image]

Used CMake options:

Here is CMake generation Debug Output: 
`Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.16299.
fatal: not a git repository (or any of the parent directories): .git

CMake Warning at tools/cmake/SetGitBasedVersion.cmake:83 (message):
  Failed to determine OPEN62541_VERSION from repository tags.  Using default
  version "".
Call Stack (most recent call first):
  CMakeLists.txt:40 (set_open62541_version)

open62541 Version: 0.0.0-unknown
Architectures included in amalgamation: freertosLWIP
The selected architecture is: freertosLWIP
   Entering             D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1/arch
   Entering             D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1/arch/posix
   Returning to         D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1/arch
   Entering             D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1/arch/win32
   Returning to         D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1/arch
   Entering             D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1/arch/freertosLWIP
   Returning to         D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1/arch
   Entering             D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1/arch/vxworks
   Returning to         D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1/arch
   Entering             D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1/arch/eCos
   Returning to         D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1/arch
   Entering             D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1/arch/wec7
   Returning to         D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1/arch
   Returning to         D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1
   Entering             D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1/doc
Could NOT find Sphinx (missing: SPHINX_EXECUTABLE) 
   Returning to         D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1
   Entering             D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1/tools/packaging
   Returning to         D:/PROJETS_BUREAU_ETUDES/PROJETS_EMBARQUES/open62541-master/open62541-1.1-rc1
Configuring done`

## Checklist

Please provide the following information:

 - [ 1.1-rc1] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ FreeRTOS] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Can you help me ? What am I doing wrong?

Thank you by advance for your help.

Rixem, 2020-05-29:
For those who have the same problem. I finally get it working.

I'm using STM32CubeIDE. I was entering the symbol -DUA_ARCHITECTURE_FREERTOS_LWIP in the wrong place.

Make sure you have this symbol here : 

[image]

Thanks to #2511 

Bye.

closed 2020-05-29 by Rixem

acer1204, 2021-01-30:
Hi, Rixem
When i add the DUA_ARCHITECTURE_FREERTOS_LWIP in true path and compile it,
it show one error "dereferencing pointer to incomplete type 'struct sockaddr_in6"
Have you ever see this problem?