open62541/open62541 #3646: Tag "v.1.0.1" in branch "1.0" confuses tools/prepare_packaging.py

jrcoding, 2020-05-29:
## Description

The tag `v.1.0.1` in branch `1.0` confuses the python script `tools/prepare_packaging.py`

## Background Information / Reproduction Steps

Check out branch 1.0

```
git checkout 1.0
```
and then calling ` /usr/bin/python ./tools/prepare_packaging.py` will give an error like
```
Traceback (most recent call last):
  File "./tools/prepare_packaging.py", line 27, in <module>
    version_major = m.group(1) if m.group(1) is not None else "0"
AttributeError: 'NoneType' object has no attribute 'group'
```
This is because ` ./tools/prepare_packaging.py` calls  
`git describe --tags --dirty --match "v*"` and the result in this case will be `v.1.0.1-1-gd5df1c62` which is not compatible with the used tag numbers so far. I guess that the tag `v.1.0.1` is a faulty leftover, right?

Two possible solutions:
1. You can delete the tag because it may be a faulty tag?!
2. I can provide a PR making the script  `prepare_packaging.py` more robust concerning this issue

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): **1.0**
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2020-06-04:
I just removed `v.1.0.1` tag. Please close the issue if it is resolved now.

jrcoding, 2020-06-04:
@Pro: Thanks!

closed 2020-06-04 by jrcoding

Johnsik, 2021-05-27:
There is a same problem with tag v.1.2.2