open62541/open62541 #3650: Example Adding method to object type and its objects

dotChris90, 2020-05-31:
Hello team, 

just a short question corresponding to https://open62541.org/doc/current/tutorial_server_object.html#object-types-type-hierarchies-and-instantiation . 

In this sample it is good to see how to add properties and variables to object type (so "a class" in OOP). 

What I was not seeing was an example how to add method to such an object type. I saw how to add methods on documentation but not to a "class" and create an object from there. 

I tried : 
```C
UA_Server_addMethodNode(server, 
                            UA_NODEID_NUMERIC(1,62541),
                            myObjectTypeID,
                            UA_NODEID_NUMERIC(0, UA_NS0ID_HASORDEREDCOMPONENT),
                            UA_QUALIFIEDNAME(1, "hello world"),
                            helloAttr, &helloWorldMethodCallback,
                            1, &inputArgument, 1, &outputArgument, NULL, node);
```

But with this i add a method to the type node. all object nodes which created based on this type have the variables and properties but no method according to opcua-client (the python tool). 

Is there a proper way to create "class / type nodes" with methods and their objects also have this methods just like in OOP? Maybe i have overseen it but until now did not find an example. 

thanks and have a good week. :)

- non-critical issue 
- documentation missing (maybe)

jpfr, 2020-05-31:
Did you make the method a mandatory child?

dotChris90, 2020-05-31:
....
embarrassing - such little mistake ..... -.- 

Yes you were totally right! i was missing the addReference function. 

the code which works like charm is : 

```C
UA_NodeId helloWorldnodeId;
UA_Server_addMethodNode(server, 
                            UA_NODEID_NUMERIC(1,62541),
                            myObjectTypeId,
                            UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                            UA_QUALIFIEDNAME(1, "hello world"),
                            helloAttr, &helloWorldMethodCallback,
                            1, &inputArgument, 1, &outputArgument, NULL, &helloWorldnodeId);

UA_Server_addReference(server, helloWorldnodeId,
                           UA_NODEID_NUMERIC(0, UA_NS0ID_HASMODELLINGRULE),
                           UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_MODELLINGRULE_MANDATORY), 
                           true);
    
```

Thanks again and wish you a good week. i will close the issue. ;)

dotChris90, 2020-05-31:
closed because issue clearified thanks to @jpfr

closed 2020-05-31 by dotChris90