open62541/open62541 #3658: Problem building on STM32CubeIDE for FreeRTOS+LwIP

Rixem, 2020-06-02:
## Description

Hi,

I try to build this stack on a STM32F429 with FreeRTOS + LwIP. Unfortunately, I have some problems and errors when build finish. I think it's a CMake problem. Some functions are not defined, like UA_Server_New.

Can you help me ?

Thanks.

[image]

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2020-06-04:
The main question here is, which CMake options did you use? Please fill out the complete Issue template!

Did you set the correct architecture with `-DUA_ARCHITECTURE=freertosLWIP`?

Example:
https://github.com/Pro/open62541-esp32/blob/master/components/open62541/CMakeLists.txt#L27

Also, when using the amalgamated version, you need to add the following compiler definitions:
`-DUA_ENABLE_AMALGAMATION -DUA_ARCHITECTURE_FREERTOSLWIP -DUA_ARCHITECTURE_FREERTOSLWIP_POSIX_CLOCK`

Rixem, 2020-06-04:
Hey,

Yup, sorry i forgot to mention that it is working like a charm now ;)

The problem was clearly simple : I had compilation problems after CMake, building with Visual Studio, because of missing includes.  I had to point CMake property "UA_ARCH_EXTRA_INCLUDES" to my FreeRTOS folder.

I'll update #2511 to tell it works on my STM32F429.

Thanks ! :)

closed 2020-06-04 by Rixem