open62541/open62541 #3661: Unable to invoke method call with array of ExtensionObjects (e.g. FieldTargetDataType)

jmachauer-codewerk, 2020-06-02:
## Description
I was trying to do some work on the subscriber side of the pubsub information model but i wasn't able to invoke a few method calls, since the data i received wasn't usable.

I hope everything is clear and thank you for your help in advance.

## Background Information / Reproduction Steps
To replicate this i tried to boil it down to the key differences. This example compares 3 method calls, a call with a scalar FieldTargetDataType, a call with an array of 2 FieldTargetDataType and as a control, that im not messing up array copies or something, a call with an array of 2 NodeIds. For this example im just interested in receiving the correct data, so i just extract the nodeids from all 3 calls and print them.
Client:
```
#include <open62541/client_config_default.h>
#include <open62541/client_highlevel.h>
#include <open62541/client_subscriptions.h>
#include <open62541/plugin/log_stdout.h>

#include <stdlib.h>

UA_StatusCode
callMethod(UA_Client *client, UA_UInt32 method_identifier, UA_Variant *input) {
    UA_StatusCode status_code = UA_STATUSCODE_GOOD;
    size_t output_size;
    UA_Variant *output;
    status_code = UA_Client_call(client, UA_NODEID_NUMERIC(0, 64005),
                                 UA_NODEID_NUMERIC(0, method_identifier), 1, input,
                                 &output_size, &output);
    if(status_code == UA_STATUSCODE_GOOD) {
        printf("Method call was successful, and %lu returned values available.\n",
               (unsigned long)output_size);
        UA_Array_delete(output, output_size, &UA_TYPES[UA_TYPES_VARIANT]);
    } else {
        printf("Method call failed: %s\n", UA_StatusCode_name(status_code));
    }
    return status_code;
}

int
main(int argc, char *argv[]) {
    UA_Client *client = UA_Client_new();
    UA_ClientConfig_setDefault(UA_Client_getConfig(client));

    UA_StatusCode status_code = UA_Client_connect(client, "opc.tcp://CWL-PF11Q7W7:4840");
    if(status_code != UA_STATUSCODE_GOOD) {
        printf("Conenction to server failed.\n");
        UA_Client_delete(client);
        return EXIT_FAILURE;
    }
    UA_FieldTargetDataType field_targets[2];
    UA_FieldTargetDataType_init(&field_targets[0]);
    field_targets[0].attributeId = 13;
    field_targets[0].targetNodeId = UA_NODEID_NUMERIC(0, 1234);
    UA_FieldTargetDataType_init(&field_targets[1]);
    field_targets[1].attributeId = 13;
    field_targets[1].targetNodeId = UA_NODEID_NUMERIC(0, 5678);

    UA_Variant input;  // no UA_Variant_init() needed
    status_code = UA_Variant_setScalarCopy(&input, field_targets,
                                           &UA_TYPES[UA_TYPES_FIELDTARGETDATATYPE]);
    if(status_code != UA_STATUSCODE_GOOD)
        printf("Failure ScalarCopy: %s\n", UA_StatusCode_name(status_code));
    callMethod(client, 64099, &input);

    status_code = UA_Variant_setArrayCopy(&input, field_targets, 2,
                                          &UA_TYPES[UA_TYPES_FIELDTARGETDATATYPE]);
    if(status_code != UA_STATUSCODE_GOOD)
        printf("Failure at ArrayCopy1: %s\n", UA_StatusCode_name(status_code));
    callMethod(client, 64100, &input);

    UA_NodeId nodeids[2];
    nodeids[0] = field_targets[0].targetNodeId;
    nodeids[1] = field_targets[1].targetNodeId;
    status_code = UA_Variant_setArrayCopy(&input, nodeids, 2, &UA_TYPES[UA_TYPES_NODEID]);
[... 200 lines cut ...]
```

Used CMake options:

Not really used here but nevertheless these are what i enable via CMake
```
UA_ENABLE_PUBSUB ON
UA_ENABLE_PUBSUB_ETH_UADP ON
UA_ENABLE_PUBSUB_DELTAFRAMES ON
UA_ENABLE_PUBSUB_INFORMATIONMODEL ON
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS ON
UA_NAMESPACE_ZERO "FULL"
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): current master branch
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu-20.04 (WSL)
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-06-03:
> UA_FieldTargetDataType *field_targets = (UA_FieldTargetDataType *)input->data;

That's the culprit.
The protocol requires us to send an array of ExtensionObject if the datatype is not "builtin".

On the receive side we could check if all ExtensionObjects carry the same content and then "unwrap". But currently we just forward the ExtensionObject.

Can you break with a debugger in the mentioned line and check the type with `input->type`?

Probably we have some work here.
As also the type-checking of the arguments should be aware of this ExtensionObject wrapping...

jmachauer-codewerk, 2020-06-08:
Thank you very much that was my oversight, i misstook ExtensionObject as Structure and thought it lost its specifics somehow. Handling the ExtensionObjects properly worked.

This is also the same reason why [addPublishedDataItemsAction](https://github.com/open62541/open62541/blob/a524accc338c9611a5ec60bf595c1a587243b457/src/pubsub/ua_pubsub_ns0.c#L615) didn't work for me then. Does this require a fix or is it planned to include the unwrapping in the near future?

jpfr, 2020-06-09:
I think automatic unwrapping is convenient and also more symmetric with our encoding API.
That should happen on master after the v1.1 release.

jpfr, 2020-06-09:
But we have to be careful on making the change and communicating it.
It could break some existing code.

jmachauer-codewerk, 2020-06-10:
Ok, thank you. This issue is resolved then

closed 2020-06-10 by jmachauer-codewerk