open62541/open62541 #3667: Request: Add package for Ubuntu 20.04 to PPA

smorokin, 2020-06-04:
## Description

Please add a package for Ubuntu 20.04 to the PPA. 
Thank you very much. 

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 20.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2020-06-04:
Thanks. I just enabled PPA for Focal and Groovy for the stable and daily build.
https://code.launchpad.net/~open62541-team/+recipe/open62541-stable

The new packages should be there in a few minutes.

closed 2020-06-04 by Pro

smorokin, 2020-06-04:
Great! Thank you very much for your work!