open62541/open62541 #3668: instantiate an object of FileType

Feras992, 2020-06-04:
## Description
Hello,

i write this new issue, cuz the old one is closed and i couldn't reopen it.
the main issue is: [2862](https://github.com/open62541/open62541/issues/2862)

i am using a nucleo board with STM32F767 which runs FreeRTOS, lwIP and OPCUA.
for the OPCUA i use the minimal server and i want to implement the FileType to be able to upload a file to the uC over the Server.

i added the FileType to the minimal server by copying its part from the full server(the functions which added the FileType to the full namespace).

i can see the FileType in my server

[image]

i started to implement the  callback functions for the FileType functions(open, read, write, set position and get position) and i tried them seperatly using the FileType in the image above without instantiate an object of the FileType and it worked. i was be able to send a file over OPCUA to the uC and resent the received file over UART to the pc.

after that i tried to instantiate an object from the FileType using this code:
```
  UA_ObjectAttributes fileAttr = UA_ObjectAttributes_default;
  fileAttr.displayName = UA_LOCALIZEDTEXT("en-US", "Firmware");
  UA_Server_addObjectNode(server,
                          UA_NODEID_NUMERIC(1, 11575),
                          UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                          UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                          UA_QUALIFIEDNAME(1, "Firmware"),
                          UA_NODEID_NUMERIC(0, UA_NS0ID_FILETYPE),
                          fileAttr,
                          NULL,
                          NULL);
```

and tried to send a file from pc to the server
[image]

but after i choose the file to be sent i get this error

[image]

i tried to used breakpoints in the callback functions to see what is the problem, but the code don't go there using the instantiated object. using the functions directly(write or read or....) works fine and in debug i can go through the breakpoints.

did i instantiate the object worngly? 

many thnx

jpfr, 2020-06-05:
You have to attach a *callback pointer* to the method.
Otherwise, the SDK does not know what to do when the method is called.

We could have a reusable internal *callback method* for file handling.
But we don't have it at the moment.
So you need to implement that (in C) for your specific file transfer.

Feras992, 2020-06-05:
@jpfr first thnx for answering me.

i did that allready and as i said, if i tired the methods Write or Read directky in the FileType in the BaseObjectType(the first photo), they work fine and i was able to send a file to the uC and resent it to the pc over UART or to save a string from uC to PC direktly.

here is an example of the code for the Write Method, i took it from the full server, cuz i was not be able to add the FileType tools/schema/datatypes_minimal.txt 

```
/* Write - ns=0;i=11588 */

static UA_StatusCode function_namespace0_generated_2309_begin(UA_Server *server, UA_UInt16* ns) {
#ifdef UA_ENABLE_METHODCALLS
UA_StatusCode retVal = UA_STATUSCODE_GOOD;
UA_MethodAttributes attr = UA_MethodAttributes_default;
attr.executable = true;
attr.userExecutable = true;
attr.displayName = UA_LOCALIZEDTEXT("", "Write");
retVal |= UA_Server_addNode_begin(server, UA_NODECLASS_METHOD,
UA_NODEID_NUMERIC(0, 11588),
UA_NODEID_NUMERIC(0, 11575),
UA_NODEID_NUMERIC(0, 47),
UA_QUALIFIEDNAME(0, "Write"),
 UA_NODEID_NULL,
(const UA_NodeAttributes*)&attr, &UA_TYPES[UA_TYPES_METHODATTRIBUTES],NULL, NULL);
return retVal;
#else
return UA_STATUSCODE_GOOD;
#endif /* UA_ENABLE_METHODCALLS */
}

static UA_StatusCode function_namespace0_generated_2309_finish(UA_Server *server, UA_UInt16* ns) {
#ifdef UA_ENABLE_METHODCALLS
return UA_Server_addMethodNode_finish(server,
UA_NODEID_NUMERIC(0, 11588)
, &writeCallback, 0, NULL, 0, NULL);
#else
return UA_STATUSCODE_GOOD;
#endif /* UA_ENABLE_METHODCALLS */
}
```
as u see, in the finish function i attached the writeCallback function

and here is the callback function:
```
UA_StatusCode writeCallback(UA_Server * server,
                            const UA_NodeId * sessionId, void * sessionHandle,
                            const UA_NodeId * methodId, void * methodContext,
                            const UA_NodeId * objectId, void * objectContext,
                            size_t inputSize, const UA_Variant * input,
                            size_t outputSize, UA_Variant * output)
    {
    /* as input we have the UInt32 fileHandle and the data */
    UA_UInt32 * fileHandle = (UA_UInt32*)input[0].data;
    if(*fileHandle == fHandle)
        {
        UA_ByteString * data = (UA_ByteString*)input[1].data;

        UARTprintf((UA_Byte*)data->data);

        return UA_STATUSCODE_GOOD;
        }
    }
```

Feras992, 2020-06-05:
@jpfr i did the same for all other methods(open, close, read, set position, get position).
i didn't use the attributes(Size, MimeType, OpenCount, Writable, UserWritable) .. i added them to the server using this code .. for example for the Size:

```
/* Size - ns=0;i=11576 */

static UA_StatusCode function_namespace0_generated_2299_begin(UA_Server *server, UA_UInt16* ns) {
UA_StatusCode retVal = UA_STATUSCODE_GOOD;
UA_VariableAttributes attr = UA_VariableAttributes_default;
attr.minimumSamplingInterval = 0.000000;
attr.userAccessLevel = 1;
attr.accessLevel = 1;
/* Value rank inherited */
attr.valueRank = -1;
attr.dataType = UA_NODEID_NUMERIC(0, 9);
attr.displayName = UA_LOCALIZEDTEXT("", "Size");
retVal |= UA_Server_addNode_begin(server, UA_NODECLASS_VARIABLE,
UA_NODEID_NUMERIC(0, 11576),
UA_NODEID_NUMERIC(0, 11575),
UA_NODEID_NUMERIC(0, 46),
UA_QUALIFIEDNAME(0, "Size"),
UA_NODEID_NUMERIC(0, 68),
(const UA_NodeAttributes*)&attr, &UA_TYPES[UA_TYPES_VARIABLEATTRIBUTES],NULL, NULL);
return retVal;
}

static UA_StatusCode function_namespace0_generated_2299_finish(UA_Server *server, UA_UInt16* ns) {
return UA_Server_addNode_finish(server,
UA_NODEID_NUMERIC(0, 11576)
);
}
```

but i didn't attached any thing with them, it was not so clear to me, how to do it, since the object dosent't show any of them.

can u plz give an example how to work with these attributes? for the Size for example?

did i intantiate the object correctly?

thnx

Feras992, 2020-06-09:
@jpfr any help? i couldn't make any progress.
when i finished, i will share the code for the others but i still have the same problem 

[image]

and as i said, i attachen callback functions for all the methodes and i used breakpoints in the callback functions and when i call the method directly i can reach the breakpoints but when i use the method from the instatiated object, i can reach the breakpoint in the open function or in the other functions

jpfr, 2020-06-09:
This error code is generated in just two places in the SDK.

What is the reference between the object and the method containing it?
This should be a `hasComponent` reference.

Feras992, 2020-06-09:
@jpfr where should i do that? do you mean when i instatiate an object? if yes i tried it and still get the same error

jpfr, 2020-06-09:
The method needs to be "contained" in the object.
Do you see a `hasComponent` reference from the relevant object to the method?

The references will be generated automatically if you instantiate from an ObjectType and the methods are declared mandatory children.

The error code is generated in only two places in /src/server/ua_services_method.c.
It should be easy to place a breakpoint for a debugger and see what is going on.

Feras992, 2020-06-10:
@jpfr the methodes are contained in the main object

[image]

the main FileType ObjectType has the method as hasComponant. Or did i understand you wrongly?

some questions:
1-how i can add the FileType using the MakeList when i build the open62541 using CMake? maybe i forget to add something what i relevant to the institiated object. I add the FileType **Manually** .. i copied the functions function_namespace0_generated_XXXX_begin and function_namespace0_generated_XXXX_finish from the full server to the reduced one.

2-how can i check that the methods are really mandatory?

Feras992, 2020-06-16:
@jpfr thnx, i solved the problem and it works now.
i made all the methodes in my code mandatory and it worked with the reduced server but not with the minimal one, the problem was that the minimal server doesn't have the ObjectType ModellingRuleType and Mandatory. i added them manually(i copied the functions function_namespace0_generated_143_begin, function_namespace0_generated_143_finish, function_namespace0_generated_144_begin, function_namespace0_generated_144_finish in my code from the full server).
is there any way to build the server with this ObjectTypes(FileTypem ModellingRuleType and Mandatory)? it is a lot of work to add them manually every time when i build the stack(for example when a new release version is out)

teenageheat, 2020-06-30:
@Feras992 I'd love to see your example code, file handling is up next for me!
Could you share it somehow? Maybe in a small Github Repo?

closed 2021-05-06 by NoelGraf

dns13, 2023-05-17:
I've got it working to use as firmware update for my arm device with minimal generated NS0. Here is the relevant code of how I've done it. Code is C++ but you should get the point. I've only implemented the required methods `open`, `write` and `close` and made them mandatory.

```c++
#define UA_LOCALIZEDTEXT_STATIC(LOCALE, TEXT) UA_LocalizedText{ \
			.locale = UA_STRING_STATIC(LOCALE), \
			.text = UA_STRING_STATIC(TEXT) \
		}

#define UA_QUALIFIEDNAME_STATIC(NSIDX, CHARS) UA_QualifiedName{ \
			.namespaceIndex = NSIDX, \
			.name = UA_STRING_STATIC(CHARS) \
		}

auto writeCallback(UA_Server * /*server*/, const UA_NodeId * /*sessionId*/, void * /*sessionHandle*/,
							const UA_NodeId * /*methodId*/, void * /*methodContext*/, const UA_NodeId * /*objectId*/,
							void * /*objectContext*/, size_t /*inputSize*/, const UA_Variant *input,
							size_t /*outputSize*/, UA_Variant * /*output*/) -> {
	/* as input we have the UInt32 fileHandle and the data */
	const auto *fileHandle = reinterpret_cast<UA_UInt32 *>(input[0].data);
	const auto *data = reinterpret_cast<UA_ByteString *>(input[1].data);
	const auto data_span = std::span<const uint8_t>(data->data, data->length);

	if (*fileHandle == 0) {
		return DFU::doMasterFirmwareUpdate(data_span) ? UA_STATUSCODE_GOOD
															   : UA_STATUSCODE_BADINVALIDARGUMENT;
	} else if (*fileHandle == 1) {
		const auto update_succesful = DFU::doNRFFirmwareUpdate(data_span);

		if (update_succesful) {
			fw_update_latch = true;

			while (fw_update_latch == true) {
				vTaskDelay(pdMS_TO_TICKS(100));
			}

			vTaskDelay(pdMS_TO_TICKS(2000));
		}

		return update_succesful ? UA_STATUSCODE_GOOD : UA_STATUSCODE_BADINVALIDARGUMENT;
	}

	return UA_STATUSCODE_BADNOTFOUND;
}

auto openCallback(UA_Server * /*server*/, const UA_NodeId * /*sessionId*/, void * /*sessionHandle*/,
						   const UA_NodeId * /*methodId*/, void * /*method_context*/, const UA_NodeId *node_id,
						   void * /*objectContext*/, size_t /*inputSize*/, const UA_Variant * /*input*/,
						   size_t /*outputSize*/, UA_Variant *output) -> UA_StatusCode {
	UA_UInt32 data{};

	if (node_id->namespaceIndex == 1 && node_id->identifier.numeric == opcua::nodeids::debug::fw) {
		data = 0;
	} else if (node_id->namespaceIndex == 1 && node_id->identifier.numeric == opcua::nodeids::debug::fw_nrf) {
		data = 1;
	} else {
		return UA_STATUSCODE_BADNOTFOUND;
	}

	UA_Variant_setScalarCopy(output, &data, getUADataType<decltype(data)>());
	return UA_STATUSCODE_GOOD;
}

auto closeCallback(UA_Server * /*server*/, const UA_NodeId * /*sessionId*/, void * /*sessionHandle*/,
							const UA_NodeId * /*methodId*/, void * /*method_context*/, const UA_NodeId *node_id,
							void * /*objectContext*/, size_t /*inputSize*/, const UA_Variant * /*input*/,
							size_t /*outputSize*/, UA_Variant * /*output*/) -> UA_StatusCode {
	if (node_id->namespaceIndex == 1 && node_id->identifier.numeric == opcua::nodeids::debug::fw) {
		return UA_STATUSCODE_GOOD;
	} else if (node_id->namespaceIndex == 1 && node_id->identifier.numeric == opcua::nodeids::debug::fw_nrf) {
		fw_update_latch = true;
[... 250 lines cut ...]
		parent_node_id_temp = *parent_node_id;
	}

	UA_QualifiedName new_qualified_name = UA_QUALIFIEDNAME_ALLOC(1, qualified_name.c_str());

	auto retval = UA_Server_addObjectNode(server, node_id, parent_node_id_temp,
										  UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES), new_qualified_name,
										  UA_NODEID_NUMERIC(0, UA_NS0ID_FILETYPE), o_attr, nullptr, &node_id);

	if (retval != UA_STATUSCODE_GOOD) {
		UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "%.*s: %s", new_qualified_name.name.length,
					 new_qualified_name.name.data, UA_StatusCode_name(retval));
	}

	return node_id;
}

auto main() -> int {
    ...
   generateCustomNS0(server);
   addFirmwareUpdateNode(server, nullptr, "Firmware", "", "firmware", UA_NODEID_NUMERIC(1, opcua::nodeids::debug::fw));
   addFirmwareUpdateNode(server, nullptr, "Firmware NRF", "", "firmware_nrf", UA_NODEID_NUMERIC(1, opcua::nodeids::debug::fw_nrf));
   ....
}
```