open62541/open62541 #3672: UA_Server_getStatistics

abdo4694, 2020-06-04:
Hello,

I am trying to check if at some moment I lose connection with the client and I am using the following function UA_Server_getStatistics. However, I get the following linkage error: undefined reference to UA_Server_getStatistics. Is there any other way to check if the client is connected or not. If its not connected , I need to handle some stuff in the server.

Thank you in advance

jpfr, 2020-06-06:
Please fill out the template you see when opening a new github issue.
Notably, which version of the library are you using?

abdo4694, 2020-06-09:
## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 18.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

jpfr, 2020-07-01:
Right.
The is a UA_EXPORT missing in the header.

abdo4694, 2020-07-02:
How can I update the library if I have installed it using:

  sudo add-apt-repository ppa:open62541-team/ppa
  sudo apt-get update
  sudo apt-get install libopen62541-1-dev

closed 2020-07-08 by jpfr

abdo4694, 2020-07-09:
Hello again,

After executing the following commands again 

sudo add-apt-repository ppa:open62541-team/ppa
sudo apt-get update
sudo apt-get install libopen62541-1-dev

I do not get a building error but I get a runtime error: symbol lookup error: undefined symbol: UA_Server_getStatistics

What could be the problem? 

Thank you in advance

commit aa4cc1a3 (2020-07-02): fix(server): Export UA_Server_getStatistics (fix #3672)