open62541/open62541 #3673: Warnings in running local discovery server

ilconte96, 2020-06-04:
## Description

When I run server_lds.c in discovery examples  I get the following warnings:
- warn/server Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
- warn/userland AcceptAll Certificate Verification. Any remote certificate will be accepted.
How can I solve these?

Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON -DUA_ENABLE_AMALGAMATION=ON ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Ubuntu 18.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2020-06-06:
As this is just an example, it's not intended to provide a secure configuration (#3243)

- it looks like there is a Username/Password configured #3243
- The example does not limit connections to trusted sources: https://github.com/open62541/open62541/blob/ba5cf14f9c39a48fc2cc0fd9a129399b240a3b26/plugins/ua_config_default.c#L708-L712

ilconte96, 2020-06-08:
Thank you for the help!

closed 2020-06-08 by ilconte96