open62541/open62541 #3682: VLAN tag pub/sub

sdick97, 2020-06-08:
## Description
Hello, 

I tried to publish VLAN tagged traffic with the tutorial_pubsub_publish example like this:
./tutorial_pubsub_publish opc.eth://0a-00-27-00-00-08:3.3 eth0

On another device (directly connected) i started the subscriber with:
./tutorial_pubsub_subscribe opc.eth://0a-00-27-00-00-08:3.3 eth0

The subscriber receives the messages. However, when I use Wireshark to analyze the traffic I can't see the VLAN tag starting with 0x8100 ...

Maybe I missed something essential. Is this the expected behaviour at the moment? 

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake 
-DUA_BUILD_EXAMPLES:BOOL="1" 
-DUA_NAMESPACE_ZERO= REDUCED 
-DUA_ENABLE_PUBSUB:BOOL="1"
-DUA_ENABLE_PUBSUB_ETH_UADP:BOOL="1"
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

johndzilva, 2020-06-09:
If you are using the tutorial_pubsub_publish.c then you should create a VLAN interface first in your system. After that you need to run your application like this

`./tutorial_pubsub_publish opc.eth://0a-00-27-00-00-08:3.3 eth0.3 `

`eth0.3` is your newly created VLAN interface

sdick97, 2020-06-09:
Thanks for that.

closed 2020-06-09 by sdick97