open62541/open62541 #3683: Connection to multiple clients in same program

hm1920, 2020-06-08:
## Description
Hello,
I am trying to connect different clients to different servers in the same program and subscribe to asynchronous responses .
It is possible to do that using ` UA_Client_run_iterate ` without being blocked ?

Thank you in advance

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.1
 - [x] Other OPC UA SDKs used (client or server): Client
 - [x] Operating system: Ubuntu 18.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-06-08:
You have to call UA_Client_run_iterate with a timeout of 0.
We are working on a solution for driving several clients in a single control flow.
But it is not there yet.

NoelGraf, 2021-04-30:
@hm1920 is something still open? Or can the issue be closed?

closed 2021-05-06 by NoelGraf