open62541/open62541 #3687: Feture/Bug: Generate random NodeIds in specific namespaces

ccvca, 2020-06-09:
## Description
https://github.com/open62541/open62541/blob/master/include/open62541/server.h#L722-L723
This documentaion is not quire right, as (at least the Zip-Tree and Hashmap Implementation) does not compare against NodeId_NULL, instead it will only check if Identifier is Numeric and 0 (so the namespaceindex is ignored).

https://github.com/open62541/open62541/blob/a524accc338c9611a5ec60bf595c1a587243b457/plugins/ua_nodestore_ziptree.c#L187-L188

https://github.com/open62541/open62541/blob/a524accc338c9611a5ec60bf595c1a587243b457/plugins/ua_nodestore_hashmap.c#L314-L315

I really like this behavior, as it is quite easy to create Nodes in specific namespaces. (see code below)
**But: Compared to the Documentation, this is a Bug.** So there are two possible solutions:

1. Update the documentation, so the actual behavior is documented.
2. Fix the implementation. 

I prefer the 1. solution, as I want to generate instances in specific namespaces and keep NamespaceIndex 0 clean (so all instances can be exported to a XML-Nodeset easily).

## Background Information / Reproduction Steps
The following code provide this behaviour by "abusing" this difference.

```cpp
UA_StatusCode generateChildNodeIdInParentNs(
    UA_Server *server,
    const UA_NodeId *sessionId, void *sessionContext,
    const UA_NodeId *sourceNodeId,
    const UA_NodeId *targetParentNodeId,
    const UA_NodeId *referenceTypeId,
    UA_NodeId *targetNodeId)
{
  if(UA_NodeId_equal(targetNodeId, &UA_NODEID_NULL)
  && !UA_NodeId_equal(targetParentNodeId, &UA_NODEID_NULL))
  {
    targetNodeId->namespaceIndex = targetParentNodeId->namespaceIndex;
  }
  return UA_STATUSCODE_GOOD;
}

[...]
UA_Server_getConfig(pServer)->nodeLifecycle.generateChildNodeId = generateChildNodeIdInParentNs;
[...]

UA_Server_addObjectNode(
          pServer,
          UA_NODEID_NUMERIC(4, 0), //<-- Specify the namespace index of the whole object here
          *ParentNodeId.NodeId,
          *referenceType.NodeId,
          *browseName.QualifiedName,
          *typeDef.NodeId,
          objAttr,
          NULL,
          member.NodeId);
```
## Checklist
 - [x] open62541 Version (release number or git tag): a524accc338c9611a5ec60bf595c1a587243b457
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-06-09:
I agree. Let's correct the documentation.

Other comments @open62541/open62541-core ?

ccvca, 2020-06-17:
Just to prevent the efford of correcting a pull request.
I would suggest adding the folloiging text instead for all places where UA_NODEID_NULL is described.
`When passing UA_NODEID_NUMERIC(X,0) an unused nodeid in namespace X will be used. E.g. passing UA_NODEID_NULL will result in a nodeid in namespace 0.`

I want to prevent the term "random" because the algorithm is not random at all. Adding a node, removing it afterwards and adding it again will result in the same NodeId.

Any other suggestions?

jpfr, 2020-06-22:
Sounds good.

PR #3746 (2020-06-25): Fix (documentation): Pass UA_NODEID_NUMERIC(X,0) for a unused NodeId in namespace X (#3687)
Resolves #3687, it feels like I missed many places, does this only occure twice?
closed 2020-06-25 by jpfr