open62541/open62541 #3688: Multinodes read get BadRequestHeaderInvalid if read nodes have bad status code

reyanvaldes, 2020-06-09:
## Description
When try to read multi nodes with bad status using the UA_Client_Service_Read got an error "BadRequestHeaderInvalid" in KepwareEx 6.8. If reading nodes with good status code, can read fine. I am following this example https://stackoverflow.com/questions/47591339/multiple-node-read-in-single-request-from-client-in-open62541. 
Thanks for any recommendation. 

## Background Information / Reproduction Steps
In kepware, create a channel with siemens PLC, that actually doesn't exist or not connected and created a couple of int tags with following address: MW0, MW2.
In the  multi nodes example try to read these two tags. 
[testing_kepware.zip](https://github.com/open62541/open62541/files/4752140/testing_kepware.zip)

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 0.3
 - [ ] Other OPC UA SDKs used (client or server):  Client
 - [ ] Operating system:  Linux
- [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
[image]
[image]

 - [ ] Critical issue

jpfr, 2020-06-11:
You write that you are on version 0.3.
Can you try with the current stable release 1.0?

NoelGraf, 2021-04-30:
@reyanvaldes what's the status?

NoelGraf, 2021-06-04:
I would close the issue for now. Pls reopen if required!

closed 2021-06-04 by NoelGraf