open62541/open62541 #3692: Open62541 LDS-ME server throwing error when processing registration request [GetEndpoints failed with BadDisconnect].

jdhanasekar, 2020-06-10:
Hi open62541,

I am working on to set up aopen62541 LDS-ME server. Already, I have build open62541 LDS sever successfully, able to register with other servers. 

**STEPS FOLLOWED:**

1. I am running ```server_multicast.c``` from /examples/discovery for LDS-ME server. 
2. After executing  ```server_multicast```  executable, I am getting IP address of LDS-ME like this,
```TCP network layer listening on opc.tcp://quanta:38691/```. 
3. Using this address ```opc.tcp://quante:38691```, I am trying to register my Milo server with open62541 LDS ME .
4. While sending registration request from ```Milo server```, I am observing errors in both LDS-ME prompt and Milo Server IDE. 

**CMAKE COMMAND USED:**
```
gcc -std=c99 -o server_multicast ../examples/discovery/server_multicast.c -lopen62541

./server_multicast
```

**ERROR OBSERVED / LOG :**
```
[2020-06-10 10:40:54.698 (UTC+0550)] warn/server	Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2020-06-10 10:40:54.698 (UTC+0550)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-06-10 10:40:54.698 (UTC+0550)] warn/network	Multicast DNS: outbound interface 0.0.0.0, it means that the first OS interface is used (you can explicitly set the interface by using 'discovery.mdnsInterfaceIP' config parameter)
[2020-06-10 10:40:54.698 (UTC+0550)] info/network	TCP network layer listening on opc.tcp://quanta:45555/
[2020-06-10 10:40:54.698 (UTC+0550)] info/server	Multicast DNS: add record for domain: Sample Multicast Server-quanta._opcua-tcp._tcp.local.
[2020-06-10 10:40:54.699 (UTC+0550)] info/server	dhanasekar retval = 0
[2020-06-10 10:40:54.699 (UTC+0550)] info/server	dhanasekar UA_STATUSCODE_GOOD = 0
[2020-06-10 10:40:54.699 (UTC+0550)] info/server	Server started. Waiting for announce of LDS Server.
[2020-06-10 10:41:16.779 (UTC+0550)] info/network	Connection 6 | New connection over TCP from 10.1.60.101
[2020-06-10 10:41:16.813 (UTC+0550)] info/channel	Creating a new SecureChannel
[2020-06-10 10:41:16.851 (UTC+0550)] info/channel	Connection 6 | SecureChannel 1 | Opened SecureChannel
[2020-06-10 10:41:16.892 (UTC+0550)] info/channel	Connection 6 | SecureChannel 1 | CloseSecureChannel
[2020-06-10 10:41:16.892 (UTC+0550)] info/network	Connection 6 | Closed
[2020-06-10 10:41:17.132 (UTC+0550)] info/network	Connection 6 | New connection over TCP from 10.1.60.101
[2020-06-10 10:41:17.133 (UTC+0550)] info/channel	Creating a new SecureChannel
[2020-06-10 10:41:17.136 (UTC+0550)] info/channel	Connection 6 | SecureChannel 2 | Opened SecureChannel
[2020-06-10 10:41:17.151 (UTC+0550)] info/channel	Connection 6 | SecureChannel 2 | Session 5b29cbda-2984-84c7-acca-fc3f80f8cf6c created
[2020-06-10 10:41:17.158 (UTC+0550)] info/session	Connection 6 | SecureChannel 2 | Session ns=1;g=5b29cbda-2984-84c7-acca-fc3f80f8cf6c | ActivateSession: Session activated
[2020-06-10 10:41:17.233 (UTC+0550)] info/server	Multicast DNS: add record for domain: opc.tcp://localhost:4840-localhost._opcua-tcp._tcp.local.
[2020-06-10 10:41:17.263 (UTC+0550)] info/server	Another server announced itself on opc.tcp://localhost:12686/milo/discovery
[2020-06-10 10:41:17.263 (UTC+0550)] info/server	LDS-ME server found on opc.tcp://localhost:12686/milo/discovery
[2020-06-10 10:41:17.263 (UTC+0550)] info/server	dhanasekar before endpointRegister
[2020-06-10 10:41:17.263 (UTC+0550)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-06-10 10:41:17.263 (UTC+0550)] info/server	dhanasekar @getRegisterEndpointFromServeretval discoveryServerUrl = opc.tcp://localhost:12686/milo/discovery
[2020-06-10 10:41:17.264 (UTC+0550)] warn/network	Connection to  failed with error: Transport endpoint is not connected
[2020-06-10 10:41:17.264 (UTC+0550)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadDisconnect
[2020-06-10 10:41:17.264 (UTC+0550)] info/server	dhanasekar @getRegisterEndpointFromServeretval retval = -2136145920
[2020-06-10 10:41:17.264 (UTC+0550)] error/server	GetEndpoints failed with BadDisconnect
Segmentation fault (core dumped)

```
**OPERATING SYSTEM:**
Linux Ubuntu 18.04

Please help me to solve this issue. 
If I committed mistake, Please correct me.

Pro, 2020-06-10:
We need the version of open62541, which you are using.
Next time please fill out the complete issue template.

If I remember correctly, this exact issue is already fixed in master.

jdhanasekar, 2020-06-10:
Sorry for incomplete information,

Yesterday, I have downloaded source like this,

```
git clone https://github.com/open62541/open62541.git
```

The version is ,

[image]

jdhanasekar, 2020-06-10:
Pro,

I am using this commit ID,

[image]

jdhanasekar, 2020-06-11:
Pro,

I have been waiting for your input. Please provide it .

I am having 4 month old source( manually downloaded zip). In that, I have tried same experiment steps which followed previously. I am receiving the  below error and but, I am not seeing any error in Milo IDE when sending registration request. 

```
user@quanta:~/dhanasekar/BMS/OPC/open62541-master/build$ gcc -std=c99 -o server_multicast ../examples/discovery/server_multicast.c -lopen62541
user@quanta:~/dhanasekar/BMS/OPC/open62541-master/build$ ./server_multicast 
[2020-06-11 10:41:48.028 (UTC+0550)] warn/server	Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2020-06-11 10:41:48.029 (UTC+0550)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-06-11 10:41:48.029 (UTC+0550)] warn/network	Multicast DNS: outbound interface 0.0.0.0, it means that the first OS interface is used (you can explicitly set the interface by using 'discovery.mdnsInterfaceIP' config parameter)
[2020-06-11 10:41:48.029 (UTC+0550)] info/network	TCP network layer listening on opc.tcp://quanta:39711/
[2020-06-11 10:41:48.029 (UTC+0550)] info/server	Multicast DNS: add record for domain: Sample Multicast Server-quanta._opcua-tcp._tcp.local.
[2020-06-11 10:41:48.029 (UTC+0550)] info/server	Server started. Waiting for announce of LDS Server.
[2020-06-11 10:42:33.732 (UTC+0550)] info/network	Connection 6 | New connection over TCP from 10.1.60.101
[2020-06-11 10:42:33.763 (UTC+0550)] info/channel	Creating a new SecureChannel
[2020-06-11 10:42:33.803 (UTC+0550)] info/channel	Connection 6 | SecureChannel 1 | Opened SecureChannel
[2020-06-11 10:42:33.843 (UTC+0550)] info/channel	Connection 6 | SecureChannel 1 | CloseSecureChannel
[2020-06-11 10:42:33.844 (UTC+0550)] info/network	Connection 6 | Closed
[2020-06-11 10:42:34.131 (UTC+0550)] info/network	Connection 6 | New connection over TCP from 10.1.60.101
[2020-06-11 10:42:34.133 (UTC+0550)] info/channel	Creating a new SecureChannel
[2020-06-11 10:42:34.134 (UTC+0550)] info/channel	Connection 6 | SecureChannel 2 | Opened SecureChannel
[2020-06-11 10:42:34.144 (UTC+0550)] info/channel	Connection 6 | SecureChannel 2 | Session e0b1ba30-3c69-e96c-d95d-c58c1b21d2fd created
[2020-06-11 10:42:34.150 (UTC+0550)] info/session	Connection 6 | SecureChannel 2 | Session ns=1;g=e0b1ba30-3c69-e96c-d95d-c58c1b21d2fd | ActivateSession: Session activated
[2020-06-11 10:42:34.201 (UTC+0550)] info/server	Multicast DNS: add record for domain: opc.tcp://localhost:4840-localhost._opcua-tcp._tcp.local.
[2020-06-11 10:42:34.230 (UTC+0550)] info/server	Another server announced itself on opc.tcp://localhost:12686/milo/discovery
[2020-06-11 10:42:34.230 (UTC+0550)] info/server	LDS-ME server found on opc.tcp://localhost:12686/milo/discovery
[2020-06-11 10:42:34.230 (UTC+0550)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-06-11 10:42:34.230 (UTC+0550)] warn/network	Connection to  failed with error: Transport endpoint is not connected
[2020-06-11 10:42:34.230 (UTC+0550)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadDisconnect
[2020-06-11 10:42:34.230 (UTC+0550)] error/server	GetEndpoints failed with BadDisconnect
[2020-06-11 10:42:34.231 (UTC+0550)] error/server	Could not find any suitable endpoints on discovery server

```

jdhanasekar, 2020-06-13:
I have debugged and found that below  line in ```UA_ClientConnectionTCP_poll()``` function of ```arch/network_tcp.c``` is causing the error.

```
/* Non blocking connect */
    int error = UA_connect(clientsockfd, tcpConnection->server->ai_addr,
                           tcpConnection->server->ai_addrlen);
```

It is throwing ```error = -1```. 

When I decode the socket address ```tcpConnection->server->ai_addr```, it is showing ``` -921510880```. 
I don know why it is showing negative value. 
Please help me to progress further.

jdhanasekar, 2020-06-13:
I have debugged further and found error occured because of ,
```
103	/* Software caused connection abort */
```
I don't know why connection is aborted. 
What has pushed the connection to abort ?

jdhanasekar, 2020-06-15:
Hi Pro,

In which ```version```, issue is fixed ?.
You mean I need to pull the ```pack/master``` branch source version. 
Please provide your comments ....

jdhanasekar, 2020-06-15:
In latest release ```v1.1-29-g6dc50720```, I am observing the same error.

I am guessing that I am using incorrect cmake build macros.  

```
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=MinSizeRel 
-DUA_NAMESPACE_ZERO=MINIMAL
 -DUA_ENABLE_AMALGAMATION=OFF 
-DUA_ENABLE_DISCOVERY_MULTICAST=ON 
-DUA_ENABLE_DISCOVERY=ON .
```
Please confirm me, Am I using correct build flag?

**STEPS FOLLOWED:**
```
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=MinSizeRel -DUA_NAMESPACE_ZERO=MINIMAL -DUA_ENABLE_AMALGAMATION=OFF -DUA_ENABLE_DISCOVERY_MULTICAST=ON -DUA_ENABLE_DISCOVERY=ON .
make
sudo make 

gcc -std=c99 -o server_multicast ../examples/discovery/server_multicast.c -lopen62541

./server_multicast
```

closed 2020-06-25 by jdhanasekar