open62541/open62541 #3697: UA Client disconnects after LifeTime expired

Pietro2018, 2020-06-12:
## Description
I'm building an OPC UA Client using amalgamated code, and I'me experiencing disconnection from server every 15', because LifeTime counter has expired. I'm connecting using Basic256Sha256 Sign (connecting to Ansi C Server) and I've tried also only Sign+Encrypt (connecting to KepServer EX 6).
The client is configured to monitor simulated values in servers.

## Background Information / Reproduction Steps

Using Unified Automation AnsiC Server I get the following every 15 minutes:

_09:50:00.380|W|47E4* ProcessOpenSecureChannelRequest: Identical nonces used.
09:50:00.380|E|47E4* OpcUa_SecureListener_ProcessRequest: Closing channel due error 0x80240000!
09:50:00.380|W|47E4* OpcUa_TcpListener_ReadEventHandler: Process Request returned an error (0x80240000)!
09:50:00.380|W|47E4* OpcUa_TcpListener_ReadEventHandler: Closing socket (0x00000000)!
10:02:52.002|W|47E4* UaServer_SubscriptionManager_PublishCallback: Lifetime counter expired! SubscriptionId 306963847_

UA_Client_getState reports UA_STATUSCODE_BADAPPLICATIONSIGNATUREINVALID for Ansi C Server.

Connecting to KepServerEX 6 I've the same problem but UA_Client_getState reports UA_STATUSCODE_BADAPPLICATIONSIGNATUREINVALID error code.

Used CMake options:

```
UA_ENABLE_AMALGAMATION x
UA_ENABLE_DA x
UA_ENABLE_DISCOVERY x
UA_ENABLE_ENCRYPTION x
UA_ENABLE_METHODCALLS x
UA_ENABLE_NODEMANAGEMENT x
UA_ENABLE_PARSING x
UA_ENABLE_SUBSCRIPTIONS x
```

## Checklist

Please provide the following information:

 - [x ] open62541 Version (release number or git tag): 1.1 rc1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x ] Operating system: Windows 10 Pro
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x ] Critical issue

Pietro2018, 2020-06-14:
Trying release 1.1: this fix the problem with UaAnsiCServer, **but not with KepServerEx 6.9**: monitored item values stop updating after 7 minutes. At this time UA_Client_connect return 0, and UA_Client_getState return all status ok (channelState = UA_SECURECHANNELSTATE_OPEN, sessionState = UA_SESSIONSTATE_ACTIVATED, connectionSTate=0x00000000).
The server is still working (checked with UAExpert).

The handler_DataChanged() callback is no more called.

I'm using synchronous api, my program structure is like client_connect_loop.c, but I do not read variables, I have monitored them.

There it is a workaround?

Pietro2018, 2020-06-14:
This the KepServerEx log at the time client stops receiving events. It seems the client stops sending PublishRequest.

[image]

Pietro2018, 2020-06-15:
**Update: UaAnsiCServer** warning for using the same "nonces" after 9 hours of run.
In the picture the UaAnsiCserver was started at 22:58, while opc ua client at 00:38.

The program correctly update values of monitored items for UaAnsiCServer.

[image]

also the LifeTimeCounter expired warning is back
[image]

**Update KepServerEX**: after a while the client has starter sending PublishRequest followed by PublishRsponse (with opc item value) BUT the handler_DataChanged() callback was not called.
Item values are not reported to my program.

Pietro2018, 2020-06-15:
14:00 updates stopped also for UaAnsiCServer... 
Tried to restart (R command and close/start application) with no success (no communication restarted).

jpfr, 2020-06-16:
Please try with v1.1 final.
Ate least the nonce warning should be gone.

jpfr, 2020-06-16:
V1.1 was tested against a range of commercial sdk.

jki2020, 2020-06-17:
I have same kind of problem with OPC UA Client in Linux environment, open62541 Version 1.1.
At first time renewing the SecureChannel works but second time it fails.
Version 1.0.1 is working fine. Tested with UAServer and UAServerCpp.

17/06/20 11:01:16.06        opcuaclt1: (D) A_Client_run_iterate 1000
[2020-06-17 11:01:16.644 (UTC+0300)] debug/channel      Connection 4 | SecureChannel -270984319 | Requesting to renew the SecureChannel
[2020-06-17 11:01:16.644 (UTC+0300)] debug/channel      Requesting to open a SecureChannel
[2020-06-17 11:01:16.682 (UTC+0300)] debug/channel      OPN message sent
[2020-06-17 11:01:16.752 (UTC+0300)] trace/channel      Connection 4 | SecureChannel -270984319 | Verifying chunk signature
[2020-06-17 11:01:16.752 (UTC+0300)] trace/channel      Connection 4 | SecureChannel -270984319 | Calculated padding size to be 118
[2020-06-17 11:01:16.752 (UTC+0300)] debug/channel      Connection 4 | SecureChannel -270984319 | Generating SecureChannel keys
[2020-06-17 11:01:16.752 (UTC+0300)] trace/channel      Connection 4 | SecureChannel -270984319 | Generating new local keys
[2020-06-17 11:01:16.752 (UTC+0300)] trace/channel      Connection 4 | SecureChannel -270984319 | Generating new remote keys
[2020-06-17 11:01:16.752 (UTC+0300)] info/channel       Connection 4 | SecureChannel -270984319 | SecureChannel renewed
17/06/20 11:01:17.07        opcuaclt1: (D) A_Client_run_iterate 1000

...

17/06/20 11:05:02.00        opcuaclt1: (D) A_Client_run_iterate 1000
[2020-06-17 11:05:02.022 (UTC+0300)] debug/channel      Connection 4 | SecureChannel -270984319 | Requesting to renew the SecureChannel
[2020-06-17 11:05:02.023 (UTC+0300)] debug/channel      Requesting to open a SecureChannel
[2020-06-17 11:05:02.064 (UTC+0300)] debug/channel      OPN message sent
[2020-06-17 11:05:02.088 (UTC+0300)] trace/channel      Connection 4 | SecureChannel -270984319 | Invalid message type
[2020-06-17 11:05:02.088 (UTC+0300)] warn/channel       Connection 4 | SecureChannel -270984319 | Receiving the response failed with StatusCode Good
[2020-06-17 11:05:02.088 (UTC+0300)] warn/channel       Connection 0 | SecureChannel 0 | Could not receive with StatusCode BadConnectionClosed
[2020-06-17 11:05:02.088 (UTC+0300)] info/client        Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good

Pietro2018, 2020-06-17:
Yes, I've used v1.1 as soon as it was released and it works only with unsecure connection. With secure connection (I tried securityMode = Sign+Encrypt) and securityPolicy=Basic256Sha256) the problem happens after 15 minutes.

[image]

Amalgamated code has been generate with with UA_MULTITHREAD = 199 (I need support to multithreading and alos wit UA_MULTITHREAD = 0. The program exhibits the same behaviour:
UA_Client_connect returns 0x80580000 UA_STATUSCODE_BADAPPLICATIONSIGNATUREINVALID

[image]

and related Ua AnsiC output
[image]

```c
/*This is my sample code (from examples) */
#include "open62541.h"
#include "common.h"
#include <stdlib.h>

#ifdef UA_ENABLE_SUBSCRIPTIONS
static void
handler_DataChanged(UA_Client *client, UA_UInt32 subId, void *subContext,
					UA_UInt32 monId, void *monContext, UA_DataValue *value) {
}
#endif

int Subscribe(UA_Client *client, UA_UInt32 *subId) {

	/* Create a subscription */
	UA_CreateSubscriptionRequest request = UA_CreateSubscriptionRequest_default();
	UA_CreateSubscriptionResponse response = UA_Client_Subscriptions_create(client, request, NULL, NULL, NULL);

	*subId = response.subscriptionId;
	if (response.responseHeader.serviceResult == UA_STATUSCODE_GOOD)
		printf("Create subscription succeeded, id %u\n", *subId);

	UA_MonitoredItemCreateRequest monRequest =
		UA_MonitoredItemCreateRequest_default(UA_NODEID_STRING(4, "Demo.Dynamic.Scalar.Int32"));

	UA_MonitoredItemCreateResult monResponse =
		UA_Client_MonitoredItems_createDataChange(client, response.subscriptionId,
			UA_TIMESTAMPSTORETURN_BOTH,
			monRequest, NULL, handler_DataChanged, NULL);
	if (monResponse.statusCode == UA_STATUSCODE_GOOD)
		printf("Monitoring 'Demo.Dynamic.Scalar.Int32', id %u\n", monResponse.monitoredItemId);

	UA_MonitoredItemCreateRequest monRequest2 =
		UA_MonitoredItemCreateRequest_default(UA_NODEID_STRING(4, "Demo.Dynamic.Scalar.Int16"));
	//UA_MonitoredItemCreateRequest_default(UA_NODEID_STRING(1, "the.answer"));

	UA_MonitoredItemCreateResult monResponse2 =
		UA_Client_MonitoredItems_createDataChange(client, response.subscriptionId,
			UA_TIMESTAMPSTORETURN_BOTH,
			monRequest2, NULL, handler_DataChanged, NULL);
	if (monResponse2.statusCode == UA_STATUSCODE_GOOD)
		printf("Monitoring 'Demo.Dynamic.Scalar.Int16', id %u\n", monResponse2.monitoredItemId);

	UA_MonitoredItemCreateRequest monRequest3 =
		//UA_MonitoredItemCreateRequest_default(UA_NODEID_STRING(4, "Demo.Dynamic.Scalar.String"));
	UA_MonitoredItemCreateRequest_default(UA_NODEID_NUMERIC(0, 2258));
	
	UA_MonitoredItemCreateResult monResponse3 =
		UA_Client_MonitoredItems_createDataChange(client, response.subscriptionId,
			UA_TIMESTAMPSTORETURN_BOTH,
			monRequest3, NULL, handler_DataChanged, NULL);
	if (monResponse3.statusCode == UA_STATUSCODE_GOOD)
		printf("Monitoring '0, 2258', id %u\n", monResponse3.monitoredItemId);

	return 0;
}

#define ENDPOINT		"opc.tcp://HP1234:48020"

int main(int argc, char *argv[]) {

	UA_Client *client = UA_Client_new();
	UA_ClientConfig *config = UA_Client_getConfig(client);

	//UA_ClientConfig_setDefault(UA_Client_getConfig(client));
	

	char szServerCertificate[_MAX_PATH];
	char szServerPrivateKey[_MAX_PATH];
	UA_ByteString certificate = loadFile("C:\\OPCUA_Client\\client_cert.der");
	UA_ByteString privateKey = loadFile("C:\\OPCUA_Client\\client_key.der");

	size_t trustListSize = 1;
	UA_STACKARRAY(UA_ByteString, trustList, trustListSize);
	trustList[0] = loadFile("C:\\OPCUA_Client\\uaservercert.der");
	//trustList[0] = loadFile("kepserverex_ua_server.der");
	UA_ByteString *revocationList = NULL;
	size_t revocationListSize = 0;

	//config->securityMode = UA_MESSAGESECURITYMODE_NONE;
	//config->securityMode = UA_MESSAGESECURITYMODE_SIGN;
	config->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;	

	UA_StatusCode retval = UA_ClientConfig_setDefaultEncryption(config, certificate, privateKey,
						trustList, trustListSize, revocationList, revocationListSize);

	//UA_StatusCode retval = UA_ClientConfig_setDefault(config);

	config->securityPoliciesSize = 2;
	// must be added: without the client doesn't connect to server
	config->securityPolicies = new UA_SecurityPolicy[config->securityPoliciesSize];
	retval = UA_SecurityPolicy_None(&config->securityPolicies[0], UA_BYTESTRING_NULL, &config->logger);
	
	config->securityPolicyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256");
	retval = UA_SecurityPolicy_Basic256Sha256(
		&config->securityPolicies[1],
		certificate,
		privateKey,
		&config->logger);

	//UA_SecurityPolicy_None(&config->securityPolicies[0], UA_BYTESTRING_NULL, NULL);
	

	UA_ByteString_clear(&certificate);
	UA_ByteString_clear(&privateKey);
	for (size_t deleteCount = 0; deleteCount < trustListSize; deleteCount++) {
		UA_ByteString_clear(&trustList[deleteCount]);
	}

	bool fReload = true;
	UA_UInt32 subId;

	while (1) {		
		UA_StatusCode retval;
		retval = UA_Client_connect(client, ENDPOINT); //anonymous access		
		//retval = UA_Client_connectUsername(client, ENDPOINT, "root", "secret"); //access to UA AnsiC Server

		if (retval != UA_STATUSCODE_GOOD) {
			fReload = true;
			Sleep(1000);
			continue;
		}

		if (fReload) {
			fReload = false;
			Subscribe(client, &subId);
		}

		// The first publish request should return the initial value of the variable
		UA_Client_run_iterate(client, 1000);
	}
	
	/* Delete the subscription */
	if (UA_Client_Subscriptions_deleteSingle(client, subId) == UA_STATUSCODE_GOOD)
		printf("Subscription removed\n");

	UA_Client_disconnect(client);
	UA_Client_delete(client);
	return EXIT_SUCCESS;
}
```

jpfr, 2020-06-18:
This commit should have resolved exactly this issue:
https://github.com/open62541/open62541/commit/745cfbdb4a5d084d861a249461703fa41a5fd868

That was merged just prior to the v1.1 release.
Are you on the v1.1-rc1 release candidate or the final release?

Will verify locally. But don't have the reproduction setup yet.

Pietro2018, 2020-06-19:
Thank you
I'm using 1.1 release, that doesn't contain the modification of the 745cfbd commit (I've just compared source code). Today I will download the latest source code and try it. I will inform you about progress.

Pietro2018, 2020-06-21:
I have tried using 1.1 master (dated 2020 06 19) but with no success. The problem still remains.

Here you can find my last sample program, you can use to reproduce the problem. 

**Library version**
master 1.1. dated 2020 06 19, with UA_THREADING 199. Amalgamated code. 

**Test condition**
Program built using VS2015.
I'm connecting KepServerEX 6.9 (to simulation tag Channel1.Device1.Tag1) with securityPolicy = BASIC256Sha256,securityMode = Sign & Encrypt,  no user and password.

**Fault**
After 7 minutes from program start,  UA_Client_connect() returns 0x80ab0000.

**Additional note**
I can reproduce the problem also on another sample program with call backs (based on you samples) and in my multithreaded opc ua client.

**console output**
as you can see the secureChannel is renewed but, after that, there it is an "unknown SecurityToken"

_monID = 1!
[2020-06-21 23:52:59.802 (UTC+0200)] warn/client        Client already connected
monID = 1!
[2020-06-21 23:53:00.296 (UTC+0200)] warn/client        Client already connected
monID = 1!
[2020-06-21 23:53:00.808 (UTC+0200)] warn/client        Client already connected
[2020-06-21 23:53:00.824 (UTC+0200)] warn/client        Client already connected
[2020-06-21 23:53:00.848 (UTC+0200)] info/channel       Connection 316 | SecureChannel 4285573788 | SecureChannel renewed
[2020-06-21 23:53:00.850 (UTC+0200)] warn/client        Client already connected
[2020-06-21 23:53:01.301 (UTC+0200)] warn/channel       Connection 316 | SecureChannel 4285573788 | **Received an unknown SecurityToken**
[2020-06-21 23:53:01.302 (UTC+0200)] warn/channel       Connection 316 | SecureChannel 4285573788 | Receiving the response failed with StatusCode BadSecureChannelTokenUnknown
[2020-06-21 23:53:01.303 (UTC+0200)] warn/channel       Connection 0 | SecureChannel 0 | Could not receive with StatusCode BadConnectionClosed
[2020-06-21 23:53:01.305 (UTC+0200)] info/client        Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good
[2020-06-21 23:53:01.305 (UTC+0200)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2020-06-21 23:53:01.359 (UTC+0200)] info/channel       Connection 316 | SecureChannel 3465276748 | Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2020-06-21 23:53:01.361 (UTC+0200)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2020-06-21 23:53:01.376 (UTC+0200)] info/client        Received a ServiceFault response
[2020-06-21 23:53:01.377 (UTC+0200)] info/client        The ServiceResult has the StatusCode BadInvalidArgument
[2020-06-21 23:53:01.377 (UTC+0200)] error/client       ActivateSession failed with error code BadInvalidArgument
[2020-06-21 23:53:01.377 (UTC+0200)] info/client        Client Status: ChannelState: Open, SessionState: ActivateRequested, ConnectStatus: BadInvalidArgument_

**Source code**
Full project available, but to big to be attached to this form.

```c
/*This is my sample code (from examples) */
#include "open62541.h"
#include "common.h"
#include <stdlib.h>

#ifdef UA_ENABLE_SUBSCRIPTIONS
static void
handler_DataChanged(UA_Client *client, UA_UInt32 subId, void *subContext,
	UA_UInt32 monId, void *monContext, UA_DataValue *value) {

	printf("monID = %d! \n", monId);

}
#endif

int Subscribe(UA_Client *client, UA_UInt32 *subId) {

	/* Create a subscription */
	UA_CreateSubscriptionRequest request = UA_CreateSubscriptionRequest_default();
	UA_CreateSubscriptionResponse response = UA_Client_Subscriptions_create(client, request, NULL, NULL, NULL);

	*subId = response.subscriptionId;
	if (response.responseHeader.serviceResult == UA_STATUSCODE_GOOD)
		printf("Create subscription succeeded, id %u\n", *subId);

[... 59 lines cut ...]

		if (retval != UA_STATUSCODE_GOOD) {
			fReload = true;
			Sleep(1000);
			continue;
		}

		if (fReload) {
			fReload = false;
			Subscribe(client, &subId);
		}

		// The first publish request should return the initial value of the variable
		UA_Client_run_iterate(client, 1000);
	}

	/* Delete the subscription */
	if (UA_Client_Subscriptions_deleteSingle(client, subId) == UA_STATUSCODE_GOOD)
		printf("Subscription removed\n");

	UA_Client_disconnect(client);
	UA_Client_delete(client);
	return EXIT_SUCCESS;
}
```

jpfr, 2020-06-23:
Fixed in #3742.
Tested against current Kepware servers.

jpfr, 2020-06-24:
Solved on 1.1.
Can you verify? Then we will tag a bugfix release.

Pietro2018, 2020-06-24:
It works fine, no more errors. Tested against KepServerEx and UaAnsiCServer. Thanks you for your work!

closed 2020-06-24 by jpfr

jki2020, 2020-06-25:
I downloaded the Version 1.1.1 but I have still the problem that renewing the SecureChannel fails in second time.

Linux environment, open62541-1.1.1.tar.gz, cmake parameters
-DOPEN62541_VERSION=v1.1.1 -DUA_ENABLE_ENCRYPTION=ON 
-DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON -DUA_LOGLEVEL=100 
-DUA_BUILD_EXAMPLES=ON

Version 1.0.1 is working fine. Tested with UAServerC and UAServerCpp.

First time

[2020-06-25 15:12:35.422 (UTC+0300)] debug/channel      Connection 4 | SecureChannel 962064972 | Generating new local nonce
[2020-06-25 15:12:35.422 (UTC+0300)] debug/channel      Connection 4 | SecureChannel 962064972 | Requesting to renew the SecureChannel
[2020-06-25 15:12:35.422 (UTC+0300)] debug/channel      Requesting to open a SecureChannel
[2020-06-25 15:12:35.446 (UTC+0300)] debug/channel      OPN message sent
[2020-06-25 15:12:35.513 (UTC+0300)] trace/channel      Connection 4 | SecureChannel 962064972 | Verifying chunk signature
[2020-06-25 15:12:35.513 (UTC+0300)] trace/channel      Connection 4 | SecureChannel 962064972 | Calculated padding size to be 118
[2020-06-25 15:12:35.513 (UTC+0300)] trace/channel      Connection 4 | SecureChannel 962064972 | Generating new local keys
[2020-06-25 15:12:35.513 (UTC+0300)] info/channel       Connection 4 | SecureChannel 962064972 | SecureChannel renewed

Second time

[2020-06-25 15:25:09.989 (UTC+0300)] warn/channel       Connection 4 | SecureChannel 962064972 | Receiving the response failed with StatusCode BadConnectionClo
sed
[2020-06-25 15:25:09.989 (UTC+0300)] warn/channel       Connection 0 | SecureChannel 0 | Could not receive with StatusCode BadConnectionClosed
[2020-06-25 15:25:09.989 (UTC+0300)] info/client        Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good

jki2020, 2020-06-29:
FIxed to define cc->connectivityCheckInterval = 2000;

jki2020, 2020-06-30:
UA_Client_run_iterate is not working in the same way in Version 1.1.1 like in Version 1.0.1.

The following program fails in second renewing of SecureChannel.
It's like the example client_connect_loop.c but the UA_Client_readValueAttribute is replaced with UA_Client_run_iterate.

/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

/**
 * Client disconnect handling
 * --------------------------
 * This example shows you how to handle a client disconnect, e.g., if the server
 * is shut down while the client is connected. You just need to call connect
 * again and the client will automatically reconnect.
 *
 * This example is very similar to the tutorial_client_firststeps.c. */

#include <open62541/client_config_default.h>
#include <open62541/client_highlevel.h>
#include <open62541/plugin/log_stdout.h>

#include <signal.h>
#include <stdlib.h>

UA_Boolean running = true;

static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_CLIENT, "Received Ctrl-C");
    running = 0;
}

int main(void) {
    signal(SIGINT, stopHandler); /* catches ctrl-c */

    UA_Client *client = UA_Client_new();
    UA_ClientConfig *cc = UA_Client_getConfig(client);
    UA_ClientConfig_setDefault(cc);

    /* default timeout is 5 seconds. Set it to 1 second here for demo */
    cc->timeout = 1000;

    /* Read the value attribute of the node. UA_Client_readValueAttribute is a
     * wrapper for the raw read service available as UA_Client_Service_read. */
    UA_Variant value; /* Variants can hold scalar values and arrays of any type */
    UA_Variant_init(&value);

    /* Endless loop reading the server time */
    while(running) {
        /* if already connected, this will return GOOD and do nothing */
        /* if the connection is closed/errored, the connection will be reset and then reconnected */
        /* Alternatively you can also use UA_Client_getState to get the current state */
        UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://cpm-geysir:48020");
        if(retval != UA_STATUSCODE_GOOD) {
            UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_CLIENT, "Not connected. Retrying to connect in 1 second");
            /* The connect may timeout after 1 second (see above) or it may fail immediately on network errors */
            /* E.g. name resolution errors or unreachable network. Thus there should be a small sleep here */
            UA_sleep_ms(1000);
            continue;
        }

#if 0
        /* NodeId of the variable holding the current time */
        const UA_NodeId nodeId =
                UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_CURRENTTIME);

        retval = UA_Client_readValueAttribute(client, nodeId, &value);
        if(retval == UA_STATUSCODE_BADCONNECTIONCLOSED) {
            UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_CLIENT,
                         "Connection was closed. Reconnecting ...");
            continue;
        }
        if(retval == UA_STATUSCODE_GOOD &&
           UA_Variant_hasScalarType(&value, &UA_TYPES[UA_TYPES_DATETIME])) {
            UA_DateTime raw_date = *(UA_DateTime *) value.data;
            UA_DateTimeStruct dts = UA_DateTime_toStruct(raw_date);
            UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                        "date is: %02u-%02u-%04u %02u:%02u:%02u.%03u",
                        dts.day, dts.month, dts.year, dts.hour, dts.min, dts.sec, dts.milliSec);
        }
        UA_Variant_clear(&value);
#else
        retval = UA_Client_run_iterate(client, 1000);
#endif
        UA_sleep_ms(1000);
    };

    /* Clean up */
    UA_Variant_clear(&value);
    UA_Client_delete(client); /* Disconnects the client internally */
    return EXIT_SUCCESS;
}

jki2020, 2020-07-01:
> This email originated from outside of your organization. Please do not click on links or open attachments unless you recognize the sender and know the content is safe.
>
> Closed #3697<https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fopen62541%2Fopen62541%2Fissues%2F3697&data=02%7C01%7C%7C996502ab96a043cd672f08d8188fba25%7C372ee9e09ce04033a64ac07073a91ecd%7C0%7C0%7C637286352867159838&sdata=k9TpyOeJapAwqRpQSGpcVX2wNLZ%2FpseMYV%2Bl3%2FiUVE4%3D&reserved=0>.

Have you seen my comments to #3697 after it has been closed?
Should I create a new issue of the case?