open62541/open62541 #3702: Client can't connect to local server on win10

matejk87, 2020-06-13:
Hi!
I compiled basic server and client example of the latest open62541(V1.1) lib for win10 (64bit). 
Server seems to work as I have tested it with UAExspert. But I can't connect to server with basic client example.
I've tested this client with different servers(localhost, and external embedded server), without difference.
Here is client response:

[image]

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ x] Operating system: win10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-07-07:
Which version do you use?
If you can, try the latest v1.1.1 release.
A networking problem on windows was resolved earlier this year.

NoelGraf, 2021-04-30:
@matejk87 what's the status

NoelGraf, 2021-05-20:
I would then close the issue. Please reopen if required!

closed 2021-05-20 by NoelGraf