open62541/open62541 #3708: Releasing nodeset_compiler as standalone Python package

wnts, 2020-06-15:
Hello,

Is there any interest (or has any prior effort been made) for releasing the nodeset_compiler as a standalone Python package?
In a project we are working on, we make use of some of the nodeset_compiler modules for parsing and representing an XMLNodeset. The nodeset_compiler being available as a standalone package would simplify reuse and deployment of its modules.

If interested, I can commence working on this.

jpfr, 2020-06-15:
We have a very strong dependency on the nodeset-compiler.
Splitting it out would introduce some friction as people have to check out a second project to build the first.

How would making it a separate project increase reuse in a way that could not be emulated in its current location?

wnts, 2020-06-16:
I might have not given enough details about what I mean with 'releasing as a standalone package'. I do not propose to split out the nodeset_compiler into a separate project, because that would indeed introduce the friction you mention. 

I propose to extend the build process so that an additional artifact is produced in the form of a Python package that is uploaded to PyPI, and **can** be installed separately.
I think this requires not much work, mostly adding a setup.py script to the nodeset_compiler, as well as uploading a source distribution to PyPI (or another PEP503 compliant repository).

That way you have the best of two worlds: 

1. The nodeset_compiler stays in the open62541 source tree, so no additional dependency is introduced, and the internal build dependency of libopen62541.so on the Python code can stay in place. 
2. For those interested only in the nodeset_compiler independent from the open62541 source tree, the Python package can be installed from PyPI.

I believe the second use case is common. Projects building against open62541 often only have a binary package or binary-devel package installed (such as the .deb or .rpm produced by the build process). But if such projects want to use the nodeset_compiler tool, they have to depend on the complete open62541 source package, which is not ideal. Additionally, option 2 also allows other Python projects/packages to depend on the separately nodeset_compiler in the form of a Python package, according to well-established practices (https://packaging.python.org/tutorials/managing-dependencies/).

closed 2025-06-03 by teccheck