open62541/open62541 #3710: The OPCF Nodeset repo was removed

jpfr, 2020-06-15:
The nodeset repository hosted by the OPC Foundation was wiped.
They only give "paying members" access now.

https://github.com/OPCFoundation/UA-Nodeset

Our git submodule dependency still works as we are tied to historic version of the repository.
But we might need another solution going forward.

ccvca, 2020-06-16:
As far as I see this, it only affects the Release Candidates (on master branch). So the submodules should be fine:
https://github.com/OPCFoundation/UA-Nodeset/tree/v1.04
https://github.com/OPCFoundation/UA-Nodeset/commits/v1.04

It is just complicated, if open62541 implements feature bfore they are released, as the required nodes from ns0 are no longer accessable for everyone.

saukijan, 2020-06-23:
I was thinking of creating a package based of [Release artifacts](https://github.com/OPCFoundation/UA-Nodeset/releases) since they are public and tie install it either under `/deps` or `${CMAKE_BINARY_DIR}`, but I need to prototype on the recipe a bit before I do it officially.

Pro, 2020-06-23:
Interesting...
Didn't they (Stefan Hoppe and Co) say in the OPC Day Event today multiple times that the O in OPC stands for openness?

Closing the repo to paying members is not really open...

Pro, 2020-06-23:
See also: https://github.com/OPCFoundation/UA-Nodeset/issues/64

ccvca, 2020-06-23:
Looking at the Nodeset2.xml-Headers (in the Members-Repository) of some Release Candidates they are still MIT...

closed 2025-02-04 by andreasebner