open62541/open62541 #3715: Code level compatibility between releases

mikini, 2020-06-17:
## Description
Please elaborate on the code level compatibility to expect between releases. 

When changes to the library's C API is being done, like fx. [swapping out the return type and adding parameters to UA_Client_getState() from 1.0 to 1.1](https://github.com/open62541/open62541/commit/97ce87470c93423f7adf6b1d8fd87cd8d3252fb3#diff-44b00b50652139af475d451264d49ffdR65), software using the library is going to break and have to be attended to by their developers, if they choose to upgrade.

An official policy, or maybe just a less formal declaration of intent, regarding the breakage to expect between release would allow new library users to better asses the future effort to expect when upgrading in different integration scenarios (distribution binaries, project binaries, custom built binaries, amalgation etc.).
The strict [semantic versioning](https://en.wikipedia.org/wiki/Software_versioning#Degree_of_compatibility) [specification](https://semver.org/) might be an inspiration.

Documentation wise, independent of policy, it would help the decision of whether to upgrade to a specific release if it was accompanied by a structured description of breaking API changes. The [v1.1 release notes](https://github.com/open62541/open62541/releases/tag/v1.1) does describe functional changes but nothing indicates that code will not compile. The above mentioned incompatibility is, however, a consequence of the functional change described as "Track the client state for SecureChannel and Session independently".

On a related note, I'm a bit puzzled that prebuilt packages of the "rc1" release were pushed into the  [Ubuntu PPA named "ppa"](https://launchpad.net/~open62541-team/+archive/ubuntu/ppa) which is described as "Stable release builds of the open62541 project" on launchpad and "Release Builds" in the [project documentation](https://github.com/open62541/open62541/blob/master/doc/installing.rst#debian). Was this intended? To me the meaning of a *candidate* is that it is not yet a release and thus not stable.

## Background Information / Reproduction Steps
My organisation is developing an OPC UA client based on open62541 to relay industrial data. The original intention was to use the prebuilt packages but seeing this I'm thinking we might need to reconsider the amalgamation approach to reduce deployment complexity.

mikini, 2020-06-19:
Below is the actual breakage I experience on `getState()` and its defines when building and following updates from the Ubuntu PPA (this on a 16.04 system).

Currently v1.1 hasn't propagated so the version exposed is 1.1-rc1 ([20200618T204527\~1.1\~rc1\~53\~g0401af07~ubuntu16.04.1](https://launchpad.net/~open62541-team/+archive/ubuntu/ppa/+packages)) and it [doesn't seem like v1.1 is being scheduled for building on "ppa"](https://launchpad.net/~open62541-team/+archive/ubuntu/ppa/+builds?build_text=&build_state=all) although ["daily" has been building 1.1 since 2020-06-13](https://launchpad.net/%7Eopen62541-team/+archive/ubuntu/daily/+builds?build_state=all&build_text=&batch=75&memo=75&start=75).

I noticed other places where I suspect similar breakages when studying [diffs from v1.0 to v1.1](https://github.com/open62541/open62541/compare/v1.0..v1.1), take a closer look at [include/open62541/client.h](https://github.com/open62541/open62541/compare/v1.0..v1.1#diff-44b00b50652139af475d451264d49ffd) for example (at least `UA_Client_run_iterate()` has had its [parameter types modified](https://github.com/open62541/open62541/commit/827d2b2d94075e3c3a59dd31509d29ed0f607e0c) too).

```
main.c: In function ‘main’:
main.c:744:5: error: unknown type name ‘UA_ClientState’
     UA_ClientState state = UA_Client_getState(uaClient);
     ^
main.c:744:28: error: too few arguments to function ‘UA_Client_getState’
     UA_ClientState state = UA_Client_getState(uaClient);
                            ^
In file included from opcua.h:1:0,
                 from main.c:9:
/usr/include/open62541/client.h:65:1: note: declared here
 UA_Client_getState(UA_Client *client,
 ^
main.c:745:18: error: ‘UA_CLIENTSTATE_SESSION’ undeclared (first use in this function)
     if (state != UA_CLIENTSTATE_SESSION && state != UA_CLIENTSTATE_SESSION_RENEWED) {
                  ^
main.c:745:18: note: each undeclared identifier is reported only once for each function it appears in
main.c:745:53: error: ‘UA_CLIENTSTATE_SESSION_RENEWED’ undeclared (first use in this function)
     if (state != UA_CLIENTSTATE_SESSION && state != UA_CLIENTSTATE_SESSION_RENEWED) {
                                                     ^

```

jpfr, 2020-06-30:
@Pro Can we define a policy of what goes into the ppa?
Stable releases or also release candidates?

We today cover a lot more features in the client than we did for the first releases.
Sometimes that growth means that an API has to be changed for the better.
We don't guarantuee a stable API across major releases. That would stifle any progress.

We try our best to keep the API stable.
But in this case, the change was absolutely necessary to ensure that the client can be used as intended by the standard.

Please reopen if required.

closed 2020-06-30 by jpfr

Pro, 2020-06-30:
> @Pro Can we define a policy of what goes into the ppa?

Currently the PPA recipe is setup to take the current `pack/1.1` branch for the stable build (see https://code.launchpad.net/~open62541-team/+recipe/open62541-stable bottom of the page).

Since 1.1 was based off master recently, it includes the breaking changes. Using semantic versoning, we should have chosen 2.0 instead of 1.1 (increase major number for breaking changes, minor number for non-breaking changes).
We already had an open issue about semantic versioning here:
#2923

As already mentioned there, I'm in favour of semantic versioning and would go for that, same as chrome does, just increase the major number if necessary.
@jpfr feel free to comment your bits directly in #2923

Instead of using the `pack/1.1` we could also create a `pack/stable` branch or similar.