open62541/open62541 #3719: Handling of asynchronous events during synchronous function calls in client

mikini, 2020-06-17:
## Description
The client documentation doesn't make it very clear that not handling asynchronous events will break SecureChannel connection/session management (["especially relevant for subscriptions"](https://github.com/open62541/open62541/blob/v1.1/include/open62541/client.h#L38-L43)). It even hints that they can potentially be handled automatically in synchronous function calls (["may be received in a synchronous service call"](https://github.com/open62541/open62541/blob/v1.1/include/open62541/client_subscriptions.h#L31-L33)), depending on whether "service call" == "function call".

Are such functionality present in the library? And if so, which function calls invoke it?

The importance of handling asynchronous events by invoking UA_Client_run_iterate() is not very clear from the current documentation so I've suggested some changes in #3718 to the best of my current knowledge and insights.

## Background Information / Reproduction Steps
Working on and testing a client I experienced disconnects of SecureChannels that baffled me until realising the *need* to call UA_Client_run_iterate() for handling [SecureToken renewal](https://reference.opcfoundation.org/v104/Core/docs/Part4/5.5.2/#5521-description--).

In this scenario the client did use a lot of synchronous calls, mostly UA_Client_readValueAttribute()s, but these seems never to trigger any handling of asynchronous events.

## Used CMake options

```bash
Not customised (using PPA package)
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): [1.0\~33\~g54e2b367](https://launchpad.net/~open62541-team/+archive/ubuntu/ppa/+sourcepub/10941478/+listing-archive-extra) (prebuilt from [Ubuntu "ppa" PPA](https://launchpad.net/~open62541-team/+archive/ubuntu/ppa/))
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 18.04 on x86-64
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

```bash
$ apt-cache policy libopen62541-1
libopen62541-1:
  Installed: 20200119T184421~1.0~33~g54e2b367~ubuntu18.04.1
  Candidate: 20200615T231217~1.1~rc1~52~g9a543201~ubuntu18.04.1
  Version table:
     20200615T231217~1.1~rc1~52~g9a543201~ubuntu18.04.1 500
        500 http://ppa.launchpad.net/open62541-team/ppa/ubuntu bionic/main amd64 Packages
 *** 20200119T184421~1.0~33~g54e2b367~ubuntu18.04.1 100
        100 /var/lib/dpkg/status
$ 
```

```bash
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.4 LTS
Release:        18.04
Codename:       bionic
$ 
```

jpfr, 2020-06-17:
The connection is kept to some extent kept alive in the synchronous processing.

`sendSymmetricServiceRequest` can renew the SecureChannel.
`receiveResponse` also handles async responses. From the SecureChannel-renew as well as other async responses.

I agree that the integration of the synchronous service calls and the connect/reconnect handshake (when the connection was completely lost) should be improved or at least documented better.

NoelGraf, 2021-04-30:
@mikini is something still open? Or can the issue be closed?

NoelGraf, 2021-06-04:
I would close the issue. Reopen if required!

closed 2021-06-04 by NoelGraf