open62541/open62541 #3720: Elliptic curve certificates

jopi2016, 2020-06-18:
Hi,
I would like to know if open62541 supports Elliptic curve certificates.

I've been trying to generate an elliptic curve certificate with the following commands:
`set RANDFILE=.rnd
openssl ecparam -genkey -name prime256v1 -out key.pem
openssl req -new -sha256 -key key.pem -out csr.csr
openssl req -x509 -sha256 -days 365 -key key.pem -in csr.csr -out certificate.pem`

Can I load the generated files into open62541 or do I still need to convert them?

Thanks in advance
jopi2016

jopi2016, 2020-06-18:
An update:
I changed my generation to the following
`openssl ecparam -genkey -name prime256v1 -out server_key.der`
`openssl req -new -sha256 -key server_key.der -out csr.csr`
`openssl req -x509 -sha256 -days 365 -key server_key.der -in csr.csr -out server_publiccert.der`

When I load the generated files and start an open62541-server in policyContext_newContext_sp_basic256() mbedtls_pk_parse_key() fails.

I saw the following lines:
`const mbedtls_md_info_t *mdInfo = mbedtls_md_info_from_type(MBEDTLS_MD_SHA1);`
`int mbedErr = mbedtls_md_setup(&pc->sha1MdContext, mdInfo, MBEDTLS_MD_SHA1);`

Would I need to change SHA1 to SHA256 as the certificate I generated above is a sha256 certificate?

ccvca, 2020-06-18:
I don't think that this is not possible at the moment, as OPC UA itself has just released a draft for ECC in OPC UA a week ago:
OPC 10001-4 (https://opcfoundation.org/developer-tools/specifications-unified-architecture/specification-release-candidates-for-review/ access limited to OPC Foundation members)

ChHaa, 2020-06-18:
ccvca is correct, ECC and DH exchange will be part of the next OPC UA update, we will think about implementing it afterwards.

jopi2016, 2020-06-18:
Hi,
thanks guys for your feedback.
@ChHaa With "OPC UA update" I suppose you mean the next update of the OPC UA specification, right?

ccvca, 2020-06-18:
@jopi2016 I think you are right, as ECC connection differs from RSA Server and Client needs to support it. This may be covered in the (free) Webinar next Tuesday https://opcfoundation.org/event-details/?ee=257 
>OPC UA Security Update
> Randy Armstrong, Lead OPC UA Security Working Group

For me, the new key exchange looks like DH in https, to pervent later decryption, when the original key is leaked.

This new key generation might be difficult in open62541 as MbedTLS does not support the generation of OPC UA compatible Certificates. (https://github.com/ARMmbed/mbedtls/pull/731 )

jopi2016, 2020-06-23:
Okay, thanks for your feedback.

NoelGraf, 2021-04-30:
@jopi2016  can the issue be closed?