open62541/open62541 #3721: UA_Cient_translateBrowsePathsToNodeIds_async question

m-drozd, 2020-06-18:
## Description

## Background Information / Reproduction Steps

Hi, I'm in the process of migrating my code to use the asynchronous API and I've stumbled into something I can't quite understand.
```
UA_StatusCode __UA_Client_translateBrowsePathsToNodeIds_async(UA_Client *client,
        char *paths[], UA_UInt32 ids[], size_t pathSize,
        UA_ClientAsyncServiceCallback callback, void *userdata,
        UA_UInt32 *reqId) {

    UA_BrowsePath browsePath;
    UA_BrowsePath_init(&browsePath);
    browsePath.startingNode = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    browsePath.relativePath.elements = (UA_RelativePathElement*) UA_Array_new(
            pathSize, &UA_TYPES[UA_TYPES_RELATIVEPATHELEMENT]);
    if (!browsePath.relativePath.elements)
        return UA_STATUSCODE_BADOUTOFMEMORY;
    browsePath.relativePath.elementsSize = pathSize;

    UA_TranslateBrowsePathsToNodeIdsRequest request;
    UA_TranslateBrowsePathsToNodeIdsRequest_init(&request);
    request.browsePaths = &browsePath;
    request.browsePathsSize = 1;

    UA_StatusCode retval = __UA_Client_AsyncService(client, &request,
            &UA_TYPES[UA_TYPES_TRANSLATEBROWSEPATHSTONODEIDSREQUEST], callback,
            &UA_TYPES[UA_TYPES_TRANSLATEBROWSEPATHSTONODEIDSRESPONSE], userdata,
            reqId);
    if (retval != UA_STATUSCODE_GOOD) {
        UA_Array_delete(browsePath.relativePath.elements,
                browsePath.relativePath.elementsSize,
                &UA_TYPES[UA_TYPES_RELATIVEPATHELEMENT]);
        return retval;
    }
    UA_BrowsePath_deleteMembers(&browsePath);
    return retval;
}
Paths and ids arguments seem to be discarded.  Is this function still WIP? The client_async.c example contains usage example for this function.

```

Used CMake options:

```bash
cmake -DUA_NO_AMALGAMATION -DUA_ENABLE_ENCRYPTION ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.0.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-06-18:
Ouch. That slipped through.
I think the API for this one is broken and has to be redone.
Deprecating the feature on v1.1 and doing a clean solution on master.
That is necessary as we won't make breaking changes on v1.1 after the release.

Can you use a workaround directly using `UA_Client_sendAsyncRequest`?

jpfr, 2020-06-18:
See #3723

m-drozd, 2020-06-23:
Yeah I can work around this for now. Thanks.

closed 2020-06-24 by jpfr