open62541/open62541 #3722: Alarms and Condition with MINIMAL or REDUCED NS0

jopi2016, 2020-06-18:
Hi,
I've got a question.
Is it intended in the future to use Alarms and Condition (Part 9) also with a MINIMAL or REDUCED NS0?

If we generate the A&C example (tutorial_server_alarms_conditions) we need to select FULL namespace in cmake. But the full namespace doesn't work on our embedded device as it is too big for it.

Is it possible to adapt the NS0 by ourselves? For us it would be okay to use the ConditionType (i=2782) together with the MINIMAL or REDUCED NS0.

Any advice on how to accomplish this would be highly appreciated.

Thanks in advance
jopi2016

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.1
 - [x ] Other OPC UA SDKs used (client or server): server
 - [x] Operating system: Embedded device
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2020-06-18:
This should be possible by adapting the CMakeLists.txt
I guess, that the interesting part begins here:
https://github.com/open62541/open62541/blob/cedb4527b4fcf96e8dd6eb45701f5062f303b1e0/CMakeLists.txt#L979

PS: If you have success, you may do a pull request with the result.

jopi2016, 2020-06-19:
Hi,
thanks for your answer.

We already tried this.
The problem we have is more from where we get the Nodeset xml-file, I mean the counterpart to „Opc.Ua.NodeSet2.EventsMinimal.xml“.

ccvca, 2020-06-19:
I would guess that they are hand crafted from these:
https://github.com/OPCFoundation/UA-Nodeset/tree/v1.04/Schema
(https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/Schema/Opc.Ua.NodeSet2.Part9.xml are all the AC nodes)

masongk, 2020-06-26:
You can try as mentioned in this issue #3285  and remove specific nodes from the generated nodeset.

jopi2016, 2020-06-27:
thank you @masongk, I'll give it a try.

jopi2016, 2020-07-10:
@masongk your suggestion helped us very much. Thanks. We could solve the issue.

closed 2020-07-10 by jopi2016