open62541/open62541 #3729: example   ./access_control_client_encrypt  failed

paopaol, 2020-06-21:
## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake .. -DUA_BUILD_EXAMPLES=ON -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_ENCRYPTION=ON
```

```bash
./access_control_client_encrypt   opc.tcp://172.16.158.132:5555   ./ser  ~/workspace/nsea/nseabridge/build/nseabridge/extra_sources/opcuaclient/pravate.der 
```
### server infomation

[image]

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):git latest
 - [x] Other OPC UA SDKs used (client or server): client
 - [x] Operating system:linux
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
   [2020-06-21 13:49:45.696 (UTC+0800)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-06-21 13:49:45.697 (UTC+0800)] info/securitypolicy        The Basic128Rsa15 security policy with openssl is added.
[2020-06-21 13:49:45.700 (UTC+0800)] warn/userland      Could not add SecurityPolicy#Basic128Rsa15 with error code BadInternalError
[2020-06-21 13:49:45.700 (UTC+0800)] info/securitypolicy        The basic256 security policy with openssl is added.
[2020-06-21 13:49:45.700 (UTC+0800)] warn/userland      Could not add SecurityPolicy#Basic256 with error code BadInternalError
[2020-06-21 13:49:45.700 (UTC+0800)] info/securitypolicy        The basic256sha256 security policy with openssl is added.
[2020-06-21 13:49:45.700 (UTC+0800)] warn/userland      Could not add SecurityPolicy#Basic256Sha256 with error code BadInternalError
[2020-06-21 13:49:45.700 (UTC+0800)] info/securitypolicy        The Aes128Sha256RsaOaep security policy with openssl is added.
[2020-06-21 13:49:45.700 (UTC+0800)] warn/userland      Could not add SecurityPolicy#Aes128Sha256RsaOaep with error code BadInternalError
[2020-06-21 13:49:45.700 (UTC+0800)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2020-06-21 13:49:45.705 (UTC+0800)] info/channel       Connection 3 | SecureChannel 2524525468 | Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2020-06-21 13:49:45.705 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2020-06-21 13:49:45.725 (UTC+0800)] info/client        Rejecting endpoint 0: security mode doesn't match
[2020-06-21 13:49:45.725 (UTC+0800)] info/client        Rejecting endpoint 1: security policy not available
[2020-06-21 13:49:45.725 (UTC+0800)] info/client        Rejecting endpoint 2: security policy not available
[2020-06-21 13:49:45.725 (UTC+0800)] error/client       No suitable endpoint found
[2020-06-21 13:49:45.725 (UTC+0800)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadInternalError

jopi2016, 2020-06-23:
Hi,
how did you generate your public and private certificate?
of which type are they (SHA1, SHA256 etc.)?

Can you connect to the open62541 demo encryption server that also comes in the examples?

opcua-tsn-team-kalycito, 2020-07-21:
Can you to provide details on the server application against which the client application was run.

Apart from that use the following command to see the usage details of access_control_client_encrypt application:
`./access_control_client_encrypt`
When we run the above command, we get a print which shows the usage details of access_control_client_encrypt   application:
Arguments are missing. The required arguments are <opc.tcp://host:port> <client-certificate.der> <client-private-key.der> [<trustlist1.crl>, ...]

Well if you are looking for an example for secure client server communication, you can run the server_encryption and client_encryption examples available in the stack.

Follow the below given steps to run the server_encryption and client_encryption examples available in the stack:
•Traverse to “open62541/tools/certs”
•Run the following commands to create the certificates for client and server application
    `python create_self-signed.py -u urn:open62541.server.application -c server ../../build/bin/examples`
    `python create_self-signed.py -u urn:unconfigured:application -c client ../../build/bin/examples`
•Traverse to “open62541/build/bin/examples”
•Run the server application
    `./server_encryption server_cert.der server_key.der client_cert.der`
•Run the client application
    Use message signing to verify if the client can be trusted
          `./client_connect -username user1 -password password -cert client_cert.der -key client_key.der -securityMode 2 -securityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 opc.tcp://localhost:4840`
    Use encryption to keep the client/server communication confidential
          `./client_connect -username user1 -password password -cert client_cert.der -key client_key.der -securityMode 3 -securityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 opc.tcp://localhost:4840`

If you are looking for more hands-on information, you can also check out these resources:
•	Free documentation: You can look at the open source documentation page: https://open62541.org/doc/current/
•	Paid online course: Practical introduction to OPC UA – code walk-through and examples in this course use the open62541 stack. It also includes examples related to security implementation: https://opcfoundation.org/products/view/practical-introduction-to-opc-ua-part-i

NoelGraf, 2021-04-30:
@paopaol what's the status?

EvgKlin, 2021-06-28:
@opcua-tsn-team-kalycito you write •Run the server application
./server_encryption server_cert.der server_key.der client_cert.der

Where do you take client_cert.der ??

EvgKlin, 2021-06-28:
If i start like you said,  i get statuscode 2147614720
after ```
UA_ServerConfig_setDefaultWithSecurityPolicies(config, 4840,
&certificate, &privateKey,
trustList, trustListSize,
issuerList, issuerListSize,
revocationList, revocationListSize);

o6-nbeier, 2026-10-21:
Closing for no activity.

closed 2026-10-21 by o6-nbeier