open62541/open62541 #3731: Channel is closed if a read request is processed whilst a secure channel is being renewed

smitha-cgi, 2020-06-22:
I have an application which has connected to a Prosys simulation server using SignAndEncrypt/Basic256Sha256. A timer thread is firing every 2s and calling UA_Client_readValueAttribute to read a double from a Prosys Simulation Server. If a read request goes out in the middle of a renewal the channel is closed due to an unknown SecurityToken being received. 

Here is what a successful read looks like:

```
17:45:49.281 : General : Scan interval (2s) expired
17:45:49.281 : Lib : Connection 33652 | SecureChannel 44 | Sending request with RequestId 2266 of type ReadRequest
17:45:49.286 : Lib : Connection 33652 | SecureChannel 44 | Verifying chunk signature
17:45:49.286 : Lib : Connection 33652 | SecureChannel 44 | Calculated padding size to be 2
17:45:49.286 : Lib : Decode a message of type ReadResponse
17:45:49.286 : General : Received 27.00
```

Here is what a successful channel renewal looks like:

```
17:45:52.990 : Lib : Connection 33652 | SecureChannel 44 | Generating new local nonce
17:45:52.990 : Lib : Connection 33652 | SecureChannel 44 | Requesting to renew the SecureChannel
17:45:52.990 : Lib : Requesting to open a SecureChannel
17:45:53.020 : Lib : OPN message sent
17:45:53.226 : Lib : Connection 33652 | SecureChannel 44 | Verifying chunk signature
17:45:53.228 : Lib : Connection 33652 | SecureChannel 44 | Calculated padding size to be 76
17:45:53.228 : Lib : Connection 33652 | SecureChannel 44 | Generating SecureChannel keys
17:45:53.228 : Lib : Connection 33652 | SecureChannel 44 | Generating new local keys
17:45:53.229 : Lib : Connection 33652 | SecureChannel 44 | Generating new remote keys
17:45:53.229 : Lib : Connection 33652 | SecureChannel 44 | SecureChannel renewed
```

Here is what happens when the read timer expires in the middle of the secure channel renewal:

```
20:46:03.140 : Lib : Connection 33652 | SecureChannel 44 | Generating new local nonce
20:46:03.140 : Lib : Connection 33652 | SecureChannel 44 | Requesting to renew the SecureChannel
20:46:03.140 : Lib : Requesting to open a SecureChannel
20:46:03.210 : Lib : OPN message sent
20:46:03.310 : General : Scan interval (2s) expired
20:46:03.310 : Lib : Connection 33652 | SecureChannel 44 | Sending request with RequestId 7698 of type ReadRequest
20:46:03.488 : Lib : Connection 33652 | SecureChannel 44 | Verifying chunk signature
20:46:03.490 : Lib : Connection 33652 | SecureChannel 44 | Calculated padding size to be 76
20:46:03.490 : Lib : Connection 33652 | SecureChannel 44 | Generating SecureChannel keys
20:46:03.490 : Lib : Connection 33652 | SecureChannel 44 | Generating new local keys
20:46:03.490 : Lib : Connection 33652 | SecureChannel 44 | Generating new remote keys
20:46:03.491 : Lib : Connection 33652 | SecureChannel 44 | SecureChannel renewed
20:46:03.492 : Lib : Connection 33652 | SecureChannel 44 | Received an unknown SecurityToken
20:46:03.492 : Lib : Connection 33652 | SecureChannel 44 | Receiving the response failed with StatusCode BadSecureChannelTokenUnknown
20:46:03.493 : Lib : The basic256sha256 security policy channel with openssl is deleted.
20:46:03.493 : Lib : Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good
```

Used CMake options:

```
UA_ENABLE_DA
UA_ENABLE_DISCOVERY
UA_ENABLE_ENCRYPTION
UA_ENABLE_ENCRYPTION_OPENSSL
UA_ENABLE_METHODCALLS
UA_ENABLE_NODEMANAGEMENT
UA_ENABLE_PARSING
UA_ENABLE_SUBSCRIPTIONS
UA_LOGLEVEL 100
```

## Checklist

Please provide the following information:

 - [x ] open62541 Version (release number or git tag): 1.1
 - [x ] Other OPC UA SDKs used (client or server): Prosys OPC UA Simulation Server 4.0.2-108
 - [x ] Operating system: Windows 7
 - [x ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2020-06-30:
This might be fixed by https://github.com/open62541/open62541/pull/3742

jpfr, 2020-07-01:
Yes. Should be solved.
Please verify.

jpfr, 2020-07-07:
Please reopen if problems remain.

closed 2020-07-07 by jpfr