open62541/open62541 #3733: Variable nodes with valueRank 0 and an array value

genuaboro, 2020-06-22:
## Description
I can not add a variable node with an array value and a value rank of 0 (UA_VALUERANK_ONE_OR_MORE_DIMENSIONS).

## Background Information / Reproduction Steps
Here is some modified tutorial example code that illustrates the issue:
```c
static void
addVariable0(UA_Server *server) {
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.displayName = UA_LOCALIZEDTEXT("en-US", "Array ValueRank 0");
    attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;

    /* Set the variable value constraints */
    attr.dataType = UA_TYPES[UA_TYPES_DOUBLE].typeId;
    attr.valueRank = UA_VALUERANK_ONE_OR_MORE_DIMENSIONS;

    /* Set the value */
    UA_UInt32 arrayDims[1] = {2};
    UA_Double zero[2] = {0.0, 0.0};
    UA_Variant_setArray(&attr.value, zero, 2, &UA_TYPES[UA_TYPES_DOUBLE]);
    attr.value.arrayDimensions = arrayDims;
    attr.value.arrayDimensionsSize = 1;

    UA_NodeId myIntegerNodeId = UA_NODEID_STRING(1, "array0");
    UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME(1, "array0");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_Server_addVariableNode(server, myIntegerNodeId, parentNodeId,
                              parentReferenceNodeId, myIntegerName,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
                              attr, NULL, NULL);
}

static void
addVariable1(UA_Server *server) {
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.displayName = UA_LOCALIZEDTEXT("en-US", "Array ValueRank 1");
    attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;

    /* Set the variable value constraints */
    attr.dataType = UA_TYPES[UA_TYPES_DOUBLE].typeId;
    UA_UInt32 arrayDimsAttr[1] = {0};
    attr.arrayDimensions = arrayDimsAttr;
    attr.arrayDimensionsSize = 1;
    attr.valueRank = UA_VALUERANK_ONE_DIMENSION;

    /* Set the value */
    UA_UInt32 arrayDims[1] = {2};
    UA_Double zero[2] = {0.0, 0.0};
    UA_Variant_setArray(&attr.value, zero, 2, &UA_TYPES[UA_TYPES_DOUBLE]);
    attr.value.arrayDimensions = arrayDims;
    attr.value.arrayDimensionsSize = 1;

    UA_NodeId myIntegerNodeId = UA_NODEID_STRING(1, "array1");
    UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME(1, "array1");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_Server_addVariableNode(server, myIntegerNodeId, parentNodeId,
                              parentReferenceNodeId, myIntegerName,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
                              attr, NULL, NULL);
}
```

And here is the corresponding log:
```
[2020-06-22 13:51:20.196 (UTC+0200)] debug/session	Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | Read the attribute 2
[2020-06-22 13:51:20.196 (UTC+0200)] warn/server	Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2020-06-22 13:51:20.196 (UTC+0200)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-06-22 13:51:20.196 (UTC+0200)] info/session	Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | WriteRequest returned status code BadTypeMismatch
[2020-06-22 13:51:20.196 (UTC+0200)] info/session	Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: The value of ns=1;s=array0 is incompatible with the variable definition
[2020-06-22 13:51:20.196 (UTC+0200)] info/session	Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Type-checking the variable node ns=1;s=array0 failed with error code BadTypeMismatch
[2020-06-22 13:51:20.196 (UTC+0200)] info/network	TCP network layer listening on opc.tcp://obsd.my.domain:4840/
[2020-06-22 13:51:24.447 (UTC+0200)] info/server	received ctrl-c
[2020-06-22 13:51:24.447 (UTC+0200)] debug/network	Socket select failed with Interrupted system call
[2020-06-22 13:51:24.447 (UTC+0200)] info/network	Shutting down the TCP network layer
```

addVariable0() and addVariable1() try to add nodes with the same value. The only difference is that the first one has a ValueRank attribute of 0 and nothing set in the ArrayDimensions attribute, while the second one has a ValueRank attribute of 1 and an ArrayDimensions attribute of {0}.

From my understanding both these cases are valid. I found this because an example of FreeOpcUa creates nodes with a ValueRank of 0 (https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/server-example.py - "myarrayvar")

The check that prohibits the creation of the node is found in compatibleValueRankValue() from src/server/ua_services_attribute.c:
```c
static UA_Boolean
compatibleValueRankValue(UA_Int32 valueRank, const UA_Variant *value) {
    /* Invalid ValueRank */
    if(valueRank < UA_VALUERANK_SCALAR_OR_ONE_DIMENSION)
        return false;

    /* Empty arrays (-1) always match */
    if(!value->data)
        return true;

    size_t arrayDims = value->arrayDimensionsSize;
    if(arrayDims == 0 && !UA_Variant_isScalar(value))
        arrayDims = 1; /* array but no arraydimensions -> implicit array dimension 1 */

    /* We cannot simply use compatibleValueRankArrayDimensions since we can have                                                                                                                                   
     * defined ArrayDimensions for the value if the ValueRank is -2 */
    switch(valueRank) {
    case UA_VALUERANK_SCALAR_OR_ONE_DIMENSION: /* The value can be a scalar or a one dimensional array */
        return (arrayDims <= 1);
    case UA_VALUERANK_ANY: /* The value can be a scalar or an array with any number of dimensions */
        return true;
    case UA_VALUERANK_SCALAR: /* The value is a scalar */
        return (arrayDims == 0);
    default:
        break;
    }

    UA_assert(valueRank >= UA_VALUERANK_ONE_OR_MORE_DIMENSIONS);

    /* case 0:  the value is an array with one or more dimensions */
    return (arrayDims == (UA_UInt32)valueRank);
}
```
At the end of the function we have a valueRank of 0 but arrayDims is never 0 if the value is an array (it would be set to 1 before).

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.0, master
 - [x] Other OPC UA SDKs used (client or server): FreeOpcUa has example nodes with such values
 - [x] Operating system: OpenBSD, Linux
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-06-22:
Fix and unit test for the v1.1 branch in #3735.

jpfr, 2020-06-23:
Merged.
Will be pulled on master soon.

closed 2020-06-23 by jpfr