open62541/open62541 #3736: how to modify the minimal server

Feras992, 2020-06-23:
## Description
hello,

is there any possible way to modify the minimal server and build it? i want to add some object types to the minimal server but i want to avoid adding them manually to the code, it will be easier if i add them using the CMake.

i am using Nucleo-F767, FreeRTOS and lwIP and due to the limited ressources i can't use the full server.

i tried to modify the Opc.Ua.NodeSet2.Minimal.xml and after that add the Object Types to datatypes_minimal.txt, however, it doesn't work.

many thanks

jopi2016, 2020-06-23:
Hello,

basically with minimal/reduced/full you define the server namespace, so to say which "features" you can use within the server.

Apart from the server namespace you need to add your own second namespace in which you add your own nodes and via this nodes you can then eg. send internal device values like CPU temperature etc to your client.

So if you don't want to add all your nodes manually into your code you can use an xml-file and a nodeset_compiler that translates your xml-file into C-code.

Hope that helped.

Feras992, 2020-06-23:
@jopi2016 thnx for the replay. 
what i want to do is to use the minimal server with some features from the full server.
in my case, i want to add the FileType Object Type and some others Object Types and Variable Types to the minimal server without using the full server cuz my hardware doesn't have enough ressources.

i don't want to creat my own namespace or somthing like that, i just want to add some Object Types to the minimal server and build it using CMake

jopi2016, 2020-06-24:
Perhaps my issue #3722 might be helpful to you.
I also asked there how to adapt the minimal/reduced namespace. ;-)

Feras992, 2020-06-25:
@jopi2016 i have the same problem and i tried to modify the xml file for the minimal server(i added some Object Types) and built the stack once again but it didn't work.
i hope that some one answer our questions.

jpfr, 2020-06-30:
Hi there,

if the namespace is set to `minimal`, then no xml-based namespace definitions are used.
Only those from `ua_server_ns0.c`.

NoelGraf, 2021-04-30:
@Feras992 what is the status?

NoelGraf, 2021-05-20:
I would close the topic for now. Please reopen if required!

closed 2021-05-20 by NoelGraf