open62541/open62541 #3737: Issue when starting renew secure channel token in version 1.1

jopi2016, 2020-06-23:
Hi,
we noticed that there's an issue when we connect via UAExpert to an open62541 version 1.1 server using encryption.

When after  450000 msec a new secure channel token is generated this fails
OpcUa_SecureStream_VerifyInputBuffer: Could not verify message signature!
OpcUa_SecureStream_AppendInput: could not verify message signature (0xA0010000)!
ReleaseSecurityKeyset: Keyset for token 3 released.
OpcUa_SecureConnection_ProcessSessionCallResponse: Could not append chunk! (0x80130000)

We tried this with different encryption modes (Basic128Rsa15 and Basic256Sha256).

Please see attached screenshot and log of UAExpert
[server.log](https://github.com/open62541/open62541/files/4818490/server.log)
[ua_expert.log](https://github.com/open62541/open62541/files/4818491/ua_expert.log)
[image]

We are running the open62541 server with the server_encryption example.

The problem does not occur always, sometimes the renewing process works succssfully but then the next process fails.

Used CMake options:
cmake -S %CURRENTDIR_OPEN% -B %BUILD_FOLDER% -G "Visual Studio 12 2013" -DUA_ENABLE_ENCRYPTION:BOOL=ON -DUA_ENABLE_AMALGAMATION:BOOL=OFF -DUA_BUILD_SELFSIGNED_CERTIFICATE:BOOL=ON  -DUA_BUILD_EXAMPLES:BOOL=ON -DBUILD_SHARED_LIBS:BOOL=ON -DUA_NAMESPACE_ZERO:STRING=MINIMAL -DMBEDCRYPTO_LIBRARY:FILEPATH="%CURRENTDIR_OPEN%..\..\mbedtls\build\library\Release\mbedcrypto.lib" -DMBEDTLS_INCLUDE_DIRS:PATH="%CURRENTDIR_OPEN%..\..\mbedtls\include" -DMBEDTLS_LIBRARY:FILEPATH="%CURRENTDIR_OPEN%..\..\mbedtls\build\library\Release\mbedtls.lib" -DMBEDX509_LIBRARY:FILEPATH="%CURRENTDIR_OPEN%..\..\mbedtls\build\library\Release\mbedx509.lib"

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.1
 - [x] Other OPC UA SDKs used (client or server): UAExpert
 - [x] Operating system: Windows 10
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jopi2016, 2020-06-23:
Additonal info:
The problem does not occur with version 1.0

jpfr, 2020-06-23:
A fix is underway. Stay tuned.

jpfr, 2020-06-24:
Solved on 1.1.
Can you verify?

jopi2016, 2020-06-25:
I can confirm that the issue is fixed.
Thank you very much for your quick response.

closed 2020-06-25 by jopi2016