open62541/open62541 #3741: Memory leek in server / Segmentation fault

abdo4694, 2020-06-23:
## Description

Hello Open62541 Team. 
I have attached valgrind output for a reference. My main problem is getting a segmentation fault randomly during operation. The segmentation fault occurs spontaneously.  While searching for the error I looked at the valgrind output and saw huge memory leaks. Can you have a look at the problem and let me know if this is a bug.

## Background Information / Reproduction Steps
In a while loop we checked some values after every 50ms, after a certain time I got an segmentation fault, By looking at the debugger, the debugger stopped every time in the read function from the server. The stop was also at some UA_Copy functions. After that I reduced the delay to 1ms, and the problem occurred more frequently. Therefore my speculation was a memory leak. I have attached my project. Please let me know if you need any other information.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 18.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue

[opcua-mock.zip](https://github.com/open62541/open62541/files/4820845/opcua-mock.zip)

jpfr, 2020-06-23:
It seems you do not properly shut down the server when the application is stopped.
Can you run valgrind with a proper shutdown to remove false positives?

abdo4694, 2020-06-24:
Thank you for your response. You are right the shutdown was missing. After adding the shutdown all the false positives where removed. But not we still have a memory leak in a UA_Server_readValue(). I have attached the new code and output of the valgrind. Can you have a look at the code? Is the leak something from the library?
Thank you

[opcua-mock.zip](https://github.com/open62541/open62541/files/4824047/opcua-mock.zip)

ccvca, 2020-06-26:
At least the members of the DataValue needs to be cleared after processing them in `double OpcUA_Server::read_double_from_node`. (Copy your double before deleting the members)

abdo4694, 2020-06-29:
Thank you very much for your help :) 
I deleted  "UA_Variant value"  by using UA_Variant_deleteMembers(&value); in case of reading from the client.
Additionally in the function create_root_objects() I deleted:
1) UA_NodeId with UA_NodeId_deleteMembers(&root_object_id);
2) UA_ObjectAttributes with UA_ObjectAttributes_deleteMembers(&oAttr);

jpfr, 2020-06-29:
Can we mark this as solved then?

abdo4694, 2020-06-30:
Yes

closed 2020-06-30 by abdo4694