open62541/open62541 #3744: Client: BadConnectionClosed when acknowlege an alarm & condition

kikia1b2, 2020-06-24:
when acknowleged an alarm, following error will be printed out:

[2020-06-24 18:22:46.656 (UTC+0800)] info/client	Error receiving the response with status code BadSecurityChecksFailed

[2020-06-24 18:22:51.644 (UTC+0800)] info/client	Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good
[2020-06-24 18:22:51.644 (UTC+0800)] info/userland	ns:0, 6006; set acknowledged failed with BadConnectionClosed
[2020-06-24 18:22:51.645 (UTC+0800)] info/userland	ns:0, 6005; set acknowledged failed with BadConnectionClosed
[2020-06-24 18:22:51.645 (UTC+0800)] info/userland	ns:0, 6004; set acknowledged failed with BadConnectionClosed
[2020-06-24 18:22:51.648 (UTC+0800)] info/userland	ns:0, 6003; set acknowledged failed with BadConnectionClosed
[2020-06-24 18:22:51.648 (UTC+0800)] info/userland	ns:0, 6002; set acknowledged failed with BadConnectionClosed
[2020-06-24 18:22:51.676 (UTC+0800)] info/userland	ns:61453, -1163005939; set acknowledged failed with BadConnectionClosed
[2020-06-24 18:22:51.685 (UTC+0800)] info/channel	Connection 664 | SecureChannel 3 | Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2020-06-24 18:22:51.685 (UTC+0800)] info/client	Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2020-06-24 18:22:51.702 (UTC+0800)] info/client	Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:  Window 7
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2020-06-26:
Could you please provide some more information:
What kind of Client/Server in which version are you using?

jpfr, 2020-06-26:
Which version of the SDK do you use?
Please try again with the latest v1.1.1.

kikia1b2, 2020-06-28:
The problem has been solved. The acknowledge method should be called at the same thread with the handler_events() func.

closed 2020-06-28 by kikia1b2