open62541/open62541 #3745: Unable to register Milo server with open62541 LDS.

jdhanasekar, 2020-06-25:
Hi open62541,

I am working on to registering my ```Milo server``` with ```open62541 LDS```. After registering, open62541 LDS showing ```Processing the message failed with error BadConnectionClosed ```. 
In my Milo IDE, I am not  observing any error in ```RegisterServer2()``` API call.
Through ``` FindServersOnNetwork()``` API function call, I have find out LDS server on my network where I got the details  like this
 
```ServerOnNetwork{RecordId=0, ServerName=LDS-quanta, DiscoveryUrl=opc.tcp://quanta:4840, ServerCapabilities=[LDS]}```

When I access this open62541 LDS```opc.tcp://quanta:4840``` in UaExpert, I am getting error like this,

[image]

I am assuming because of this ```BadConnectionClosed ``` error, I am getting above error.
Please help me to fix ```BadConnectionClosed``` error.  I am working on to find why it is happening. 
As I am new to open62541 and opc ua, I am making mistakes. Please bear with me. 
Please correct me, If I am wrong. 

**STEPS FOLLOWED:**

1. sudo cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=MinSizeRel -DUA_NAMESPACE_ZERO=MINIMAL -DUA_ENABLE_AMALGAMATION=OFF -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON ..
2. sudo make
3. sudo make install
4. sudo gcc -std=c99 -o lds_me ../examples/discovery/server_lds.c -lopen62541

**OUTPUT :**
```
[2020-06-25 14:14:58.150 (UTC+0550)] info/network       Connection 8 | New connection over TCP from 10.1.60.208
[2020-06-25 14:14:58.182 (UTC+0550)] info/channel       Creating a new SecureChannel
[2020-06-25 14:14:58.221 (UTC+0550)] info/channel       Connection 8 | SecureChannel 1 | Opened SecureChannel
[2020-06-25 14:14:58.265 (UTC+0550)] info/channel       Connection 8 | SecureChannel 1 | CloseSecureChannel
[2020-06-25 14:14:58.265 (UTC+0550)] info/network       Connection 8 | Processing the message failed with error BadConnectionClosed
[2020-06-25 14:14:58.265 (UTC+0550)] info/network       Connection 8 | Closed
[2020-06-25 14:14:58.543 (UTC+0550)] info/network       Connection 8 | New connection over TCP from 10.1.60.208
[2020-06-25 14:14:58.544 (UTC+0550)] info/channel       Creating a new SecureChannel
[2020-06-25 14:14:58.546 (UTC+0550)] info/channel       Connection 8 | SecureChannel 2 | Opened SecureChannel
[2020-06-25 14:14:58.557 (UTC+0550)] info/channel       Connection 8 | SecureChannel 2 | Session 9f6292b1-f023-14a6-8868-96999829b28b created
[2020-06-25 14:14:58.566 (UTC+0550)] info/session       Connection 8 | SecureChannel 2 | Session ns=1;g=9f6292b1-f023-14a6-8868-96999829b28b | ActivateSession: Session activated
[2020-06-25 14:14:58.617 (UTC+0550)] info/server        Multicast DNS: add record for domain: opc.tcp://localhost:4840-10-1-60-208._opcua-tcp._tcp.local.
[2020-06-25 14:14:58.656 (UTC+0550)] info/session       Connection 8 | SecureChannel 2 | Session ns=1;g=9f6292b1-f023-14a6-8868-96999829b28b | CloseSession
[2020-06-25 14:14:58.658 (UTC+0550)] info/channel       Connection 8 | SecureChannel 2 | CloseSecureChannel
[2020-06-25 14:14:58.658 (UTC+0550)] info/network       Connection 8 | Processing the message failed with error BadConnectionClosed
[2020-06-25 14:14:58.658 (UTC+0550)] info/network       Connection 8 | Closed
```

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [*] open62541 Version (release number or git tag):
   open62541 version = v1.1-29-g6dc50720

 - [*] Other OPC UA SDKs used (client or server): 
     ```server_lds.c```
 - [*] Operating system:
          LINUX
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-06-25:
@Pro Any ideas?

Pro, 2020-06-25:
Works fine for me.

Tested with current 1.1 branch (320d71b77564964379365e1390d1e372964388d7)

```
cd open62541
mkdir build && cd build
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=MinSizeRel -DUA_NAMESPACE_ZERO=MINIMAL -DUA_ENABLE_AMALGAMATION=OFF -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON -DUA_BUILD_EXAMPLES=ON ..
make -j
./bin/examples/discovery_server_lds
```

And in eclipse milo, Version 0.4.1 used the following code:

```java
/*
 * Copyright (c) 2019 the Eclipse Milo Authors
 *
 * This program and the accompanying materials are made
 * available under the terms of the Eclipse Public License 2.0
 * which is available at https://www.eclipse.org/legal/epl-2.0/
 *
 * SPDX-License-Identifier: EPL-2.0
 */

package org.eclipse.milo.examples.client;

import java.util.List;
import java.util.concurrent.CompletableFuture;
import java.util.concurrent.ExecutionException;

import org.eclipse.milo.opcua.sdk.client.OpcUaClient;
import org.eclipse.milo.opcua.stack.core.Identifiers;
import org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject;
import org.eclipse.milo.opcua.stack.core.types.builtin.LocalizedText;
import org.eclipse.milo.opcua.stack.core.types.builtin.NodeId;
import org.eclipse.milo.opcua.stack.core.types.builtin.StatusCode;
import org.eclipse.milo.opcua.stack.core.types.enumerated.ApplicationType;
import org.eclipse.milo.opcua.stack.core.types.enumerated.BrowseDirection;
import org.eclipse.milo.opcua.stack.core.types.enumerated.BrowseResultMask;
import org.eclipse.milo.opcua.stack.core.types.enumerated.NodeClass;
import org.eclipse.milo.opcua.stack.core.types.structured.*;
import org.slf4j.Logger;
import org.slf4j.LoggerFactory;

import static org.eclipse.milo.opcua.stack.core.types.builtin.unsigned.Unsigned.uint;
import static org.eclipse.milo.opcua.stack.core.util.ConversionUtil.toList;

public class BrowseExample implements ClientExample {

    public static void main(String[] args) throws Exception {
        BrowseExample example = new BrowseExample();

        new ClientExampleRunner(example, false).run();
    }

    @Override
    public String getEndpointUrl() {
        return "opc.tcp://localhost:4840";
    }

    private final Logger logger = LoggerFactory.getLogger(getClass());

    @Override
    public void run(OpcUaClient client, CompletableFuture<OpcUaClient> future) throws Exception {
        // synchronous connect
        client.connect().get();

        MdnsDiscoveryConfiguration myDiscoveryConfig = new  MdnsDiscoveryConfiguration("test.name", new String[]{"NA"});

[... 51 lines cut ...]
```
[2020-06-25 12:24:46.435 (UTC+0200)] info/network	Connection 6 | New connection over TCP from 127.0.0.1
[2020-06-25 12:24:46.452 (UTC+0200)] info/channel	Creating a new SecureChannel
[2020-06-25 12:24:46.473 (UTC+0200)] info/channel	Connection 6 | SecureChannel 1 | Opened SecureChannel
[2020-06-25 12:24:46.494 (UTC+0200)] info/channel	Connection 6 | SecureChannel 1 | CloseSecureChannel
[2020-06-25 12:24:46.494 (UTC+0200)] info/network	Connection 6 | Processing the message failed with error BadConnectionClosed
[2020-06-25 12:24:46.494 (UTC+0200)] info/network	Connection 6 | Closed
[2020-06-25 12:24:46.602 (UTC+0200)] info/network	Connection 6 | New connection over TCP from 127.0.0.1
[2020-06-25 12:24:46.603 (UTC+0200)] info/channel	Creating a new SecureChannel
[2020-06-25 12:24:46.604 (UTC+0200)] info/channel	Connection 6 | SecureChannel 2 | Opened SecureChannel
[2020-06-25 12:24:46.607 (UTC+0200)] info/channel	Connection 6 | SecureChannel 2 | Session ce7e1393-aca2-5f24-c68c-48b4c0fcff0f created
[2020-06-25 12:24:46.611 (UTC+0200)] info/session	Connection 6 | SecureChannel 2 | Session ns=1;g=ce7e1393-aca2-5f24-c68c-48b4c0fcff0f | ActivateSession: Session activated
[2020-06-25 12:24:47.097 (UTC+0200)] info/server	Multicast DNS: add record for domain: test.name-localhost._opcua-tcp._tcp.local.
[2020-06-25 12:24:47.100 (UTC+0200)] info/session	Connection 6 | SecureChannel 2 | Session ns=1;g=ce7e1393-aca2-5f24-c68c-48b4c0fcff0f | CloseSession
[2020-06-25 12:24:47.101 (UTC+0200)] info/channel	Connection 6 | SecureChannel 2 | CloseSecureChannel
[2020-06-25 12:24:47.101 (UTC+0200)] info/network	Connection 6 | Processing the message failed with error BadConnectionClosed
[2020-06-25 12:24:47.101 (UTC+0200)] info/network	Connection 6 | Closed
^C[2020-06-25 12:25:23.345 (UTC+0200)] info/network	Shutting down the TCP network layer

```

If you then open UAExpert, it will show you the registered client. The UaExpert error messages comes from the fact that you are probably passing a wrong endpoint url to the RegisterServer2 request and therefore UAExpert is trying to connect to that one.

Note that the LDS example server is not visible in UaExpert:
https://github.com/open62541/open62541/blob/master/examples/discovery/server_multicast.c#L258

jdhanasekar, 2020-06-25:
Pro,

Thanks for the details. 
I will follow your steps and implementation as well. 
I will Let you know the result.

jdhanasekar, 2020-06-25:
After correcting my endpoint url (in RegisterServer2() call), I can access open62541 LDS server in UaExpert. But I am not seeing the details of registered server. 
[image]

My registration code will look like this,

```
/*
 * Copyright (c) 2019 the Eclipse Milo Authors
 *
 * This program and the accompanying materials are made
 * available under the terms of the Eclipse Public License 2.0
 * which is available at https://www.eclipse.org/legal/epl-2.0/
 *
 * SPDX-License-Identifier: EPL-2.0
 */

package org.eclipse.milo.examples.client;

import java.util.List;
import java.util.concurrent.CompletableFuture;
import java.util.concurrent.Semaphore;

import com.google.common.collect.ImmutableList;
import org.eclipse.milo.opcua.sdk.client.OpcUaClient;
import org.eclipse.milo.opcua.sdk.client.api.nodes.VariableNode;
import org.eclipse.milo.opcua.stack.client.UaStackClient;
import org.eclipse.milo.opcua.stack.core.Identifiers;
import org.eclipse.milo.opcua.stack.core.Stack;
import org.eclipse.milo.opcua.stack.core.serialization.SerializationContext;
import org.eclipse.milo.opcua.stack.core.serialization.UaStructure;
import org.eclipse.milo.opcua.stack.core.types.builtin.DataValue;
import org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject;
import org.eclipse.milo.opcua.stack.core.types.builtin.LocalizedText;
import org.eclipse.milo.opcua.stack.core.types.builtin.NodeId;
import org.eclipse.milo.opcua.stack.core.types.enumerated.ApplicationType;
import org.eclipse.milo.opcua.stack.core.types.enumerated.ServerState;
import org.eclipse.milo.opcua.stack.core.types.enumerated.TimestampsToReturn;
import org.eclipse.milo.opcua.stack.core.types.structured.*;
import org.slf4j.Logger;
import org.slf4j.LoggerFactory;

import static ch.qos.logback.core.joran.action.ActionConst.NULL;

public class ReadExample implements ClientExample {

    public static void main(String[] args) throws Exception {
        ReadExample example = new ReadExample();

        new ClientExampleRunner(example, false).run();

    }

    private final Logger logger = LoggerFactory.getLogger(getClass());

    @Override
    public void run(OpcUaClient client, CompletableFuture<OpcUaClient> future) throws Exception {

        //String ServerURI ="urn:eclipse:milo:examples:client";
        String ServerURI ="urn:eclipse:milo:examples:server";
        String ProductURI = "https://opcfoundation.org/";
        LocalizedText[] myServerName = new LocalizedText[]{LocalizedText.english("eclipse milo opc-ua server")};
        ApplicationType ServerType =  ApplicationType.Server;
        String[] myDiscoveryURLs =  new String[]{ "opc.tcp://localhost.localdomain:12686/milo"};
        Boolean IsOnline = Boolean.TRUE;

        UaStackClient stackClient = client.getStackClient();

        MdnsDiscoveryConfiguration mymDNS = new MdnsDiscoveryConfiguration("opc.tcp://localhost:4840",new String[] {"NA,LDS"});

        SerializationContext mySerial =  stackClient.getSerializationContext();
[... 58 lines cut ...]
        });
    }

    private CompletableFuture<List<DataValue>> readServerStateAndTime(OpcUaClient client) {
        List<NodeId> nodeIds = ImmutableList.of(
            Identifiers.Server_ServerStatus_State,
            Identifiers.Server_ServerStatus_CurrentTime);

        Logger logger = LoggerFactory.getLogger(ReadExample.class);
        logger.info("dhanasekar @ReadExample nodeIds : {}", nodeIds);

        //dhanasekar - This call defaults to reading the “value” attribute

        return client.readValues(0.0, TimestampsToReturn.Both, nodeIds);
    }

    @Override
   public String getEndpointUrl() {
        
        return  "opc.tcp://quanta:4840/"; //registering with LDS server
        
    }

}
````

Pro, 2020-06-25:
The UA Expert screenshot does have nothing to do with discovery. 
In UAExpert you are connecting to the (Milo?) server which does not have the Product Info set.

As mentioned previously, the LDS will not show up in UAExpert.
I strongly suggest to play around with ALL the examples here and make sure that you understand what is happening there:
https://github.com/open62541/open62541/tree/master/examples/discovery

Then, continue with the Milo implementation and check e.g., if the provided client in our examples is listing the milo server.

I will close this issue, as the initial question is solved. Feel free to open another issue if there are other questions.

closed 2020-06-25 by Pro