open62541/open62541 #3747: Why does ua_generate_nodeset_and_datatypes requires a .bsd file when a .csv file is passed?

ccvca, 2020-06-26:
## Description
Why does ua_generate_nodeset_and_datatypes requires a .bsd File when a .csv file is passed?
Internally the .bsd is passed to ua_generate_datatypes and the .csv to ua_generate_nodeid_header, so there is no internal call that requires both.

This problem occures with namespace that does not contain any data type definitions (e.g. enumerations) as no .bsd generated.

## Background Information / Reproduction Steps
Create only object/variable types in a separate namespace using a modelling tool and export xml and csv.

Try to use ua_generate_nodeset_and_datatypes will result in 
https://github.com/open62541/open62541/blob/2d6915d1928e7243979cd47b629b78dabf9455ab/tools/cmake/macros_public.cmake#L476

```cmake
 ua_generate_nodeset_and_datatypes(
     NAME "machinery"
     FILE_NS "${PROJECT_SOURCE_DIR}/model/machinery.xml"
     FILE_CSV "${PROJECT_SOURCE_DIR}/model/machinery.csv"
#     FILE_BSD "${PROJECT_SOURCE_DIR}/model/machinery.bsd" #<-- Not available
     OUTPUT_DIR "${PROJECT_SOURCE_DIR}/src_generated"
     NAMESPACE_IDX 3
     DEPENDS_TYPES "UA_TYPES_DI"
 )
```

The following workaround is possible but requires more code and some internal knowledge to prevent breaking subsequent calls of ua_generate_nodeset_and_datatypes:

```cmake
# Workaround/Hack, as ua_generate_nodeset_and_datatypes expects a types-File.
file(TOUCH "${PROJECT_SOURCE_DIR}/src_generated/types_machinery_generated.h")

ua_generate_nodeid_header(
    NAME "machinery"
    FILE_CSV "${PROJECT_SOURCE_DIR}/model/machinery.csv"
    OUTPUT_DIR "${PROJECT_SOURCE_DIR}/src_generated"
    TARGET_SUFFIX "ids-machinery"
    ID_PREFIX "MACHINERY_"
)

ua_generate_nodeset(
    NAME "machinery"
    FILE "${PROJECT_SOURCE_DIR}/model/machinery.xml"
    OUTPUT_DIR "${PROJECT_SOURCE_DIR}/src_generated"
    # First NS0 Types, then other dependent namespaces
    DEPENDS_TYPES "UA_TYPES;UA_TYPES_DI"
    DEPENDS_NS "${open62541_NODESET_DIR}/Schema/Opc.Ua.NodeSet2.xml;${open62541_NODESET_DIR}/DI/Opc.Ua.Di.NodeSet2.xml"
)

```

I would suggest to add this workaround in ua_generate_nodeset_and_datatypes when no bsd file is passed.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master (2d6915d1928e7243979cd47b629b78dabf9455ab)
 - [x] Other OPC UA SDKs used (client or server): UaModeler
 - [x] Operating system: Win 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2020-08-31:
This is explained in my comment here:
https://github.com/open62541/open62541/issues/2784#issuecomment-499750909

NoelGraf, 2021-04-30:
@ccvca can the issue be closed?

ccvca, 2021-05-17:
I think it's still not resolved. I known, that the bsd-Files are not exported from the nodeset.

But I still don't get the reason, why it is not possible to pass an nodeset and a CSV file (no bsd required and available) to ua_generate_nodeset_and_datatypes to get the namespace and the definitions for the Nodes.
Fine:
```
 ua_generate_nodeset_and_datatypes(
     NAME "machinery"
     FILE_NS "${PROJECT_SOURCE_DIR}/model/any.xml"
     FILE_CSV "${PROJECT_SOURCE_DIR}/model/any.csv"
     FILE_BSD "${PROJECT_SOURCE_DIR}/model/any.bsd"
     OUTPUT_DIR "${PROJECT_SOURCE_DIR}/src_generated"
     NAMESPACE_IDX 3
     DEPENDS_TYPES "UA_TYPES_DI"
 )

 ua_generate_nodeset_and_datatypes(
     NAME "machinery"
     FILE_NS "${PROJECT_SOURCE_DIR}/model/any.xml"
#     FILE_CSV "${PROJECT_SOURCE_DIR}/model/any.csv"
#     FILE_BSD "${PROJECT_SOURCE_DIR}/model/any.bsd"
     OUTPUT_DIR "${PROJECT_SOURCE_DIR}/src_generated"
     NAMESPACE_IDX 3
     DEPENDS_TYPES "UA_TYPES_DI"
 )
```

Not fine:
```
 ua_generate_nodeset_and_datatypes(
     NAME "machinery"
     FILE_NS "${PROJECT_SOURCE_DIR}/model/any.xml"
     FILE_CSV "${PROJECT_SOURCE_DIR}/model/any.csv"
#     FILE_BSD "${PROJECT_SOURCE_DIR}/model/any.bsd"
     OUTPUT_DIR "${PROJECT_SOURCE_DIR}/src_generated"
     NAMESPACE_IDX 3
     DEPENDS_TYPES "UA_TYPES_DI"
 )
```