open62541/open62541 #3755: Possible free of a non-heap object

insensiv-ah, 2020-07-01:
## Description

While following the tutorial to learn the lib I ran into the following problem: The examples cannot be compiled as a release build while building open62541 as a static library as the compiler/linker detects a call to free of a stack variable:

```
...
[ 55%] Building C object examples/CMakeFiles/access_control_client.dir/access_control/client_access_control.c.o
[ 55%] Linking C executable ../bin/examples/access_control_client
In function ‘__Subscriptions_DeleteData_free’,
    inlined from ‘__Subscriptions_delete_handler.constprop’ at /home/ahorst/opc-ua/open62541/src/client/ua_client_subscriptions.c:292:9,
    inlined from ‘UA_Client_Subscriptions_delete.constprop’ at /home/ahorst/opc-ua/open62541/src/client/ua_client_subscriptions.c:361:5,
    inlined from ‘UA_Client_Subscriptions_deleteSingle’ at /home/ahorst/opc-ua/open62541/src/client/ua_client_subscriptions.c:372:36:
/home/ahorst/opc-ua/open62541/src/client/ua_client_subscriptions.c:230:5: error: attempt to free a non-heap object ‘data’ [-Werror=free-nonheap-object]
     UA_free(data);
     ^
In function ‘__Subscriptions_delete_handler.constprop’,
    inlined from ‘UA_Client_Subscriptions_delete.constprop’ at /home/ahorst/opc-ua/open62541/src/client/ua_client_subscriptions.c:361:5,
    inlined from ‘UA_Client_Subscriptions_deleteSingle’ at /home/ahorst/opc-ua/open62541/src/client/ua_client_subscriptions.c:372:36:
/home/ahorst/opc-ua/open62541/src/client/ua_client_subscriptions.c:293:9: error: attempt to free a non-heap object ‘cc’ [-Werror=free-nonheap-object]
         UA_free(cc);
         ^
lto1: all warnings being treated as errors
lto-wrapper: fatal error: /usr/bin/cc returned 1 exit status
compilation terminated.
...
```

When I build the library as a shared library the build process succeeds.  
My guess is that due to less opportunities to optimize and inline code the compiler isn't able to detect this potential error.  
As far as I understand the linker error this isn't a problem of the examples themselves but linking in the open62541 library as a static lib and using the problematic functions that stack allocate the structures.

I'm not sure if this is in fact a bug or a false-positive due to a code path that will never be taken. I'm new to the lib and too overwhelmed to follow and reason the internal code right now.  
I'm fine using a shared library. This is just something I ran into by accident.

## Background Information / Reproduction Steps

```bash
git clone --depth 1 --recurse-submodules https://github.com/open62541/open62541.git
mkdir open62541/build
cd open62541/build
cmake -DCMAKE_BUILD_TYPE=Release -DUA_BUILD_EXAMPLES=ON ..
make
```

`master` branch is `febe2e8` as I wrote and tested this.  
Same issue with the `v1.1` tag.

Used CMake options:

```bash
cmake -DCMAKE_BUILD_TYPE=Release -DUA_BUILD_EXAMPLES=ON ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): Tag: v1.1, Branch: master at febe2e8
 - [ ] Other OPC UA SDKs used (client or server): *not relevant*
 - [x] Operating system: Ubuntu 18.04.4 on x86_64
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached (see snippet of build process above)
 - [ ] Wireshark network dump attached *not relevant*
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2020-07-01:
The compiler is absolutely right. This one is cleaned up in __Subscriptions_delete_handler:

https://github.com/open62541/open62541/blob/febe2e8fd4c88dc3384188dca3db479194563c76/src/client/ua_client_subscriptions.c#L342

jpfr, 2020-07-01:
Good find!
You’re welcome to send a pr against the 1.1 branch.

ccvca, 2020-07-02:
After a short look at it, this should be already handled here:
https://github.com/open62541/open62541/blob/febe2e8fd4c88dc3384188dca3db479194563c76/src/client/ua_client_subscriptions.c#L344-L346
And this line prevents the prevents cleanup anyway as cc.isAsync is always false.
https://github.com/open62541/open62541/blob/febe2e8fd4c88dc3384188dca3db479194563c76/src/client/ua_client_subscriptions.c#L343

@insensiv-ah  What compiler do you use? We might need to update the "__clang_analyzer__" condition.

insensiv-ah, 2020-07-02:
I'm using gcc 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04) form the gcc-7 package.

teccheck, 2025-05-20:
Seems to be solved as discussed