open62541/open62541 #3757: How to enforce an encrypted connection

undroidwish, 2020-07-01:
As stated in the subject, what are the server side programmatic steps to enforce that only encrypted client connections are accepted and all others early terminated? The documentation doesn't provide enough guidance unfortunately. Access control by username/password isn't of interest, too, since the sole objective is to authenticate by cert/key and exclude everything else. Please provide some insight.

undroidwish, 2020-07-04:
After some code reading I believe this requirement can be easily achieved by

- adding a field `UA_MessageSecurityMode securityMode` to `struct UA_AccessControl` which is initialized to the value `UA_MESSAGESECURITYMODE_NONE` in function `UA_AccessControl_default()`

- adding some logic early in function `activateSession_default()` which verifies the specified mode in `struct UA_AccessControl` against the current `UA_EndpointDescription.securityLevel`, e.g.

```C
    if(ac->securityMode != UA_MESSAGESECURITYMODE_NONE) {
        if(endpointDescription->securityLevel < ac->securityMode) {
            return UA_STATUSCODE_BADSECURITYCHECKSFAILED;
        }
    }
```
BTW, a byproduct is symmetry between `struct UA_ClientConfig` and `struct UA_ServerConfig` in that in both the minimum required security level can be specified.

Opinions?

jpfr, 2020-08-13:
Cf. to what we have in the /examples folder.

Better docs are much appreciated.
If you could write down your findings, we can add a chapter in the documentation on encryption configuration.

NoelGraf, 2021-04-30:
@undroidwish what is the status? can the issue be closed?

teccheck, 2025-05-20:
Because there's been no further comment, I'll close this as done

closed 2025-05-20 by teccheck