open62541/open62541 #3760: Client alarm&event: after acknowledging several alarms continually, only get one notification

kikia1b2, 2020-07-02:
I made alarm&event Client and Server based on open62541.  I can subscribe and monitor item event correctly.  If I get an alarm and acknowledge it soon, I can get an acknowledged alarm notification.  But if I acknowlege several alarms continually, I can only get one notification which is the newest ackowleged. 

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.1.1
 - [ ] Other OPC UA SDKs used (client or server): client  & server
 - [ ] Operating system: Win7
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

oriolorra, 2022-07-11:
Could you share your server & client source code?
Please

mlang-de, 2022-07-11:
I guess what you requires could be Alarm and Conidition Branches.
https://reference.opcfoundation.org/v104/Core/docs/Part9/5.5.3/
The master says "Condition Branch not implemented".
Without this feature every time the latest Alarm are only visible.

NickBark, 2024-07-23:
Hi, I'm also very interested in implementing a client (open62541). I'm trying to create a subscription, but it's not working as it does with regular nodes. Can someone provide a hint if they've managed to get client notifications?

closed 2026-10-20 by o6-nbeier