open62541/open62541 #3761: UA Client disconnects after LifeTime counter expired second time

jki2020, 2020-07-02:
#2506 Description

Client fails in second renewing of SecureChannel. Tested against UaAnsiCServer.

## Background Information / Reproduction Steps

libopen62541.a is build of open62541-1.1.1.tar.gz files
cmake .. -DUA_ENABLE_ENCRYPTION=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON -DUA_LOGLEVEL=100

Test program is compiled
gcc -std=c99 client_connect_test.c -o client_connect_test -flto /usr/local/lib64/libopen62541.a

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.1.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

## client_connect_test.c

/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

#include <open62541/client_config_default.h>
#include <open62541/client_highlevel.h>
#include <open62541/plugin/log_stdout.h>

#include <signal.h>
#include <stdlib.h>

UA_Boolean running = true;

static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_CLIENT, "Received Ctrl-C");
    running = 0;
}

int main(void) {
    signal(SIGINT, stopHandler); /* catches ctrl-c */

    UA_Client *client = UA_Client_new();
    UA_ClientConfig *cc = UA_Client_getConfig(client);
    UA_ClientConfig_setDefault(cc);

    /* default timeout is 5 seconds. Set it to 1 second */
    cc->timeout = 1000;

    cc->secureChannelLifeTime = 2 * 60 * 1000; /* 2 minutes */

    /* Endless loop */
    while(running) {
        /* if already connected, this will return GOOD and do nothing */
        /* if the connection is closed/errored, the connection will be reset and then reconnected */
        /* Alternatively you can also use UA_Client_getState to get the current state */
        UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://cpm-geysir:48020");
        if(retval != UA_STATUSCODE_GOOD) {
            UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_CLIENT, "Not connected. Retrying to connect in 1 second");
            /* The connect may timeout after 1 second (see above) or it may fail immediately on network errors */
            /* E.g. name resolution errors or unreachable network. Thus there should be a small sleep here */
            UA_sleep_ms(1000);
            continue;
        }
        retval = UA_Client_run_iterate(client, 1000);
    };
[... 656 lines cut ...]
[2020-07-02 12:56:21.408 (UTC+0300)] debug/client       Client Status: ChannelState: HELSent, SessionState: Created, ConnectStatus: Good
[2020-07-02 12:56:21.408 (UTC+0300)] trace/client       Client connect iterate
[2020-07-02 12:56:21.409 (UTC+0300)] debug/network      Received ACK message
[2020-07-02 12:56:21.409 (UTC+0300)] debug/client       Client Status: ChannelState: AckReceived, SessionState: Created, ConnectStatus: Good
[2020-07-02 12:56:21.409 (UTC+0300)] trace/client       Client connect iterate
[2020-07-02 12:56:21.409 (UTC+0300)] debug/channel      Connection 3 | SecureChannel 0 | Generating new local nonce
[2020-07-02 12:56:21.409 (UTC+0300)] debug/channel      Connection 3 | SecureChannel 0 | Requesting to open a SecureChannel
[2020-07-02 12:56:21.409 (UTC+0300)] debug/channel      Requesting to open a SecureChannel
[2020-07-02 12:56:21.409 (UTC+0300)] debug/channel      OPN message sent
[2020-07-02 12:56:21.409 (UTC+0300)] debug/client       Client Status: ChannelState: OPNSent, SessionState: Created, ConnectStatus: Good
[2020-07-02 12:56:21.409 (UTC+0300)] trace/client       Client connect iterate
[2020-07-02 12:56:21.409 (UTC+0300)] trace/channel      Connection 3 | SecureChannel 0 | Verifying chunk signature
[2020-07-02 12:56:21.409 (UTC+0300)] trace/channel      Connection 3 | SecureChannel 2675354676 | Generating new local keys
[2020-07-02 12:56:21.409 (UTC+0300)] info/channel       Connection 3 | SecureChannel 2675354676 | Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2020-07-02 12:56:21.409 (UTC+0300)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2020-07-02 12:56:21.409 (UTC+0300)] trace/client       Client connect iterate
[2020-07-02 12:56:21.409 (UTC+0300)] debug/channel      Connection 3 | SecureChannel 2675354676 | Sending request with RequestId 8 of type ActivateSessionRequest
[2020-07-02 12:56:21.409 (UTC+0300)] debug/client       Client Status: ChannelState: Open, SessionState: ActivateRequested, ConnectStatus: Good
[2020-07-02 12:56:21.409 (UTC+0300)] trace/client       Client connect iterate
[2020-07-02 12:56:21.409 (UTC+0300)] trace/channel      Connection 3 | SecureChannel 2675354676 | Generating new remote keys
[2020-07-02 12:56:21.410 (UTC+0300)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2020-07-02 12:56:22.411 (UTC+0300)] warn/client        Client already connected
[2020-07-02 12:56:23.412 (UTC+0300)] warn/client        Client already connected
[2020-07-02 12:56:24.413 (UTC+0300)] warn/client        Client already connected
[2020-07-02 12:56:25.414 (UTC+0300)] warn/client        Client already connected

jpfr, 2020-07-08:
Thanks for the detailed description. Will investigate.

jpfr, 2020-07-08:
The problem occurred if no messages are exchanged between two SecureChannel renew.
This fix for 1.1 and later is in #3774 and will be merged when green.

NoelGraf, 2021-04-30:
I close the issue because the pull request #3774 is merged now.

closed 2021-04-30 by NoelGraf